MKFirmware/MK4duo #200: Is it possible to add in the configurator - Motherboard - Own layout defined in userpins.h?

DAGal86, 2017-08-20:
Hello.
I have a DUE and a board production of my friend.
Is it possible to add in the configurator - Motherboard - Own layout defined in userpins.h?

Such a thing is in Repetier, but with him a problem arose for some reason only in Repetier-Host 1.6.1 it is possible to control the motor of the extruder in other similar programs the motor of the extruder does not react ... and the smooth printing of circles does not work. G2 / G3 does not work.

In Simplify3D 4.0.0 there is a function that allows you to smoothly print circles, but there does not work controlling the motor of the extruder ...

Team Repetier not in a hurry to solve the problem.

With the Simplify3D command, we went through all the possible options for setting ...

Thank you.

DAGal86, 2017-08-20:
I have the files for my Shield for DUE.
Userpins - feet for DUE.
Uiconfig - display management.
[RaMS userpins uiconfig.zip](https://github.com/MKFirmware/MK4duo/files/1237100/RaMS.userpins.uiconfig.zip)

iosonopersia, 2017-08-20:
Hi, as far as custom pins are concerned, in MK4duo you can manually edit the file src/boards/99.h, which is intended for using custom boards. Be careful because pin macros have different names across different firmwares... You can take as an example one of the other files in src/boards. Then you should configure the printer MOTHERBOARD as BOARD_99 in Configuration_Basic.h .

On the other hand, I believe MK4duo doesn't support custom display management. Here only @MagoKimbra can help you.

DAGal86, 2017-08-20:
Hi, I'm trying so hard but I had a question I do not have a paramitra
#define HEATER_BED_PIN
He is in all Marlin
- here it is called
#define ORIG_HEATER_BED_PIN, there is a leg 4 for it.
Can you tell me what kind of foot is it for the table? Or if it is not in my case and it can be ignored and put "-1"?

iosonopersia, 2017-08-20:
Have you got an heated bed? If so, you should also have a dedicated pin on your board to heat it up. If not, you can leave it to -1, which means "disabled"

DAGal86, 2017-08-20:
Heated bed is, but there is no this link is only #define ORIG_HEATER_0_PIN.

It turns out in marlin two pins on the table 
#define ORIG_HEATER_0_PIN 
and 
#define ORIG_HEATER_BED_PIN.

What is the second pin used for - #define ORIG_HEATER_BED_PIN?

iosonopersia, 2017-08-21:
ORIG_HEATER_0_PIN is the pin of the nozzle heating cartridge, while ORIG_TEMP_0_PIN is the pin of the nozzle temperature sensor. (There's a 0 because there are also ORIG_HEATER_1_PIN, ORIG_HEATER_2_PIN, ORIG_HEATER_3_PIN and their relative temperature sensors, just in case your board supports up to 4 nozzles).

ORIG_HEATER_BED_PIN is the pin to heat the bed, while ORIG_TEMP_BED_PIN is the pin of the bed temperature sensor.

DAGal86, 2017-08-21:
Thanks for the detailed explanation.
Now everything is clear to me.
I really have three ports for the extruder.
I use one port for the second Z-axis motor.

I have somewhere to hide.
I will look for.
On the board I know these feet are left to find them on the list.
But I think it's a worthwhile task.

Can you tell marlin what command is responsible for the second Z axis motor?

iosonopersia, 2017-08-21:
Info about WHAT pins you have in your custom board should go in 99.h file (and as I've understood you've already done that.

Info about HOW you use those pins should go in the Configuration files. I think the best and simpler way for you to configure the printer is by using the online configurator (http://marlinkimbra.it/configurator/v4_3_x/firmware_configurator.php), which is also very similar to the Repetier online configurator.

In the Mechanics tab, you have to scroll down to the bottom of the page to find the "Z two stepper" option. Select it and select also "Z Two EndStop" if you have an endstop for each Z motor. Then scroll up and in the "Z2 stepper socket" combobox choose the extruder socket of your board that you will use as the Z2 motor (choose between Extruder 0, 1 and 2, which will correspond to the pins you have set in 99.h).

Then configure every aspect of your printer.

AAA: please pay attention, the online configurator doesn't show you the option to select BOARD_99 as your MOTHERBOARD. You can choose casually one board, then, when you will have finished the configuration and downloaded the firmware, you will have to open Configuration_Overall.h and replace "#define MOTHERBOARD something" with "#define MOTHERBOARD BOARD_99", otherwise error will occur while compiling.

If you prefer downloading not the latest version (4.3.26) but the last stable (4.3.2), here is the link: http://marlinkimbra.it/configurator/v4_3_2/firmware_configurator.php

DAGal86, 2017-08-21:
Thanks for the valuable tips!

iosonopersia, 2017-08-22:
@DAGal86 remember to tell us if you succeed or not!

DAGal86, 2017-08-22:
Currently prescribed pins. 
came the question. 
#define X_STOP_PIN, and similar for Y and Z, as I understand Endstop ?

iosonopersia, 2017-08-22:
X_STOP_PIN can be used to overwrite the ORIG_X_MIN_PIN and the ORIG_X_MAX_PIN values. The same for Y and Z axis. You should do it if you have only one endstop per axis, but I recommend you to leave them disabled and to set ORIG_X_MIN_PIN and ORIG_X_MAX_PIN instead. 
``` cpp
#define X_STOP_PIN -1
#define Y_STOP_PIN -1
#define Z_STOP_PIN -1
```

This is the logic behind it (you can find it in src/pins.h):
``` cpp
#if PIN_EXISTS(X_STOP)
  #undef ORIG_X_MIN_PIN
  #undef ORIG_X_MAX_PIN
  #if X_HOME_DIR < 0
    #define ORIG_X_MIN_PIN X_STOP_PIN
    #define ORIG_X_MAX_PIN -1
  #else
    #define ORIG_X_MIN_PIN -1
    #define ORIG_X_MAX_PIN X_STOP_PIN
  #endif
#endif
```

DAGal86, 2017-08-23:
Now when I moved all the pins to 99.h, when I compile, I get this error:

In file included from sketch\src\../base.h:82:0,

                 from sketch\src\MK_Main.cpp:31:

sketch\src\MK_Main.cpp: In function 'bool pin_is_protected(uint8_t)':

sketch\src\../src/pins.h:444:24: warning: narrowing conversion of '4294967295u' from 'unsigned int' to 'const int' inside { } [-Wnarrowing]

                        }

                        ^

sketch\src\MK_Main.cpp:582:39: note: in expansion of macro 'SENSITIVE_PINS'

   static const int sensitive_pins[] = SENSITIVE_PINS;

                                       ^

sketch\src\../src/pins.h:444:24: warning: narrowing conversion of '4294967295u' from 'unsigned int' to 'const int' inside { } [-Wnarrowing]

                        }

                        ^

sketch\src\MK_Main.cpp:582:39: note: in expansion of macro 'SENSITIVE_PINS'

   static const int sensitive_pins[] = SENSITIVE_PINS;

                                       ^

sketch\src\../src/pins.h:444:24: warning: narrowing conversion of '4294967295u' from 'unsigned int' to 'const int' inside { } [-Wnarrowing]

                        }

                        ^

sketch\src\MK_Main.cpp:582:39: note: in expansion of macro 'SENSITIVE_PINS'

   static const int sensitive_pins[] = SENSITIVE_PINS;

                                       ^

sketch\src\../src/pins.h:444:24: warning: narrowing conversion of '4294967295u' from 'unsigned int' to 'const int' inside { } [-Wnarrowing]

                        }

                        ^

sketch\src\MK_Main.cpp:582:39: note: in expansion of macro 'SENSITIVE_PINS'

   static const int sensitive_pins[] = SENSITIVE_PINS;

                                       ^

sketch\src\MK_Main.cpp: In function 'void gcode_M92()':

sketch\src\MK_Main.cpp:7660:88: error: invalid conversion from 'int' to 'AxisEnum' [-fpermissive]

         const float value = code_value_per_axis_unit((AxisEnum)E_AXIS + TARGET_EXTRUDER);

                                                                                        ^

sketch\src\MK_Main.cpp:1184:16: error:   initializing argument 1 of 'float code_value_per_axis_unit(AxisEnum)' [-fpermissive]

   inline float code_value_per_axis_unit(const AxisEnum axis) { return code_value_float() / axis_unit_factor(axis); }

                ^

exit status 1
Ошибка компиляции для платы Arduino Due (Native USB Port).

DAGal86, 2017-08-23:
My firmware.

iosonopersia, 2017-08-23:
I'll check your firmware. You accidentally closed this issue, please re-open it...

DAGal86, 2017-08-23:
Thank you very much! 
re-opened.

iosonopersia, 2017-08-23:
Ok, I've found something interesting:

1. to solve your error, a newer version of the firmware should be used. I chose the 4.3.26 dev;
2. I moved your configuration to the new version of the firmware. I've found two little bugs (one in temperature.cpp and one in stepper.cpp, these will be fixed soon). I've manually fixed them immediately for you;
3. then I've encountered another problem: you didn't define the ORIG_TEMP_BED_PIN in your 99.h (it's set to -1). If you have a temperature sensor pin on your board, you should set it properly. If you don't have that pin, then you CAN'T select in the configurator the PIDTEMPBED option, as you did, because this requires the sensor and the sensor pin is disabled. I solved the error by defining the pin in 99.h as 10, but you should fix it as soon as possible!
4. then, another problem again. Are you really sure your board has EEPROM support? If so, you should define some other pins in your 99.h:
``` cpp
#define SPI_EEPROM
#define SPI_CHAN_EEPROM1      <PIN>
#define SPI_EEPROM1_CS        <PIN>
#define SPI_EEPROM2_CS        <PIN>
```
In order to get the firmware to compile, I've modified your configuration to disable EEPROM settings. Now the firmware compiles, so you can start making your tests with it.

If you need help with the EEPROM support we will discuss it another time. The important thing now is to verify if MK4duo works well on your board :+1: 

Download the fixed version of your firmware here (and fix ORIG_TEMP_BED_PIN in 99.h!):
https://www.dropbox.com/s/tf4u3ezt7r7cg6z/MK4duo-for-DAGal86.zip?dl=0

DAGal86, 2017-08-23:
Thank you very much for your work!
Unfortunately, I have not yet determined which pins are responsible for the temperature sensor and the heating of the bed. The author of the board is also not responding yet.
 Because I decided to turn off these pins.
In the evening I will try to look for more.
EEPROM I have an installation.
I probably forgot to pin it for him, I'll figure out their numbers.
But to begin with I will check without it.

If you can not find the pins for the bed, I'll make a new firmware in the configurator, replace the file Configuration_Overall.h, I'll set up 99.h
The results will be reported!

DAGal86, 2017-08-24:
Unfortunately the result is disappointing.

I still could not find the pin numbers for the bed. Tried free.

The firmware that you sent was compiled and poured into the controller.

Repetier-Host 1.6.1 is connected, but no command is executed.
I tried to give commands to the movement along the axes, go home - each axis separately and all together.
Even asked for the status EndStop.
All without success.
No response and no action.

Simplify3D 4.0.0 was not even able to connect.

I tried to use both ports of DUE for connection (Navite and Programming) - the result is the same.

I made the firmware through the configurator 4.3.26dev with the disabled hot-sticking and EEPROM

I copied you the modified Configuration_Overall.h

Adjusted the pins in 99.h

But there is an error:

Arduino: 1.8.3 (Windows 7), Fee: "Arduino Due (Native USB Port)"

Sketch \ src \ HAL \ HAL_DUE \ HAL_spi_Due.cpp: In function 'void spiSend (const uint8_t *, size_t)':

Sketch \ src \ HAL \ HAL_DUE \ HAL_spi_Due.cpp: 117: 51: error: 'void spiSend (const uint8_t *, size_t)' was declared 'extern' and later 'static' [-fpermissive]

   Static void spiSend (const uint8_t * buf, size_t n) {

                                                   ^

In the file included from sketch \ src \ HAL \ HAL_DUE \ ../../../ base.h: 85: 0,

                 From sketch \ src \ HAL \ HAL_DUE \ HAL_spi_Due.cpp: 40:

Sketch \ src \ HAL \ HAL_DUE \ ../../../ src / HAL / HAL.h: 47: 6: error: previous declaration of 'void spiSend (const uint8_t *, size_t)' [-fpermissive]

 Void spiSend (const uint8_t * buf, size_t n);

      ^

Exit status 1
Compilation error for the Arduino Due card (Native USB Port).

DAGal86, 2017-08-24:
4.3.26dev firmware with disabled hot crawl and EEPROM

I copied you the modified Configuration_Overall.x
Adjusted the pins in 99.h

https://www.dropbox.com/s/9zwmz64videqi6r/MK4duo_4_3_26_dev_24-08-2017.rar?dl=0

iosonopersia, 2017-08-24:
Hi, sorry for that. The situation is this: the owner of the repository hasn't been active for 8 days, so a lot of fixes haven't been merged yet. For this reason, 4.3.26dev is NOT the most updated version of MK4duo. When you use the online configurator, it downloads automatically the 4.3.26 version.

The error you had compiling 4.3.26 was already fixed in my repository but my changes were not merged (for the reason I told you before). When I sent you the link to the firmware to download from Dropbox, I did not used 4.3.26, I used my more up to date version, and then I replaced configuration files with yours (also fixing them, as I've already told you).

**Now, what should you do?**
Start from the version I've sent you. That version compiles (and hopefully works).
If you need to modify the configuration, follow these steps:
- **pin values edits**: change manually 99.h and recompile
- **other changes**: open a browser, go to the online configurator. Load the `Configuration_Overall.h`, apply your changes, download the firmware. That firmware is 4.3.26 so it won't compile, it will give always the same error (until MagoKimbra will become active again). You should do 3 things:
1. Copy the newly downloaded `Configuration_Overall.h` in the firmware I sent you
2. Copy the newly downloaded `Configuration_Pins.h` in the firmware I sent you
3. Recompile the firmware I sent you

I know this maybe can create a bit of confusion... But we have to be patient and to wait the return of @MagoKimbra, until then fixes won't be merged in the main repository.

Let me know if you didn't understand something

iosonopersia, 2017-08-24:
You should always use the most updated version of the firmware.
If you can't connect and communicate with your board, maybe is because you set the wrong serial port in the firmware. To change it in the online configurator, try selecting "default port" in "Primary port" section.

DAGal86, 2017-08-24:
Your firmware is being compiled.
I changed the Primary Port (SERIAL_PORT) to all three options.
Everything works the same way.
Repetier-Host connects, but the commands do not work.
Simplify3D does not connect.

I tried replacing the files in the firmware without the bed and EEPROM for, but the error is the same.
Also found and replaced the files that you said before temperature.cpp and stepper.cpp
But it did not help either.
Perhaps you have something else corrected somewhere!

iosonopersia, 2017-08-24:
Nooo, the 2 config files I told you to replace should be taken from the folder downloaded from the online configurator and put in the folder of the firmware I sent you via Dropbox, you did the exact contrary :) You should compile the firmware I sent you. Use the configurator only to generate Configuration_Overall.h and Configuration_Pins.h. Then discard the firmware from the configurator and use the new config files in the firmware I sent you. Otherwise you will continue to get that same error over and over...

That being said, we should fix the serial communication problem you have. When you connect the board, do you see any messages? Are you selecting the right com port? The right baudrate (the same you set during configuration)? Are you using a good USB cable?

DAGal86, 2017-08-24:
I understand you about the configurator!

When I connect to a computer, I monitor the status of the COM port.
Everything corresponds.

USB cable is reliable.

iosonopersia, 2017-08-24:
And do you get any messages from the board???

DAGal86, 2017-08-24:
I see that the controller is connected and in the Device Manager I see the COM port number and to which DUE (Navite or Programming) port is connected.
Arduino IDE also sees all this data and it matches the Device Manager data.

iosonopersia, 2017-08-24:
Ok, I've forgotten to say that before replacing Configuration_Overall.h you should also correct the MOTHERBOARD value with BOARD_99...

iosonopersia, 2017-08-24:
Ok, look at this picture: it's Pronterface but it's the same with Repetier Host. Do you see those messages coming from the board when you connect or not? This is what I wanted to know... In case you get similar messages, could you please copy and paste them here as I previously asked you? They can be useful to undestand what's happening, it's too difficult to me wondering what happens on your computer, I'm not a magician...

[image]

DAGal86, 2017-08-24:
I understand!
We say that - not clairvoyant!
I tried to connect through Pronterfake from 02/03/2015
Here's what he said: 
Connecting...
Traceback (most recent call last):
  File "printrun\pronterface.pyc", line 1053, in connect
  File "printrun\pronsole.pyc", line 720, in connect_to_printer
  File "printrun\printcore.pyc", line 46, in inner
  File "printrun\printcore.pyc", line 208, in connect
  File "encodings\utf_8.pyc", line 16, in decode
UnicodeDecodeError: 'utf8' codec can't decode byte 0xcd in position 36: invalid continuation byte

DAGal86, 2017-08-24:
Filled Repetier-Firmware which at least somehow works for me.
The printer responds to Pronterface commands.
The truth is also not controlled motor of the extruder.
The team is sent but there is no reaction.

iosonopersia, 2017-08-24:
Ahah ok, my native language is Italian, I didn't know the word "clairvoyant" :-)

Sorry, but sometimes it's really difficult to understand the problem, always send every info that you have...

What do you get from Repetier-Host?

DAGal86, 2017-08-24:
Repetier-Host:
23: 05: 59.998: No start signal detected - forcing start
23: 06: 00.029: N1 M110 * 34
23: 06: 00.029: N2 M115 * 36
23: 06: 00.029: N3 M105 * 36
23: 06: 00.029: N4 M114 * 35
23: 06: 00.029: N5 M111 S6 * 98
23: 06: 00.029: N6 T0 * 60
23: 06: 00.029: N7 M20 * 22
23: 06: 00.029: N8 M80 * 19
23: 06: 03.071: N9 M105 * 46
23: 06: 06.129: N10 M105 * 22
23: 06: 09.187: N11 M105 * 23
23: 06: 12.244: N12 M105 * 20
23: 06: 15.302: N13 M105 * 21
23: 06: 18.359: N14 M105 * 18

Constantly the team M105
And if you give your commands, then 3-4 teams hang.

DAGal86, 2017-08-24:
Attempting connection at \\.\COM4...
Simplify3D:
  Testing plaintext communication protocol...
  Testing binary communication protocol...
  Testing alternate communication protocols...
  Connection failed.

iosonopersia, 2017-08-24:
Ok, actually you cannot connect with any host software... That's not good, but you said that you're able to connect when using Repetier Firmware. That suggest no problems with usb drivers or similar things.

Just a last thing: can you please send here the last version of the firmware you tried? (The one with every config change you made) When MagoKimbra will become active again, we'll investigate the problem, thanks. I'll also keep a copy of that, I will update this issue if I have any update. Don't close this issue. Bye

MagoKimbra, 2017-08-24:
Hi I'm back from vacation now, I read everything and then I get you some help !!!

DAGal86, 2017-08-24:
Actually this is the firmware that you sent me! 
https://www.dropbox.com/s/6cotcgl4wlejotl/MK4duo-for-DAGal86_24_08_217.rar?dl=0

DAGal86, 2017-08-24:
Hello!
I hope you are well rested!
I will be glad!
Good luck to all of us!

DAGal86, 2017-08-30:
Kind time of the day!
I saw that there were new update dates in the code list.
Do I correctly understand that you have updated the firmware and you can try to configure and use it?

iosonopersia, 2017-08-30:
Hi @DAGal86. Yesterday MagoKimbra made big modifications to the heaters management code, and other changes will follow... The problem is that the new code has still to be tested and I think you will find new errors :-(  We encourage you to try the newer version, it would be helpful to spot new errors, it would help us fixing the code. Sorry for this...

DAGal86, 2017-08-30:
And the configurator is not done yet for this firmware?
I already tried to download and configure, but the Configuration_Overall.h file in this version is empty.
And where these settings have been transferred, I do not know where to look and where else in the other place to adjust, I still do not know well.
And I can not exactly find where to turn on the second Z axis motor
Do not worry, I understand that this is hard and tedious work!
"And only those who do not do not have mistakes!"

DAGal86, 2017-09-04:
Saw that there was an online configurator for * .27 but in the end got: 

Not Found

The requested URL /configurator/v4_3_x/Parse error: syntax error, unexpected ')' in /customers/5/9/7/marlinkimbra.it/httpd.www/configurator/v4_3_x/config.php on line 454 was not found on this server.

MagoKimbra, 2017-09-04:
Sorry.. my error. i fixed file config.php in configurator online

DAGal86, 2017-09-04:
the firmware was compiled.
Thank you!
but the control programs did not connect.
I thought maybe there was a reason in choosing a number for #define SERIAL_PORT
Became me numbers and switch between the Native and Programming port
and in the reconnection process, the DUE is out of order.
Now I can not determine the reason.
Unfortunately, we'll have to wait until the new DUE board arrives ...

I have also Mega2560
Does it apply to your patches in the new version?

iosonopersia, 2018-09-07:
Hi @DAGal86 , an entire year has passed. A lot of changes have been made to the code. Just to answer to your last post: yes, every patch that I made during this discussion was merged in the new version of the firmware.

This discussion was interrupted because you had to wait for a new Arduino DUE board, and the conversation stopped at that point. We don't know if you have fixed that problem in the meantime.

I'm going to close this old issue. If you need help with newer versions of MK4duo (see V4_3_6), feel free to open a new issue.

closed 2018-09-07 by iosonopersia