MKFirmware/MK4duo #204: A couple compile warnings/errors in branch 'dev'

remspoor, 2017-08-27:
### Issue description
I get compile warnings/errors when trying to get de current 'dev' branch running.

While i used the 'dev' branch a couple of months, the last working commit for me is 97b1a8ef3b4de935efcfb55c3dc73063f69b99be. I left the compile errors alone for a while hoping they would go away by itself ;) and used my own 'working' branch for the time being.
My config was initially generated with the web configurator but edited after that. Hopefully it's still accurate for current 'dev'
#### How to create the errors:
- ```git checkout upstream/dev```
- copy my Configuration_Overall.h file
- open arduino
- build
### Relevant compile warnings/errors
```
In file included from /tmp/arduino_build_802134/sketch/src/commands/../../base.h:41:0,
                 from /tmp/arduino_build_802134/sketch/src/commands/commands.cpp:29:
/tmp/arduino_build_802134/sketch/src/commands/../gcode/calibrate/g29_mbl.h: In function 'void gcode_G29()':
/tmp/arduino_build_802134/sketch/src/commands/../../src/macros.h:113:37: warning: comparison is always true due to limited range of data type [-Wtype-limits]
 #define WITHIN(V,L,H)     ((V) >= (L) && (V) <= (H))
                                     ^
/tmp/arduino_build_802134/sketch/src/commands/../gcode/calibrate/g29_mbl.h:65:10: note: in expansion of macro 'WITHIN'
     if (!WITHIN(state, 0, 5)) {
          ^
In file included from /tmp/arduino_build_802134/sketch/src/commands/../gcode/gcode.h:67:0,
                 from /tmp/arduino_build_802134/sketch/src/commands/commands.cpp:30:
/tmp/arduino_build_802134/sketch/src/commands/../gcode/control/m42.h: In function 'void gcode_M42()':
/tmp/arduino_build_802134/sketch/src/commands/../gcode/control/m42.h:55:35: warning: suggest parentheses around assignment used as truth value [-Wparentheses]
       if (fans.pin[f] = pin_number) {
                                   ^
In file included from /tmp/arduino_build_802134/sketch/src/commands/../gcode/debug/m43.h:31:0,
                 from /tmp/arduino_build_802134/sketch/src/commands/../gcode/gcode.h:87,
                 from /tmp/arduino_build_802134/sketch/src/commands/commands.cpp:30:
/tmp/arduino_build_802134/sketch/src/commands/../gcode/debug/../../utility/pinsdebug.h: In function 'bool pwm_status(uint8_t)':
/tmp/arduino_build_802134/sketch/src/commands/../gcode/debug/../../utility/pinsdebug.h:808:10: warning: case label value is less than minimum value for type [enabled by default]
     case NOT_ON_TIMER:
          ^
/tmp/arduino_build_802134/sketch/src/commands/../gcode/debug/../../utility/pinsdebug.h:770:8: warning: unused variable 'buffer' [-Wunused-variable]
   char buffer[20];   // for the sprintf statements
        ^
In file included from /tmp/arduino_build_802134/sketch/src/commands/../gcode/gcode.h:87:0,
                 from /tmp/arduino_build_802134/sketch/src/commands/commands.cpp:30:
/tmp/arduino_build_802134/sketch/src/commands/../gcode/debug/m43.h: In function 'void gcode_M43()':
/tmp/arduino_build_802134/sketch/src/commands/../gcode/debug/m43.h:259:9: error: 'wait_for_user' was not declared in this scope
         wait_for_user = true;
         ^
In file included from /tmp/arduino_build_802134/sketch/src/commands/../gcode/gcode.h:137:0,
                 from /tmp/arduino_build_802134/sketch/src/commands/commands.cpp:30:
/tmp/arduino_build_802134/sketch/src/commands/../gcode/mixing/m163_m165.h: In function 'void gcode_M164()':
/tmp/arduino_build_802134/sketch/src/commands/../gcode/mixing/m163_m165.h:63:23: error: 'normalize_mix' was not declared in this scope
         normalize_mix();
                       ^
/tmp/arduino_build_802134/sketch/src/commands/../gcode/mixing/m163_m165.h: In function 'void gcode_M165()':
/tmp/arduino_build_802134/sketch/src/commands/../gcode/mixing/m163_m165.h:87:48: error: 'gcode_get_mix' was not declared in this scope
   inline void gcode_M165(void) { gcode_get_mix(); }
                                                ^
In file included from /tmp/arduino_build_802134/sketch/src/commands/../gcode/gcode.h:211:0,
                 from /tmp/arduino_build_802134/sketch/src/commands/commands.cpp:30:
/tmp/arduino_build_802134/sketch/src/commands/../gcode/tools/tcode.h: In function 'void gcode_T()':
/tmp/arduino_build_802134/sketch/src/commands/../gcode/tools/tcode.h:41:11: warning: unused variable 'tool_id' [-Wunused-variable]
   uint8_t tool_id = parser.codenum;
           ^
```

### Firmware version
Current 'dev' branch
### My setup
- Arch linux
- Arduino version 1.8.4
- 'Arduino SAM boards' extension version 1.6.11
- Ultratronics board (1705.h)
- Diamond hotend ( 3 extruders )

My config can be found on github [here](https://github.com/remspoor/MK4duo/blob/cebe4d3b303ab36c2af27c6cf26e0c05577fca70/MK4duo/Configuration_Overall.h) or the attached version [Configuration_Overall.h.txt](https://github.com/MKFirmware/MK4duo/files/1254751/Configuration_Overall.h.txt)
### Thanks!!
I'm not a programmer, but i can always try to help debug stuff with an Ultratronics board. I also like the rework of the code going on at the moment. I think it's becoming more readable.
Keep up the good work.

iosonopersia, 2017-08-28:
Thank you very much for the valuable information you gave us. A lot of the warnings/errors you found will be fixed in my new pull request. Thank you again for helping us writing better code!

@MagoKimbra , after fixing these problems I get a lot of other errors from ultralcd_impl_HD44780.h which are very similar to the errors we get in #199 ... I don't know how that part of the code works, could you please try to fix that issue? In both cases 32bit electronics is used...
[error_messages.txt](https://github.com/MKFirmware/MK4duo/files/1256405/error_messages.txt)

iosonopersia, 2017-09-03:
Hi @remspoor , some fixes were made, now the firmware should compile... Some config macros are now used and defined in a different way, so you can't use your Configuration_Overall.h with the the latest version of the firmware... Could you please redo the configuration here: http://marlinkimbra.it/configurator/v4_3_x/firmware_configurator.php ?

remspoor, 2017-09-06:
Hello,
I'm back home again and first thing i did was merge my old config to a recent one. I did a diff from a configurator config to double check.

A lot of errors seem to have gone away, but i got new/other errors, see attached 
[new-errors-git-a7f2ec3.txt](https://github.com/MKFirmware/MK4duo/files/1281040/new-errors-git-a7f2ec3.txt) file. It looks like the same errors already mentioned above about the MENU_ITEMS appear as well.

My config file can be found in my repo (see above) or just this [Configuration_Overall.h.txt](https://github.com/MKFirmware/MK4duo/files/1281050/Configuration_Overall.h.txt) file.

MagoKimbra, 2017-09-06:
Ok i found the problem.. I fix it and upload new file the envening...

remspoor, 2017-09-06:
That would be great!
I tried 3bca5b9, but that's still not working though (hope i'm not to fast today ;) )

commit af4b692b (2017-09-06): Fix #204
closed 2017-09-06 by MagoKimbra