MKFirmware/MK4duo #207: Does MBL work with z probe?

hon-po, 2017-08-30:
V4.3.2 stable on 8 bit Mega + Ramps. The manual MBL works wonderful. But with fix mounted probe, G29 S1 stop after homing, no probing occurs. I tried 4.3.26 dev, but it fail to compile.

Does MBL work with zprobe? If yes, which version and with which probe type is known to work? Config for V4.3.2 attached.
[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/1263263/Configuration_Overall.txt)

MagoKimbra, 2017-08-30:
MBL work with probe.. Does the Zprobe work properly? Have you tried with M119?

hon-po, 2017-08-30:
Thanks for the reply. Z probe is good as G30 works.

My setup is corexy with a magnetically attached switch as z probe, it will be mounted by hand to do occasional probing. Could not find a suitable probe type, so I just try fix mounted probe to try the function first. Not sure if it is a config problem.

Any known working config.h for MBL with z probe that I could reference?

hon-po, 2017-08-31:
May be V4.3.2 is too old? I found probe_pt() in:
double_home_z()
gcode_G29() ---- ABL_GRID part
gcode_G29() ---- AUTO_BED_LEVELING_3POINT part
gcode_G30()
gcode_G33()

But not in the MBL part.

In V4.3.26 dev, couldn't identify where the probing is done. Any hint?

MagoKimbra, 2017-08-31:
Ok, I'll do MBL control with the probe, for now I can do the linear or bilinear ABL while I find the cause of the problem ...

hon-po, 2017-09-04:
After enable bilinear ABL, I have stuttering on XY movement. May be 8 bit electronics not fast enough.

closed 2018-08-02 by iosonopersia