MKFirmware/MK4duo #208: Passare da Marlin 1.1.4 a MarlinKimbra?

EnricoSx, 2017-08-30:
[Marlin.zip](https://github.com/MKFirmware/MK4duo/files/1264088/Marlin.zip)
c'è un modo facile per passare da marlin 1.1.4 a marlinkimbramk4duo?
perchè cercando i parametri del configuratore online per la delta anycubic linear plus, non trovo i corrispettivi o simili parametri nei file di marlin.

MagoKimbra, 2017-08-30:
Ci sono!!! Controlla bene il file configuration.h e advanced..
Sono gli stessi perché cmq MK deriva da marlin poi alcune cose sono cambiate, ma le basi sono quelle..

EnricoSx, 2017-08-30:
i valori sono diversi da quelli nel manuale della linear plus

MagoKimbra, 2017-08-30:
Strano quella configurazione è stata fatta da chi aveva una linear plus.
Domani vedo se riesco a farti la configurazione partendo dal marlin che hai messo...

EnricoSx, 2017-08-30:
io credo che abbiano cambiato versione nel frattempo.
io ne so zero di stampanti 3d, ero partito da questo originale aggiornato
http://www.anycubic3d.com/en/col.jsp?id=158
nel manuale che ho a casa, c'è scritto se hai la linear plus cambia tot parametri, anche se ho l'impressione che siano pure sbagliati, come la lunghezza dei "bracci"

MagoKimbra, 2017-08-30:
E ho capito, ma da qualche parte dobbiamo partire... Io ti ripeto che quella configurazione è stata fatta da chi ha la linear plus quindi presumo siano valori esatti visto che lo usa...
Informati e vediamo..

EnricoSx, 2017-08-30:
ti ho allegato le foto

EnricoSx, 2017-08-30:
una differenza con la configurazione preconfigurata del sito è che propone una ramps 1.3, ma la mia ha una trigorilla

EnricoSx, 2017-08-31:
[image]
la Anycubic mi ha scritto questi dati, chissa se sono reali :/

Drakelive, 2017-09-25:
Per esperienza vi dico che le Anycubic, sia le Mini che le Plus hanne sempre le Rods di lunghezza differente quindi è bene controllare ogni volta e riportare i valori. 
Poi su 10 stampanti acquistate per un worlshop abbiamo anche visto che ci sono differenze gia tra le 6 Rods presenti nel singolo kit.... un delirio...... 

Abbiamo cercato di sistemare la cosa "accoppiando" quelle che erano della stessa misusa per Torre e correggendo i valori del firmware ( MK4Duo per fortuna lo permette ) ed effettivamente la situazione è migliorata...... ma per avere una situazione ottimale abbiamo cambiato le Rods.
Grazie ad una dima abbiamo rifatto le Rods con dei tubicini di Alluminio preso al brico e recuperato i giunti a sfera. Un po di colla e via.... come nuovi .... vanno benissimo. 

Poi ogni tanto capita di trovare qualche snodo un po lasco .... anche se questa cosa si risolve aggiungendo dei buoni elestici. 

Alla fine non ninete da dire da questi kit delta, sono migliorati tantissimo nel tempo, ma l'unico punto negativo sono le Rods.

MagoKimbra, 2017-09-25:
Allora rispetto a marlin normale su MK ci sono delle piccole differenze HEATER_PWM_SPEED serve per aumentare o diminuire la frequenza pwm per i riscaldatori, per ora lascia come trovi, poi con calma e quando capirai cosa significa potrai cambiarla e fare prove.
La trigorilla c'è ma l'ho messa solo sulla versione in sviluppo la dev. Per ora metti la ramps.
Per le misure facci un post su FB nel gruppo Anycubic...

iosonopersia, 2018-07-31:
Da quello che ho capito questa issue può essere chiusa @MagoKimbra

closed 2018-07-31 by MagoKimbra