MKFirmware/MK4duo #210: DELTA_HEIGHT ?

EnricoSx, 2017-08-31:
perdona il disturbo e l'ignoranza.
ho usato il configuratore online, dentro al quale ho messo come delta height 295.3, caricato il tutto con arduino ide, in configuration_overall.h c'è scritto #define DELTA_HEIGHT 295.3
premuto > carica.
dal display muovo l'asse z, porto il carrello in alto ed arriva a Z300 .
Cosa sbaglio? grazie

MagoKimbra, 2017-08-31:
Fai M666 L e leggi il valore di Z height

EnricoSx, 2017-08-31:
sono quelli scritti sopra Config:H (Z-Height): 300.000

MagoKimbra, 2017-08-31:
A non lo avevo letto... Scusami.. Ok allora sono i valori in eeprom... Quando usi la eeprom il firmware carica quei valori e non quelli scritti nei define, puoi modificare i valori sempre con M666, oppure resettare i valori della eeprom con quelli messi nel firmware con il comando M502 e poi memorizzarli con M500

EnricoSx, 2017-08-31:
OTTIMO, Grazie infinite ,sei il migliore!
Ero convinto che caricare il firmware resettasse tutta la stampante.

closed 2017-08-31 by EnricoSx

MagoKimbra, 2017-08-31:
No la eeprom rimane, questo consente aggiornamenti del firmware senza dover cambiare i parametri settati...

EnricoSx, 2017-08-31:
ultima domandina ,spero :) da firmware come faccio girare l'estrusore nella direzione opposta?