MKFirmware/MK4duo #214: #define TOWER_C_ENDSTOP_ADJ , non ho capito

EnricoSx, 2017-09-02:
sempre io :)
sto cercando di imparare...
voglio calibrare il bed manualmente sulla delta, vado nel menu, scelgo calibra delta> calibra z, con la rotelliva vado giu fino a sfiorare il foglio, mi segna 0.4 .
So che posso regolare la vite sopra al carello, ma si va un pò ad occhio ed è una rottura, quindi vorrei imparare a farlo da firmare.
sono andato in configuration_overall
ho impostato #define TOWER_C_ENDSTOP_ADJ 0.4 
ho provato anche #define TOWER_C_ENDSTOP_ADJ -0.4

poi come mi hai insegnato, ho eseguito dopo il caricamento M502 e M500
riprovo, ma resta sempre uguale. E' forse il parametro sbagliato? grazie

MagoKimbra, 2017-09-02:
E magari fosse cosi facile... Cosi come stai facendo tu calibrare una sarebbe una passeggiata di salute...
Innanzitutto usa il comando M666 che modifichi i valori senza andare ogni volta ad aggiornare il firmware, ala massimo lo fai alla fine.. Ma lo devi fare più e più volte per ogni torre, per il centro e via dicendo finché non ottieni una regolazione giusta...

EnricoSx, 2017-09-02:
ahahah mi chiedevo perché tutti si lamentavano, se era cosi facile :D 
ma M666 e basta o poi anche M502 e M500?

MagoKimbra, 2017-09-02:
M666 con relativi sotto comandi. Non serve che memorizzi finche non spegni la stampante quelli restano... Quando finisci allora memorizzi e poi se vuoi metti anche nel firmware..

EnricoSx, 2017-09-02:
non trovo documentazione al riguardo :(

MagoKimbra, 2017-09-02:
M666 L hai la lista di tutti i sotto comandi...
In documentation quì in github c'è la lista di tutti i comandi e sotto comandi...

EnricoSx, 2017-09-02:
grazie .
in documentation ho trovato solo
M666 - Delta geometry adjustment.

EnricoSx, 2017-09-10:
Dubbio sull'uso di M666 X su una DELTA
inizio con
09:53:54.300 : Config:X (Endstop Adj): 0.000
09:53:54.300 : Config:Y (Endstop Adj): 0.000
09:53:54.300 : Config:Z (Endstop Adj): 0.000
dal display premo calibra delta->calibra X
vado giu fino al pezzo di carta e segna Z=2.8
eseguo M666 X-2.8 e poi G28
dal display premo calibra delta->calibra X
si impianta nel piatto perchè diventa Z=5.5
10:01:04.743 : Config:X (Endstop Adj): -2.800
10:01:28.976 : X:-103.92 Y:-60.00 Z:5.50 E:0.00 Count Alpha:21794 Beta:13276 Teta:13276

iosonopersia, 2017-10-21:
Questa issue può essere chiusa? Mi sembra di ricordare che il problema si fosse risolto nel gruppo Facebook...

MagoKimbra, 2017-10-21:
Yes!!

closed 2017-10-21 by MagoKimbra