MKFirmware/MK4duo #219: #error "LCD_BED_LEVELING requires MESH_BED_LEVELING or PROBE_MANUALLY

EnricoSx, 2017-09-04:
nel file configuration_overall ho scritto
```
#define PROBE_MANUALLY
#define LCD_BED_LEVELING
```

ma compilando mi da l'errore 
`#error "LCD_BED_LEVELING requires MESH_BED_LEVELING or PROBE_MANUALLY`

ps. c'è qualche documentazione su come si fa il livellamento con PROBE_MANUALLY e LCD?

MagoKimbra, 2017-09-04:
Quando esce quel segnale non è un errore di compilazione è un errore di configurazione...
MBL è già manuale quindi se abiliti lcd bed leveling con MBL non devi attivare probe_:manually.
Probe manually serve per ABL.
C'è un file che si chiama sanity check che controlla appunto gli errori di configurazione e ti dice che errore stai commettendo, basta seguire quello che ti dice.
LCD BED LEVELING richiede MESH_BED_LEVELING o PROBE_MANUALLY
Quindi o attivi MBL o ATTIVI PROBE_MANUALLY

EnricoSx, 2017-09-04:
interessante, non avevo capito che fosse gia attivo MBL, grazie.

closed 2017-09-04 by EnricoSx