MKFirmware/MK4duo #220: Online Configurator Feature Request

apphead, 2017-09-05:
Thanks for fixing the download error.
Since the Online Configurator is kind of a one-way street, I would much appreciate the option to just download the Configuration_Overall.h instead of the whole package.
One more minor thing: the github link at the bottom of the configurator points to your old repo.

Sidenote: It would serve the following workflow a lot, if one could upload the Configuration_Overall.h without the JSON, as I usually edit the config in notepad++ and not in configurator itself. So all changes I made are basically lost upon upload unless I change the settings in the JSON as well. Kind of cumbersome. I'm aware that this entails a lot of coding to make that change, but in terms of usability, configurator would enable to update an existing config with new variables easily - as opposed to being just a starting point for an initial config.

iosonopersia, 2017-09-06:
As for now the online configurator is supposed to be used also for modifying the Configuration_Overall.h file. You should keep the JSON section because for the configurator the remaining part of the file is useless: it can read only the JSON data. Editing the configurator in order to make possible what you ask would be such a mess to mantain... The online configurator it's just a tool, not the solution to every existing problem.

So you're supposed to upload to the configurator the Configuration_Overall.h and edit it there. The problem is that then you're constrained to download the entire firmware with it... So I totally agree with you: it should be possible to download also the Configuration_Overall.h file alone

apphead, 2017-09-06:
I totally understand your point. It surely entails a lot of work for my suggestion and would probably get messy too. Unfortunately, it does not help my workflow much that way.

closed 2017-09-06 by apphead