MKFirmware/MK4duo #221: bltouch clone deploys randomly

HyGy, 2017-09-05:
Hi,

I'm using mk4duo 4.3.2. I have a bltouch clone. I set it up as blotuch sensor (couse if not, then it leaves the tip out when moving the next calibration point).

Sometimes when measuring a calibration point it drops down the tip right after a successfull point metering. A you can see on the video at the last point:
https://goo.gl/photos/kXpebAWkLm2EgfB66

What i'm doing wrong?

Here is my config:

`/**
 * Configuration_Overall.h
 * Here you can define all your custom settings and they will overwrite configurations in the main configuration files.
 */

#define CONFIGURATION_OVERALL

/***********************
 * Configuration_Basic *
 ***********************/
#define SERIAL_PORT 0
#define BAUDRATE 115200
//#define BLUETOOTH
#define BLUETOOTH_PORT 0
#define BLUETOOTH_BAUD 115200
#define STRING_CONFIG_H_AUTHOR "(none, default config)"
#define MACHINE_UUID "00000000-0000-0000-0000-000000000000"
#define KILL_METHOD 0
#define NO_TIMEOUTS 1000
//#define ADVANCED_OK
//#define EMERGENCY_PARSER
//#define HOST_KEEPALIVE_FEATURE
#define DEFAULT_KEEPALIVE_INTERVAL 2
#define MOTHERBOARD BOARD_RAMPS_13_HFB
#define MECHANISM MECH_CARTESIAN
//#define MECHANISM MECH_COREXY
//#define MECHANISM MECH_COREYX
//#define MECHANISM MECH_COREXZ
//#define MECHANISM MECH_COREZX
//#define MECHANISM MECH_COREYZ
//#define MECHANISM MECH_COREZY
//#define MECHANISM MECH_DELTA
//#define MECHANISM MECH_MORGAN_SCARA
//#define MECHANISM MECH_MAKERARM_SCARA
//#define MECHANISM MECH_MUVE3D
#define POWER_SUPPLY 0
//#define PS_DEFAULT_OFF
#define DELAY_AFTER_POWER_ON 5
#define EXTRUDERS 1
#define DRIVER_EXTRUDERS 1

/*****************************
 * Configuration_Temperature *
 *****************************/
#define TEMPERATURE_UNITS_SUPPORT
#define TEMP_SENSOR_0 5
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_BED 1
#define TEMP_SENSOR_CHAMBER 0
#define TEMP_SENSOR_COOLER 0
#define TEMP_SENSOR_AD595_OFFSET 0
#define TEMP_SENSOR_AD595_GAIN 1
#define DUMMY_THERMISTOR_998_VALUE 25
#define DUMMY_THERMISTOR_999_VALUE 25
//#define SHOW_TEMP_ADC_VALUES
//#define MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED 0
//#define MILLISECONDS_PREHEAT_TIME 0
[... 1249 lines cut ...]
"deltaYprobeoffset": 0,
"deltaZprobeoffset": -10,
"E0retractionspeed": 150,
"E1retractionspeed": 150,
"E2retractionspeed": 150,
"E3retractionspeed": 150,
"Xinvert": 0,
"Yinvert": 0,
"Zinvert": 0,
"E0invert": 0,
"E1invert": 0,
"E2invert": 0,
"E3invert": 0,
"filamentswitch": "0",
"pausepin": 19,
"Xservo": "-1",
"Yservo": "-1",
"angleextendservosX": 0,
"angleretractservosX": 0,
"angleextendservosY": 0,
"angleretractservosY": 0,
"uilanguages": 7
}
========== End configuration string ==========
*/`

iosonopersia, 2017-09-06:
[image]

It's written in the Online Configurator: if you have a BLTouch clone, you must select Z Probe Servo!
Bye

HyGy, 2017-09-06:
At least I set to this parameter. But when using this, the tip is always down while moving, so I moved the head upper. But when it measures and start to move up the head it releases the pin before it is reaching the top position, so the tip is always knocking on the glass. Is there a parameter to change this?

Here is the video:
https://photos.app.goo.gl/jXByKxDSWeTsF7H13

weed2all, 2018-09-23:
Same thing here with bltouch clone set as bltouch...if i set it up as a z probe servo...when the time to autoleveling the bed the pin is always up...inverted in the firmware but behave the same...mk4duo 4.3.6