MKFirmware/MK4duo #222: Compilation error when BLTouch is enabled (and other probes as well)

apphead, 2017-09-05:
In file \src\gcode\debug\m43.h
I get compilation error, which is fixed by adding _probe._ to z_servo_angle where it is missing below:
[image]

apphead, 2017-09-05:
Also, if EEPROM_SETTINGS are enabled, I get errors. Only this time I cannot figure out, how to fix them...

`[]\src\eeprom\eeprom.cpp.o: In function `EEPROM::read_data(int&, unsigned char*, unsigned short, unsigned short*)':
[]\src\eeprom/eeprom.cpp:271: undefined reference to `eeprom_read_byte(unsigned char*)'
[]\src\eeprom\eeprom.cpp.o: In function `EEPROM::write_data(int&, unsigned char const*, unsigned short, unsigned short*)':
[]\src\eeprom/eeprom.cpp:248: undefined reference to `eeprom_read_byte(unsigned char*)'
[]\src\eeprom/eeprom.cpp:249: undefined reference to `eeprom_write_byte(unsigned char*, unsigned char)'
[]\src\eeprom/eeprom.cpp:250: undefined reference to `eeprom_read_byte(unsigned char*)'
`

MagoKimbra, 2017-09-05:
Fix servo..
I do not understand your problem with EEPROM

apphead, 2017-09-05:
Thx.
Title was misleading, issue is not related to BLTouch. What I meant about EEPROM_SETTINGS: what causes compilation with above errors to fail is setting EEPROM_SETTINGS to true.  
Tested with downloaded dev-firmware from online configurator.

iosonopersia, 2017-09-08:
Ok, this problem is caused by the fact that MK4duo believes RADDS has no eeprom in it. It can be easily fixed by adding this line somewhere in `src/boards/1401.h`:
```cpp
#define I2C_EEPROM
```
, as it's also done in Marlin.

apphead, 2017-09-08:
I see. RADDS (1.5) does indeed lack an EEPROM. Luckily, it's fairly easy and cheap to add one as described here: https://www.instructables.com/id/Adding-an-24LC256-EEPROM-to-the-Arduino-Due/
Compiling fine now, will need to test if EEPROM actually works (HW setup works with Repetier)

closed 2017-09-08 by MagoKimbra

apphead, 2017-09-08:
More compilation errors with RADDS:
`[]\src\lcd\buzzer.cpp.o: In function `WRITE':
[]\src\lcd/buzzer.cpp:27: multiple definition of `Buzzer::buzz(long, unsigned short)'
[]\src\buzzer\buzzer.cpp.o:[]\src\buzzer/buzzer.cpp:27: first defined here
collect2.exe: error: ld returned 1 exit status`

Set ORIG_BEEPER_PIN to -1, then its ok (I disabled LCD entirely)

iosonopersia, 2017-09-08:
Download again the firmware, Arduino IDE has messed up something, in src/lcd/ there is no buzzer.cpp file...

apphead, 2017-09-08:
Right, Arduino IDE was messed up somehow. Another source of potential errors....nice

commit 5ed130ab (2017-09-08): Merge pull request #230 from iosonopersia/master

Fix #222 and some warnings
commit 6aa70dbc (2017-09-08): Fix #222 and some warnings