MKFirmware/MK4duo #225: Temperature readings on RADDS faulty

apphead, 2017-09-06:
### Issue description
1 extruder and 1 hotbed. 
Temperatures for both just follow one sensor, completely ignoring the other one
[image]

The same hardware setup works fine with Repetier FW.

[Configuration_Overall.h.txt](https://github.com/MKFirmware/MK4duo/files/1282065/Configuration_Overall.h.txt)

iosonopersia, 2017-09-06:
How did you managed to compile the firmware if you previously told us that you got errors?

apphead, 2017-09-06:
Well, time and coffee did the trick ;-) I disabled EEPROM_SETTINGS and THERMAL_PROTECTION_HOTENDS, commented some parts of the code that caused trouble (min / max temp errors). 
After pulling your latest changes, THERMAL_PROTECTION_HOTENDS seems fine whereas EEPROM_SETTINGS enabled still won't compile as mentioned here https://github.com/MKFirmware/MK4duo/issues/222#issuecomment-327136729

apphead, 2017-09-06:
As for the faulty temps on RADDS, that seems to be fixed now. Much appreciated!

iosonopersia, 2017-09-08:
You said:
> I [...] commented some parts of the code that caused trouble (min / max temp errors).

There's a chance your problem with temperature readings is caused by your edits. Please undo them or download again MK4duo (**remember to wait until PR #230 is merged!**)

iosonopersia, 2017-09-08:
#230 was merged, please, retry now with the latest version of the firmware **without editing the code** and tell us if the problem still occurs.

iosonopersia, 2017-09-14:
@apphead any update? Does it work now?

apphead, 2017-09-20:
@iosonopersia I am lacking the time to check right now. Will report back as soon as I do.

apphead, 2017-10-20:
Finally got around to check up with V4.3.29 and temp reports are ok now. Thx for fixing

closed 2017-10-20 by apphead

iosonopersia, 2017-10-20:
That's great! Have a good time with MK4duo!