MKFirmware/MK4duo #232: Configurator issue on Z axis

carlonb, 2017-09-12:
Related to my discussion here [http://forums.reprap.org/read.php?361,787373](url) in order to add the Flux capacitor 4 ways I need 3 more extruder driver outside the controller (MKS Gen2Z V1.1) as this controller uses 2 drivers for Z axis (X, Y, Z, E1 for second Z, E0).
So discussing with Alberto we have defined the new 3 needed extruder as E2,E3,E4, then I used the configurator, then I've compiled correctly.
The strange issue in configurator is that if I want create a new identical configuration using as input the previous correct config_overall, the configurator do not allow to define the second Z axis as "Extruder 1". 
See in annexed pitcure, allows only to select Etruder 0 or 2.
See also my overall.
Thanks
[Configuration_Overall.zip](https://github.com/MKFirmware/MK4duo/files/1296431/Configuration_Overall.zip)

[image]

MagoKimbra, 2017-09-12:
Strano che non te lo faccia fare, comunque non serve attivarlo perché la tua scheda lo fa via hardware cioè i pin dir step enable di E1 sono stati presi in parallelo da Z. In teoria dovresti avere anche liberi i pin di E1, il problema è come prelevarli se non li hanno proprio portati dalla cpu...

carlonb, 2017-09-12:
Ciao Mago,
Spero di essermi spiegato bene, questo problema nasce solo se parto col configuratore inserendo l'overall precedentemente compilato con queste aggiunte dei 3 estrusori.
Se invece parto inserendo un overall che usa solo un estrusore (o da zero), allora il config si comporta correttamente e devo solo fare attenzione a selezionare tutto con cura.

Si è vero, il motore Z2 dovrebbe avere le piste sul c.st. in parallelo tra Z e E1, ed anch'io ho pensato la stessa cosa, ma i pin non sono portati su nessun connettore.
La scheda è montata ed è un casino verificare o hackerare, se mi girano le b.... metto la RAMPS.
Ciao

carlonb, 2017-09-12:
A proposito, sempre sul configuratore,
Ieri mi hai dato la dritta sull'errore di compilazione del FAN e mi hai suggerito di mettere "orig_fan0_pin" ed effettivamente ho risolto.
Ho capito perchè mi è successo questo, il configuratore, se parto inserendo un overall versione 4.3.24Dev, i campi del controller fan0 sono vuoti, mentre fan1,2 ecc. contengono valori di default.
Io li avevo lasciati vuoti....quindi errore...

closed 2018-08-02 by MagoKimbra