MKFirmware/MK4duo #239: Radds wont start with eeprom enabeled

smoki3, 2017-09-17:
my Radds board did not start with eeprom enabled

rafaljot, 2017-09-23:
same with Due + RAMPS-FD

iosonopersia, 2017-10-24:
Hi @smoki3 and @rafaljot , does this still happen to you? Have you tried again compiling? Eeprom code was changed recently, maybe now it works

rafaljot, 2017-11-01:
Hi @iosonopersia 
Now EPROM works with my Due + RAMPS-FD

lordwimsey, 2017-12-23:
If I may piggyback on this, My Due+Radds is flashed with MK4Due (latest version, but my half-year old version did the same)
It worked in the beginning, but after a while (after a few re-flashes/config try-outs) it stopped working.
M500 gives 
`Recv: Echo:Error writing to EEPROM!` (OctoPrint Serial Monitor)

M501:
`Send: M501
Recv: Echo:EEPROM version mismatch (EEPROM=?? MK4duo=MKV44)
Recv: Echo:Hardcoded Default Settings Loaded
Recv: Config:  G21 ; Units in mm
Recv: Config:  M149 C ; Units in Celsius
Recv: Config:Steps per unit:
Recv: Config:  M92 X80.000 Y80.000 Z1600.000 T0 E438.000
Recv: Config:Maximum feedrates (units/s):
Recv: Config:  M203 X300.000 Y300.000 Z10.000 T0 E75.000
Recv: Config:Maximum Acceleration (units/s2):
Recv: Config:  M201 X3000 Y3000 Z100 T0 E10000
Recv: Config:Acceleration (units/s2): P<print_accel> V<travel_accel> T* R<retract_accel>
Recv: Config:  M204 P1500.000 V3000.000 T0 R3000.000
Recv: Config:Advanced variables: S<min_feedrate> V<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> T* E<max_e_jerk>
Recv: Config:  M205 S0.000 V0.000 B20000 X10.000 Y10.000 Z0.400 T0 E5.000
Recv: Config:Fans:
Recv: Config:  M106 P0 F250 U9 L1 I0
Recv: Config:  M106 P1 F250 U8 L1 I0
Recv: Config:Material heatup parameters:
Recv: Config:  M145 S0 H190 B60 F255
Recv: Config:  M145 S1 H240 B100 F255
Recv: Config:  M145 S2 H230 B60 F255
Recv: Config:PID settings:
Recv: Config:  M301 H0 P4.543903 I0.294533 D17.525266
Recv: ok`

I googled the EEPROM Mismatch thing and found, that the eeprom is somehow indexed. With a mismatch (might happen when frequently re-flashing) it gets disabled.
I tried to clear it with a sketch, but failed until now. :-(

Could we get a "renovate EEPROM"-Feature?

closed 2018-08-02 by MagoKimbra

iosonopersia, 2018-08-02:
@lordwimsey , you just have to overwrite eeprom every time you update the firmware. Just send M502 and then M500.

lordwimsey, 2018-08-04:
Thanks for your comment!
Unfortunately, M500 doesn't work, even after flashing new firmware. I just put some inits in the start g-code of the slicer. This works OK for me.
(Right now, I am running the Marlin 2.0 FW)

MagoKimbra, 2018-08-04:
Sorry, but i have radds and eeprom function perfectly. But i don't have extension board. Do you have extension board on radds?

lordwimsey, 2018-08-04:
I don't have an extension board, just the standard RADDS. 
However, I hooked up an OLED Display, which I understand is also communicating via I2C. (I'm not very smart with this...)
I tried disconnecting it though, no success.

Please mind, this error also occurs on Marlin 2, which I am running at the moment. That's why I am certain, the error is either in the hardware or somewhere in my configuration. (I did the config from scratch for both firmwares, but they corresponded.)

This is the output from Marlin2, from a minute ago.

`Recv: FIRMWARE_NAME:Marlin bugfix-2.0.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Voron CoreXY EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:0
Recv: Cap:Z_PROBE:0
Recv: Cap:LEVELING_DATA:0
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: ok
Send: M21
Recv: echo:SD init fail
Recv: ok
Send: M155 S2
Recv: echo:SD init fail
Recv: ok
[...]
Send: M502
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok
[...]
Send: M500
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: start
Printer sent 'start' while already operational. External reset? Resetting line numbers to be on the safe side
Recv: echo: Watchdog Reset
Recv: Marlin bugfix-2.0.x
Recv: 
Recv: echo: Last Updated: 2018-01-20 | Author: (lordwimsey)
Recv: echo:Compiled: Jun  6 2018
Recv: echo: Free Memory: 84176  PlannerBufferBytes: 1344
Recv: echo:EEPROM version mismatch (EEPROM=�> Marlin=V50)
Recv: echo:Hardcoded Default Settings Loaded
Recv: echo:SD init fail`

Sanitar777, 2018-08-05:
> Recv: Marlin bugfix-2.0.x

> Recv: echo:EEPROM version mismatch (EEPROM=�> Marlin=V50) 

This is not the MK4due firmware

lordwimsey, 2018-08-05:
> This is not the MK4due firmware.

Yes. 
I pointed that out already, apologies if I have been unclear.  Please note:

> Please mind, this error also occurs on Marlin 2, which I am running at the moment. That's why I am certain, the error is either in the hardware or somewhere in my configuration. (I did the config from scratch for both firmwares, but they corresponded.)