MKFirmware/MK4duo #241: bug? std dev:-2147483.750

EnricoSx, 2017-09-20:
credo che un bug di Marlin si rifletta su MK4DUO, ve lo segnalo
https://github.com/MarlinFirmware/Marlin/issues/7402

Eseguo G33 con
https://www.thingiverse.com/thing:1976680

e mi risulta 
**std dev:-2147483.750**
Anycubic Delta Linear Plus
[G33.txt](https://github.com/MKFirmware/MK4duo/files/1318797/G33.txt)

MagoKimbra, 2017-09-20:
Qual'è il bug?

MagoKimbra, 2017-09-20:
Si se non riesce a prendere un punto perché fuori piatto, per via dell'offset..
Ma con MK puoi cambiare il probe radius sempre con m600 in modo che stia sempre dentro.
Se salta qualche punto lo riduci.. Ma quella probe che hai messo non si mette al posto del nozzle proprio per avere offset = 0 almeno per X e Y, perché è quello il grosso problema delle sonde che dovrebbero stare nello stesso punto del nozzle. Se hai un offset allora la trovo inutile...

EnricoSx, 2017-09-21:
Grazie, ho capito :)