MKFirmware/MK4duo #245: Smart Ramps  & EEPROM Compile error

Drakelive, 2017-09-25:
Hi 

On Smart Ramps ( board 1411 ) I get some errors compiling, when EEPROM is enabled.

/home/drake/Sviluppo/workspaces/Marlin_test/MK_32_2/Release/../src/eeprom/eeprom.cpp:284: undefined reference to `eeprom_read_byte(unsigned char*)'
./src/eeprom/eeprom.cpp.o: In function `EEPROM::write_data(int&, unsigned char const*, unsigned short, unsigned short*)':
/home/drake/Sviluppo/workspaces/Marlin_test/MK_32_2/Release/../src/eeprom/eeprom.cpp:261: undefined reference to `eeprom_read_byte(unsigned char*)'
makefile:122: set di istruzioni per l'obiettivo "MK_32_2.elf" non riuscito
/home/drake/Sviluppo/workspaces/Marlin_test/MK_32_2/Release/../src/eeprom/eeprom.cpp:262: undefined reference to `eeprom_write_byte(unsigned char*, unsigned char)'
/home/drake/Sviluppo/workspaces/Marlin_test/MK_32_2/Release/../src/eeprom/eeprom.cpp:263: undefined reference to `eeprom_read_byte(unsigned char*)'
collect2: error: ld returned 1 exit status
make: *** [MK_32_2.elf] Errore 1

alejoh90, 2017-09-28:
me2

iosonopersia, 2017-10-22:
Hi everyone. Which version of the firmware are you referring to? Could you please give us some feedback from the latest version available (master branch, V4.3.29 dev), maybe this error was already fixed and this issue could be closed. Thanks.

jmlcatarino, 2017-12-05:
@iosonopersia 
I'm trying to use the last master branch, which was downloaded a few minutes go and I'm having a similar compile error:
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_406026/sketch/src/eeprom/eeprom.cpp.o: In function `EEPROM::read_data(int&, unsigned char*, unsigned short, unsigned short*)':
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_406026/sketch/src/eeprom/eeprom.cpp:320: undefined reference to `eeprom_read_byte(unsigned char*)'
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_406026/sketch/src/eeprom/eeprom.cpp.o: In function `EEPROM::write_data(int&, unsigned char const*, unsigned short, unsigned short*)':
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_406026/sketch/src/eeprom/eeprom.cpp:299: undefined reference to `eeprom_read_byte(unsigned char*)'
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_406026/sketch/src/eeprom/eeprom.cpp:300: undefined reference to `eeprom_write_byte(unsigned char*, unsigned char)'
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_406026/sketch/src/eeprom/eeprom.cpp:301: undefined reference to `eeprom_read_byte(unsigned char*)'
collect2: error: ld returned 1 exit status

I'm trying to use an external EEPROM and already included #include <extEEPROM.h>. 

Thanks in advance.

MagoKimbra, 2017-12-05:
Smart ramps have the EEPROM??

jmlcatarino, 2017-12-05:
@MagoKimbra no, as I mentioned I'm using an external I2C EEPROM.

MagoKimbra, 2017-12-05:
Ok It's your modification...
You must add #define I2C_EEPROM in file 1411.h or the one you use

jmlcatarino, 2017-12-05:
It worked! No more compile errors. Tomorrow I'll upload it and test it.
So this means I have to add the #define I2C_EEPROM in all the boards I pretend to use the external eeprom?

MagoKimbra, 2017-12-05:
Yes, unfortunately for the cards that do not have the default eeprom I can not know if you mount an eeprom and what kind (spi or i2c). Others use the eeprom on SD.

jmlcatarino, 2017-12-05:
Ok I see, you have a point. Thank you for your help, as usual. :) 
Now that this one is solved, back to a new problem, but this one is more hardware than firmware. Need to get a full graphic LCD to work with this board. The pins are there, the backlight is on but now graphics. :)

closed 2018-08-02 by MagoKimbra