MKFirmware/MK4duo #246: Compiling Error with MAX31855

Drakelive, 2017-09-25:
Hello 
I Have some problem when I compiling MK4Duo  on Ramps 1.4 ( 8 Bit ) and Smart Ramps ( 32 Bit ) when enabled  MAX31855 (  -3  : thermocouple controller ).

Ramps 1.4
--------------------------
### ../src/sanitycheck.h:1642:6: error: #error DEPENDENCY ERROR: HEATER_0_PIN not EXIST for this board
     #error DEPENDENCY ERROR: HEATER_0_PIN not EXIST for this board
      ^
In file included from ../base.h:115:0,
                 from ../.ino.cpp:9:
../src/temperature/temperature.h: In static member function 'static bool Temperature::tooColdToExtrude(uint8_t)':
../src/temperature/temperature.h:73:45: error: 'heaters' was not declared in this scope
         return allow_cold_extrude ? false : heaters[HOTEND_INDEX].current_temperature < extrude_min_temp;

Smart Ramps
---------------------------------------------

### ../src/sanitycheck.h:1670:6: error: #error DEPENDENCY ERROR: HEATER_0_PIN not EXIST for this board
     #error DEPENDENCY ERROR: HEATER_0_PIN not EXIST for this board
      ^
In file included from ../base.h:112:0,
                 from ../.ino.cpp:8:
../src/temperature/temperature.h: In static member function 'static bool Temperature::tooColdToExtrude(uint8_t)':
../src/temperature/temperature.h:73:45: error: 'heaters' was not declared in this scope
         return allow_cold_extrude ? false : heaters[HOTEND_INDEX].current_temperature < extrude_min_temp;

MagoKimbra, 2017-09-25:
Ok . i found the problem.. i think i can fix it in the day. Thank's

MagoKimbra, 2017-09-25:
Test it now... Choice for Temp pin your MAX31855 pin.

Drakelive, 2017-09-25:
We have other problems.  
**With MAX6675  ( -2 ) :** 

../src/temperature/temperature.cpp: In static member function 'static int16_t Temperature::read_max6675(Pin, int8_t)':
../src/temperature/temperature.cpp:1007:55: error: 'idx' was not declared in this scope
     if (HAL::timeInMilliseconds() - last_max6675_read[idx] > 230) {

**With MAX31855  ( -3 ) :** 
../src/temperature/temperature.h: In static member function 'static bool Temperature::tooColdToExtrude(uint8_t)':
../src/temperature/temperature.h:73:45: error: 'heaters' was not declared in this scope
         return allow_cold_extrude ? false : heaters[HOTEND_INDEX].current_temperature < extrude_min_temp;

closed 2018-08-06 by iosonopersia