MKFirmware/MK4duo #253: Few questions about MK4duo

vido89, 2017-10-07:
Where I can get MK4Duo board and where I can see all features of what it has ?
And last i assume that this board can work on [this](https://www.aliexpress.com/item/Free-shipping-5pcs-lot-LilyPad-328-Main-Board-ATmega328P-ATmega328-16M-For-Arduino/1701138585.html?spm=2114.search0104.3.1.TK5mMI&ws_ab_test=searchweb0_0,searchweb201602_3_10152_10065_10151_10068_10344_10342_10343_10340_10341_5470011_10307_10137_10060_10155_10154_10056_10055_10054_10059_100031_10099_5460020_10338_10339_10103_10102_440_10052_10053_10142_10107_10050_10051_5380020_10326_10084_10083_10080_10082_10081_10110_10111_10112_10113_10114_10179_143_10312_10313_10314_10078_10079_10073,searchweb201603_2,ppcSwitch_5&btsid=9ef74a67-bd69-4d11-8908-29b2cef41949&algo_expid=1edce67e-03ee-4885-8921-1195aa0db951-0&algo_pvid=1edce67e-03ee-4885-8921-1195aa0db951)

iosonopersia, 2017-10-07:
Hi, MK4duo is not a board, it's a firmware. You can configure and download it here: http://marlinkimbra.it/configurator/v4_3_2/firmware_configurator.php?lingua=en

There are a lot of boards you can use. Some have the microcontroller built-in, others are only shields and you need also an Arduino (Mega 2560 is 8bit, Due is 32bit but it's more expensive, 32 bit shields are more expensive too). 32bit is faster and more powerful than 8bit, anyway.

If you don't know where to start from, I suggest using an Arduino Mega 2560 with a RAMPS 1.4 shield. It's very cheap and quite reliable.

vido89, 2017-10-07:
Thanks for explanation I was thinking that MK4duo is shield for Due R3 Board not firmware Im looking for good solution to go to 32 bit board since Im using delta tnx

iosonopersia, 2017-10-22:
@MagoKimbra, this can be closed.

iosonopersia, 2018-08-02:
This issue was solved on 7 Oct 2017. It can be safely closed, @MagoKimbra

closed 2018-08-02 by MagoKimbra