MKFirmware/MK4duo #254: Inverted FAN PWM and unuseable case light switch

sarpba, 2017-10-08:
Firmware version dev 4.3.28. 
(In 4.3.2 evrithing worked fine.)

[Configuration_Overall.zip](https://github.com/MKFirmware/MK4duo/files/1366725/Configuration_Overall.zip)

My setup:
Ramps-FD v1 + Arduino DUE

used pins: 
digital pin2 for fan0
digital pin12 for hotend cooler
digital pin11 for case light

Something wrong with FAN PWM it's work, but in inverted way. If I set 255 then stoped and at 0 it's working with full speed.

The hotend cooler is working with ful speed if tte hotend is cool, but if warming up then stoped.

The case light is turned on, when power on the printer, and can't power off.

BTW a little feature request: Can you make higher PWM speed for hotend cooler too?

Best regards,
sarpba

Hans007a, 2017-10-15:
look at here for the solution #261

rafaljot, 2017-11-02:
hmm. it looks like Thera are more versions of RAMPS-FD .... 
v1.x - heaters inverted, fans not inverted
v1.y - heaters and fan inverted,
v2 - heater and fans not inverted 

:/

what exactly RAMPS do you have? Where did you buy it?

sarpba, 2017-11-02:
I have v1.something... 

anyway it work great with 3.2 firmware and ramps-FD V1 board setting.

closed 2017-12-10 by sarpba