MKFirmware/MK4duo #258: ultratronics 1.0 pro & I2C eeprom

silversat, 2017-10-12:
### Issue description

connecting an external i2c eeprom, or others devices,  to I2C connector on this board doesn't work at all. these pins are not connected to the first i2c port (sda0, scl0) but to the second one (sda1, scl1) and is referenced as wire1. you should mod I2cEeprom.cpp to switch correctly wire/wire1...

sda0/scl0 are wired to ANALOG connector.

thanks, bye

remspoor, 2018-08-06:
It would be really great if there was an option to select the i2c bus.

My setup is a DIY CoreXY printer with an Ultratronics 1.0 (an adruino due). And as metioned above, the connections for a usable i2c connection, are on the second bus. I connected an i2c FRAM chip (FM24CL64) which is compatible with eeprom to that bus. In the code i use the patch below, and i works like a charm.

```diff --git a/MK4duo/src/HAL/I2cEeprom.cpp b/MK4duo/src/HAL/I2cEeprom.cpp
index 47bf717e..49c62e1d 100644
--- a/MK4duo/src/HAL/I2cEeprom.cpp
+++ b/MK4duo/src/HAL/I2cEeprom.cpp
@@ -77,7 +77,7 @@ static uint8_t eeprom_device_address = 0x50;
 static void eeprom_init(void) {
   static bool eeprom_initialised = false;
   if (!eeprom_initialised) {
-    Wire.begin();
+    Wire1.begin();
     eeprom_initialised = true;
   }
 }
@@ -87,11 +87,11 @@ void eeprom_write_byte(uint8_t* pos, uint8_t value) {
 
   eeprom_init();
 
-  Wire.beginTransmission(eeprom_device_address);
-  Wire.write((int)(eeprom_address >> 8));   // MSB
-  Wire.write((int)(eeprom_address & 0xFF)); // LSB
-  Wire.write(value);
-  Wire.endTransmission();
+  Wire1.beginTransmission(eeprom_device_address);
+  Wire1.write((int)(eeprom_address >> 8));   // MSB
+  Wire1.write((int)(eeprom_address & 0xFF)); // LSB
+  Wire1.write(value);
+  Wire1.endTransmission();
 
   // wait for write cycle to complete
   // this could be done more efficiently with "acknowledge polling"
@@ -104,23 +104,23 @@ void eeprom_update_block(const void* pos, void* eeprom_address, size_t n) {
 
   eeprom_init();
 
-  Wire.beginTransmission(eeprom_device_address);
-  Wire.write((int)((unsigned)eeprom_address >> 8));   // MSB
-  Wire.write((int)((unsigned)eeprom_address & 0xFF)); // LSB
-  Wire.endTransmission();
+  Wire1.beginTransmission(eeprom_device_address);
+  Wire1.write((int)((unsigned)eeprom_address >> 8));   // MSB
+  Wire1.write((int)((unsigned)eeprom_address & 0xFF)); // LSB
+  Wire1.endTransmission();
 
   uint8_t *ptr = (uint8_t*)pos;
   uint8_t flag = 0;
-  Wire.requestFrom(eeprom_device_address, (byte)n);
-  for (byte c = 0; c < n && Wire.available(); c++)
-    flag |= Wire.read() ^ ptr[c];
+  Wire1.requestFrom(eeprom_device_address, (byte)n);
+  for (byte c = 0; c < n && Wire1.available(); c++)
+    flag |= Wire1.read() ^ ptr[c];
 
   if (flag) {
-    Wire.beginTransmission(eeprom_device_address);
-    Wire.write((int)((unsigned)eeprom_address >> 8));   // MSB
-    Wire.write((int)((unsigned)eeprom_address & 0xFF)); // LSB
-    Wire.write((uint8_t*)(pos), n);
-    Wire.endTransmission();
+    Wire1.beginTransmission(eeprom_device_address);
+    Wire1.write((int)((unsigned)eeprom_address >> 8));   // MSB
+    Wire1.write((int)((unsigned)eeprom_address & 0xFF)); // LSB
+    Wire1.write((uint8_t*)(pos), n);
+    Wire1.endTransmission();
 
     // wait for write cycle to complete
     // this could be done more efficiently with "acknowledge polling"
@@ -134,25 +134,25 @@ uint8_t eeprom_read_byte(uint8_t* pos) {
 
   eeprom_init ();
 
-  Wire.beginTransmission(eeprom_device_address);
-  Wire.write((int)(eeprom_address >> 8));   // MSB
-  Wire.write((int)(eeprom_address & 0xFF)); // LSB
-  Wire.endTransmission();
-  Wire.requestFrom(eeprom_device_address, (byte)1);
-  return Wire.available() ? Wire.read() : 0xFF;
+  Wire1.beginTransmission(eeprom_device_address);
+  Wire1.write((int)(eeprom_address >> 8));   // MSB
+  Wire1.write((int)(eeprom_address & 0xFF)); // LSB
+  Wire1.endTransmission();
+  Wire1.requestFrom(eeprom_device_address, (byte)1);
+  return Wire1.available() ? Wire1.read() : 0xFF;
 }
 
 // maybe let's not read more than 30 or 32 bytes at a time!
 void eeprom_read_block(void* pos, const void* eeprom_address, size_t n) {
   eeprom_init();
 
-  Wire.beginTransmission(eeprom_device_address);
-  Wire.write((int)((unsigned)eeprom_address >> 8));   // MSB
-  Wire.write((int)((unsigned)eeprom_address & 0xFF)); // LSB
-  Wire.endTransmission();
-  Wire.requestFrom(eeprom_device_address, (byte)n);
+  Wire1.beginTransmission(eeprom_device_address);
+  Wire1.write((int)((unsigned)eeprom_address >> 8));   // MSB
+  Wire1.write((int)((unsigned)eeprom_address & 0xFF)); // LSB
+  Wire1.endTransmission();
+  Wire1.requestFrom(eeprom_device_address, (byte)n);
   for (byte c = 0; c < n; c++ )
-    if (Wire.available()) *((uint8_t*)pos + c) = Wire.read();
+    if (Wire1.available()) *((uint8_t*)pos + c) = Wire1.read();
 }
 
 #endif // HAS_EEPROM_I2C
diff --git a/MK4duo/src/boards/1705.h b/MK4duo/src/boards/1705.h
index ded9a10d..4d3ea49b 100644
--- a/MK4duo/src/boards/1705.h
+++ b/MK4duo/src/boards/1705.h
@@ -160,7 +160,9 @@
 
 
 //###UNKNOWN_PINS
-#define EEPROM_SD
+//#define EEPROM_SD
+#define EEPROM_I2C
+#define E2END 0x1FFF
 #define MAX6675_SS_PIN    65
 #define MAX31855_SS0_PIN  65
 #define MAX31855_SS1_PIN  52```diff

iosonopersia, 2018-08-06:
Thanks a lot @remspoor for your suggestion. Tomorrow I'm going to add the option for selecting which port will be used by EEPROM. Let's see how it will go.

iosonopersia, 2018-08-06:
I had a try with #549. Please review that Pull Request to give us some suggestions! @remspoor @silversat

silversat, 2018-08-06:
I resolved this simple way
[I2cEeprom.zip](https://github.com/MKFirmware/MK4duo/files/2264079/I2cEeprom.zip)

iosonopersia, 2018-08-07:
Hi @silversat and @remspoor. I need you to check the changes made here: #550 . If you give me the OK, I'm going to merge them in the master branch. Let me know!

silversat, 2018-08-07:
hi, cannot try now but it seems to be ok. I used that solution for a long time on ultratronics with i2c eeprom connected.

what about #define E2END 0x1FFF ? eeprom max size?

iosonopersia, 2018-08-07:
Those changes have been merged. I'm closing this issue since it's now solved. I'm going to try understand how E2END macro works exactly and then do something for it.

closed 2018-08-07 by iosonopersia

silversat, 2018-08-07:
ok, thanks

remspoor, 2018-08-07:
Looks great!
Maybe it's an idea for the future to have the web configurator set EEPROM_SD or EEPROM_I2C when a DUE board is selected. But for now, this works out of the box for us ;)
Thanks for this merge!!
You made at least 2 guys happy with this..