MKFirmware/MK4duo #259: no Scara configuration works

Aghoma, 2017-10-12:
(French) : I would like to know when you intend to upgrade a MAKERARM_SCARA version online? I only have compilation problems as soon as I choose the MECH_Scara.

Also, I can not get firmware Scara MakerARM via the configurator " http://marlinkimbra.it " can import the version.

If there is no possibility of having a functional version, I should turn to (Repetier - https://github.com/wangsamas) that I do not like the interface at all ...

**ERROR Compil :  configurator 4_3_x / 4_3_2** 

In file included from sketch\base.h:84:0,
                 from C:\Users\admhc\Desktop\MK4duo_12-10-2017_150042\MK4duo\MK4duo.ino:247:
sketch\src/sanitycheck.h:1283:4: error: #error DEPENDENCY ERROR: Missing setting Z3_MIN_ENDSTOP_LOGIC
   #error DEPENDENCY ERROR: Missing setting Z3_MIN_ENDSTOP_LOGIC
    ^
sketch\src/sanitycheck.h:1286:4: error: #error DEPENDENCY ERROR: Missing setting Z4_MIN_ENDSTOP_LOGIC
   #error DEPENDENCY ERROR: Missing setting Z4_MIN_ENDSTOP_LOGIC
    ^
sketch\src/sanitycheck.h:1301:4: error: #error DEPENDENCY ERROR: Missing setting Z3_MAX_ENDSTOP_LOGIC
   #error DEPENDENCY ERROR: Missing setting Z3_MAX_ENDSTOP_LOGIC
    ^
sketch\src/sanitycheck.h:1304:4: error: #error DEPENDENCY ERROR: Missing setting Z4_MAX_ENDSTOP_LOGIC
   #error DEPENDENCY ERROR: Missing setting Z4_MAX_ENDSTOP_LOGIC
    ^
sketch\src/sanitycheck.h:1471:6: error: #error DEPENDENCY ERROR: Missing setting LINKAGE_1
     #error DEPENDENCY ERROR: Missing setting LINKAGE_1
      ^
sketch\src/sanitycheck.h:1474:6: error: #error DEPENDENCY ERROR: Missing setting LINKAGE_2
     #error DEPENDENCY ERROR: Missing setting LINKAGE_2
     ^
sketch\src/sanitycheck.h:1483:6: error: #error DEPENDENCY ERROR: Missing setting SCARA_RAD2DEG
     #error DEPENDENCY ERROR: Missing setting SCARA_RAD2DEG
      ^
exit status 1
Erreur de compilation pour la carte Arduino/Genuino Mega or Mega 2560

thanks in advance

iosonopersia, 2017-10-14:
Hi, you won't get those errors if you choose to compile the latest dev version (4.3.29). Sadly, we've not implemented a full support for the SCARA mechanics, so even compiling the dev version won't work... I'm afraid you'll have to wait for the full support. Actually SCARA users are not so many and we cannot get a lot of feedback...

Agohma, 2017-10-18:
Thanks for your feedback ; I will wait for a functional version for scara makerARM
cordially

Power3DPrinting, 2018-04-08:
How can I help with giving back feed back? I've been hunting down a firmware that can handle SCARA kinematics but haven't had much luck.

iosonopersia, 2018-08-11:
Hi @Power3DPrinting your help would be really appreciated!!! What type of SCARA do you own? A MORGAN_SCARA or a MAKERARM_SCARA?

Actually I've now made a big step forward in terms of SCARA support. For years MK4duo did not compile under SCARA option enabled since a lot of functions were missing in our codebase, and some were not correct. Now for the first time since I've started working for MK4duo, **I've seen a MORGAN_SCARA configuration compile well in Arduino IDE**. I still don't know if it will work in real world tests (I'm pretty sure it won't work 100% well) but it's still a BIG step forward.

This is why I'm asking you if you own a MORGAN_SCARA, since MAKERARM_SCARA still misses 3 fundamental functions:
```cpp
bool move_to_cal(uint8_t delta_a, uint8_t delta_b);
void Transform(const float &a, const float &b);
void InverseTransform(const float logical[XYZ]);
```

**Please, I would be really grateful if you could test [this branch here](https://github.com/iosonopersia/MK4duo/tree/scaraDevelopment)!**

Power3DPrinting, 2018-08-11:
Hey, thanks for the heads up about Parallel/Morgan SCARA’s, I unfortunatly
have a Single arm Serial SCARA what a lot of people call makerarm SCARA’s
where one motor is stacked upon the other, Wangsamas firmware might have
the missing 3 or so lines you’re looking for as it’s the only 8-bit
firmware I’ve had some luck with for the Serial/makerarm style machine I
have.

Thanks and sorry I can’t help test it! I’m still looking for a better
firmware for my machine so give another heads up if I can help test some
code!

iosonopersia, 2018-08-11:
@Power3DPrinting , @MagoKimbra just made it compile also for MAKERARM printers!!! Please tell us how it behaves! If needed, make videos and/or describe carefully everything that needs to be fixed. That would be very important for MK4duo!
Thanks in advance!
Just download the newest master branch of MK4duo!

closed 2018-08-26 by iosonopersia

Power3DPrinting, 2018-08-29:
Hi, got my hands full between school and work but I finally uploaded the latest version of Mk4Duo to test the changes. Where can I share pictures or videos to show how my machine works? It uploaded successfully which is fantastic and the Z-axis works but I cannot get the X or Y (Shoulder or Elbow) moving.

iosonopersia, 2018-08-30:
Hi @Power3DPrinting ! Please, firstly open a new issue. Then you can share photos just by drag and dropping them in the comment box (if I'm not wrong). As for videos, you could upload them on YouTube or on a cloud storage like Dropbox and provide us a link to view them. In the issue try to explain with as many details as you can what exactly is the problem, how and when it happens, in what circumstances, and so on... Thanks in advance for the help you're giving us!

Power3DPrinting, 2018-08-30:
Thanks! I'll go do that, I really appreciate the quick response

ricardocjcosta, 2019-01-26:
Hi!
I would also like to try to implement this firmware on a single scara using Arduino Mega2560 / Ramps1.4.
It is still under construction. However, when trying to compile, I get the following list of errors. I do not have many programming skills. I tried to figure out how I could solve it but I could not.
I saw that Power3DPrinting was able to do the compilation and load it to the edge, even though it got errors similar to these.
Is it possible to have an explanation of how you did it?
Thank you

sketch\src\core\mechanics\scara_mechanics.cpp: In static member function 'static void Scara_Mechanics::factory_parameters()':

sketch\src\core\mechanics\scara_mechanics.cpp:76:32: error: assignment of read-only location 'Mechanics::homing_feedrate_mm_s[((int)i)]'

       homing_feedrate_mm_s[i]  = pgm_read_float(&tmp_homefeedrate[i]);

                                ^

sketch\src\core\mechanics\scara_mechanics.cpp: In static member function 'static void Scara_Mechanics::do_homing_move(AxisEnum, float, float)':

sketch\src\core\mechanics\scara_mechanics.cpp:532:36: error: 'struct mechanics_data_t' has no member named 'home_dir'

     const bool is_home_dir = (data.home_dir[axis] > 0) == (distance > 0);

                                    ^

sketch\src\core\mechanics\scara_mechanics.cpp:548:38: error: 'sync_plan_position_mech_specific' was not declared in this scope

     sync_plan_position_mech_specific();

                                      ^

sketch\src\core\mechanics\scara_mechanics.cpp: In static member function 'static void Scara_Mechanics::homeaxis(AxisEnum)':

sketch\src\core\mechanics\scara_mechanics.cpp:1065:67: error: 'struct mechanics_data_t' has no member named 'home_dir'

     mechanics.do_homing_move(axis, 1.5f * max_length[axis] * data.home_dir[axis]);

                                                                   ^

sketch\src\core\mechanics\scara_mechanics.cpp:1068:29: error: 'struct mechanics_data_t' has no member named 'home_dir'

     const float bump = data.home_dir[axis] * (

                             ^

sketch\src\core\mechanics\scara_mechanics.cpp:1091:38: error: 'sync_plan_position_mech_specific' was not declared in this scope

     sync_plan_position_mech_specific();

                                      ^

exit status 1
In file included from sketch\src\core\commands\gcode/gcode.h:48:0,

                 from sketch\src\core\commands\commands.cpp:30:

sketch\src\core\commands\gcode/config/m92.h: In function 'void gcode_M92()':

sketch\src\core\commands\gcode/config/m92.h:51:17: error: 'class Scara_Mechanics' has no member named 'print_M92'

       mechanics.print_M92();

                 ^

In file included from sketch\src\core\commands\gcode/gcode.h:51:0,

                 from sketch\src\core\commands\commands.cpp:30:

sketch\src\core\commands\gcode/config/m201.h: In function 'void gcode_M201()':

sketch\src\core\commands\gcode/config/m201.h:43:17: error: 'class Scara_Mechanics' has no member named 'print_M201'

       mechanics.print_M201();

                 ^

In file included from sketch\src\core\commands\gcode/gcode.h:52:0,

                 from sketch\src\core\commands\commands.cpp:30:

sketch\src\core\commands\gcode/config/m203.h: In function 'void gcode_M203()':

sketch\src\core\commands\gcode/config/m203.h:43:17: error: 'class Scara_Mechanics' has no member named 'print_M203'

       mechanics.print_M203();

                 ^

In file included from sketch\src\core\commands\gcode/gcode.h:53:0,

                 from sketch\src\core\commands\commands.cpp:30:

sketch\src\core\commands\gcode/config/m204.h: In function 'void gcode_M204()':

sketch\src\core\commands\gcode/config/m204.h:47:17: error: 'class Scara_Mechanics' has no member named 'print_M204'

       mechanics.print_M204();

                 ^

In file included from sketch\src\core\commands\gcode/gcode.h:54:0,

                 from sketch\src\core\commands\commands.cpp:30:

sketch\src\core\commands\gcode/config/m205.h: In function 'void gcode_M205()':

sketch\src\core\commands\gcode/config/m205.h:50:17: error: 'class Scara_Mechanics' has no member named 'print_M205'

       mechanics.print_M205();

                 ^

MagoKimbra, 2019-01-26:
Fix this problem...

ricardocjcosta, 2019-01-26:
Thank you very much for the very quick answer and solution. The firmware compiles successfully choosing the MECHANISM MECH_MAKERARM_SCARA definition.
I can not get an answer from the engines when I use the Repetier Host, however I can make the move, very well, using Pronterface and that's pretty good! ;)
I think my robot structure will be finished within 3 weeks or a month. As soon as I have it, I'll test it with your firmware and give you feedback!
Thank you for your help once more!