MKFirmware/MK4duo #260: Ultratronics & reprapworld graphical lcd #2

silversat, 2017-10-13:
#### Compile error

compiling for ULTRATRONICS board with REPRAPWORLD_GRAPHICAL_LCD you get the follow error:

````

In file included from sketch\base.h:102:0,

                 from F:\Downloads\3D_Printing\Firmware\Ultratronics\MK4duo\MK4duo.ino:264:

sketch\src/planner/planner.h: In static member function 'static uint16_t Planner::block_buffer_runtime()':

sketch\src/planner/planner.h:297:11: error: expected ';' before 'millis_t'

           millis_t bbru = block_buffer_runtime_us;

           ^

sketch\src/planner/planner.h:302:9: error: expected ';' before 'bbru'

         bbru >>= 10;

         ^

In file included from sketch\base.h:41:0,

                 from F:\Downloads\3D_Printing\Firmware\Ultratronics\MK4duo\MK4duo.ino:264:

sketch\src/planner/planner.h:304:16: error: 'bbru' was not declared in this scope

         NOMORE(bbru, 0xFFFFul);

                ^

sketch\src/macros.h:116:27: note: in definition of macro 'NOMORE'

 #define NOMORE(v,n)       v = (v > n) ? n : v

                           ^

In file included from sketch\base.h:102:0,

                 from F:\Downloads\3D_Printing\Firmware\Ultratronics\MK4duo\MK4duo.ino:264:

sketch\src/planner/planner.h: In static member function 'static void Planner::clear_block_buffer_runtime()':

sketch\src/planner/planner.h:310:11: error: expected ';' before 'block_buffer_runtime_us'

           block_buffer_runtime_us = 0;

           ^

sketch\src/planner/planner.h:312:7: error: expected ';' before '}' token

       }

       ^

exit status 1
Errore durante la compilazione per la scheda Arduino Due (Programming Port).
````

lost last semicolon in HAL_due.h, lines 149, 150 
````
#define CRITICAL_SECTION_START	uint32_t primask=__get_PRIMASK(); __disable_irq()
#define CRITICAL_SECTION_END    if (!primask) __enable_irq()
````

MagoKimbra, 2017-10-13:
Please send me your configuration_overall.h
I try with Ultratronics and display and haven't error...

MagoKimbra, 2017-10-13:
Ok i found the problem... I fix it...

closed 2017-10-13 by MagoKimbra