MKFirmware/MK4duo #261: Solution for Due + RAMPS_FD v1 fan inverting

Hans007a, 2017-10-14:
Hello!

I found a solution for DUE + RAMPS-FD v1 fan right inverting:

change in HAL_Due.cpp file those lines:

line from 575
```
 if (pwm_count_fan == 0) {
    #if FAN_COUNT >0
      LOOP_FAN() {
        if (!fans[f].pwm_hardware && ((fans[f].pwm_pos = (fans[f].Speed & FAN_PWM_MASK)) > 0))
          HAL::digitalWrite(fans[f].pin, fans[f].hardwareInverted ? HIGH : LOW);
      }
    #endif
```
 and line from 591

 ```
 #if FAN_COUNT > 0
    LOOP_FAN() {
      if (fans[f].Kickstart == 0 && !fans[f].pwm_hardware && fans[f].pwm_pos == pwm_count_fan && fans[f].pwm_pos != FAN_PWM_MASK)
        HAL::digitalWrite(fans[f].pin, fans[f].hardwareInverted ? LOW : HIGH);
    }
  #endif
```
for me it works with the last dev version

Regards

Hans

Hans007a, 2017-10-14:
I think here should be a standalone inverted value and not this:
`fans[f].hardwareInverted `

Hans007a, 2017-10-14:
I forgot, i changed in conditionals_post.h line from 1040:

```
#if ENABLED(INVERTED_FAN_PINS)
  #define _WRITE_FAN(pin, v) WRITE(pin, 255-v)//WRITE(pin, !v)
  #define FAN_INVERTED true
#else
  #define _WRITE_FAN(pin, v) WRITE(pin, v)
  #define FAN_INVERTED false
#endif
```

iosonopersia, 2017-10-14:
Hi, thank you very much for your effort! Are you able to open a pull request with your modifications? It would be much simpler to handle them.
Bye

MagoKimbra, 2017-10-14:
Sorry i do not understand your problem...
This line is obsolete and aren't use in FW... I clear it.
 #define _WRITE_FAN(pin, v) WRITE(pin, 255-v)//WRITE(pin, !v)
WRITE is DIGITAL 1 or 0 is possible. 255 - v is a error.

Hans007a, 2017-10-14:
Yes, You are right with _WRITE_FAN, but I wrote here all my changes including the dumb too :)

But important is, that now it's working with the fan :D

Hans007a, 2017-10-14:
My problem was, that the fan pwm was inverted after the v4.3.2 and #define INVERTED_FAN_PINS does not help. the fan was full on = 0 and disabled at 255. with my changes the fan works now in the right order.

thats all...

MagoKimbra, 2017-10-15:
Ok, if you swap LOW with HIGH as you did, then it means you have the inverted fan output, but just turn on INVERTED_FAN_PINS.
If you have activated it then just turn it off.
Because what it does is just that, if it is true it does LOW otherwise it does HIGH and vice versa later. So if your solution is to invert LOW with HIGH, it means you have to select INVERTED_FAN_PINS or deselect it.

rafaljot, 2017-10-31:
I made the same modifications in Repetier some days ago. But I found out the same result I could achieve, but it works with Repetier.
#define BOARD_FAN_SPEED 128
#define BOARD_FAN_MIN_SPEED 255

rafaljot, 2017-11-01:
@MagoKimbra , @Hans007a
Boys, in RAMPS-FD fan1, fan2 works normally. not inverted.  
Heaters are inverted. Pls check it again.
You have issue if you want to use Heater2 as fan controller.  And this issue is still unsolved. So this change did not make any sense. 
Frankly speaking only an individual INVERTED parameter for each MOSFET makes sense.

MagoKimbra, 2017-11-01:
But you ave INVERTED_HEATERS and INVERTED_FAN, if you have heaters inverted enabled only INVERTED_HEATERS

rafaljot, 2017-11-01:
Exactly. 
Look at this configuration for RAMPS-FD:
FAN1 ---> E3D fan (hardware 1 means on)
FAN2 ---> Cooler (hardware  1 means on)
EXT2 ---> system board fan (hardware  1 means off)

1. FAN1 and FAN2 are not inverted in RAMPS-FD, so current 1403.h is wrong.

2. if you define INVERTED_FAN_PINS it inverts all FANs

MagoKimbra, 2017-11-01:
Ok so you're telling me I have to remove #define INVERTED_FAN_PINS from card 1403??

Ok but with command M106 you can change inverted for each fan and save this setting in eeprom

rafaljot, 2017-11-01:
yes, remove it pls.

manusch-hz, 2018-01-28:
As far as I know, the RAMPS-FD board doesn't have an EEPROM.
So, MagoKimbra, do you know how to fix this bug?
INVERTED_FAN_PINS does not help here...

rafaljot, 2018-01-29:
This is not a bug. Arduino DUE doesn't have an EEPROM. 
You can buy i2c EEPROM, from $0.60 and just plug correctly to i2c port:
https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20180129111251&SearchText=i2c+eeprom

closed 2018-08-02 by iosonopersia