MKFirmware/MK4duo #262: Question about corexy and tmc2130 stallguard

Dzikibiszkopt, 2017-10-14:
Hello,. 
I have a question which is bothering me from few weeks because i work on corexy mechanics printer with arduino mega +ramps 1.4. Setup.  Im thinking on using tmc2130 stepsticks and want to use sensorless endstops with stallguard it is possible to use it on Yours firmware at corexy printer? (because i think its not from movement principles of this kind of machine).

Best regards, 
Dzikibiszkopt

kandidaten, 2017-11-19:
Did you get it to work? Im also building a coreXY wiht tmc2130 drivers, and i cant get it to work. Mega+RAMPS 1.4. Thanks 
/Adam

Dzikibiszkopt, 2017-11-19:
it is problem with fw i did it by modifying homing procedure as temporary fix because i can't even use advanced options section in config of tmc2130(compilation error).  I will write full fix wheen i figure it out.

iosonopersia, 2017-11-19:
@Dzikibiszkopt do you mean you modified the `src/mechanics/core/core_mechanics.cpp` function named `homeaxis(...)` ? I'm looking forward for your modifications. Thanks a lot for your effort.

By the way, have you tried with Marlin? @MagoKimbra tries always tries to catch up with Marlin's devs work, but sometimes something can be left behind by accident. If it works in Marlin, than we would just have to take a look at Marlin code.

Also it would be helpful if you submitted the compilation errors you found: that could be very useful when finding problems inside the code.

Bye!

Dzikibiszkopt, 2017-11-20:
Hi

MagoKimbra, 2017-11-20:
This solution is already available on the new fw v4.3.3 shortly available after Maker Faire Rome 2017 1-2-3 December.

Dzikibiszkopt, 2017-11-20:
@MargoKimbra good to hear that but i have question which solution ? Homing or/and skip strp detection?

iosonopersia, 2017-12-09:
Hi @Dzikibiszkopt , 4.3.3 has just come out, please have a try!

Dzikibiszkopt, 2017-12-17:
Hi @iosonopersia sorry for late reply but i don't receive any email notification. Yes it is working like a charm and btw. now tmc2130 can work with sdcard(earlier has bug with freezing when print from sd starts) ;)

iosonopersia, 2017-12-17:
@Dzikibiszkopt That's great news! We're happy to hear that it now works as expected!
@MagoKimbra this issue was solved. It can now be closed.

closed 2018-08-02 by iosonopersia