MKFirmware/MK4duo #265: Is it possible to change configurations e run time?

alequeiroz, 2017-10-17:
I'm trying to modify some #defines to use variables, always keeping an eye on memory used, of course. But I would like to hear the possibilities of modifying the code so that it is possible to implement modules, for example to exchange the cnc module for diamond hotend, in runtime, without having to make the modifications and recompile / upload the code.

MagoKimbra, 2017-10-17:
If you have a 32 bit it is possible, ability cnc e diamond and with gcode command switch to FFF, LASER or CNC mode.

alequeiroz, 2017-10-17:
But on mega?

iosonopersia, 2017-10-22:
MK4duo is moving towards that direction, @alequeiroz , but ATMega chips have got very little SRAM so a system in which everything is configurable at runtime would be impossible to implement, at least on those low spec chips. It's an hardware limitation that we cannot overcome

iosonopersia, 2018-08-02:
This issue can be safely closed, @MagoKimbra

closed 2018-08-02 by MagoKimbra