MKFirmware/MK4duo #267: Support for Mini Rambo?

johannes-otto, 2017-10-18:
Hi,
is it possible to support Mini Rambo? I tryed to use 301.h (Rambo) as  a template and used the pin assignments from Marlin.
Compiling fails, due to the DIGIPOTSS_PIN not beeing set (Minirambo doesn't seem to have one?).

iosonopersia, 2017-12-08:
Hi @johannes-otto . Marlin has recently added the Mini Rambo pins support to its firmware. In the next days we may add that also in MK4duo (we will inform you just in case). We've now reached `4.3.3 stable` version of the firmware with a lot of improvements and we hope you'll like it.

iosonopersia, 2017-12-09:
Hey, @johannes-otto . MiniRambo and MiniRambo 1.0a have now been added to MK4duo! This can now be closed, I believe.

johannes-otto, 2017-12-09:
Thanks very much.

closed 2017-12-09 by johannes-otto