MKFirmware/MK4duo #268: ultratronics & tmc2130

silversat, 2017-10-19:
### Issue description
compiling with HAVE_TMC2130 switch the fw hangs on start.
stepsticks are connected on x,y,z axis and e3 extruder socket is used instead of e0 (pins exchanged with extruder 3). miso/mosi/clk are connected to SPI connector and CSs are connected as in attached 1705.h file. 
Anyway, disconnecting all the stepsticks, the result is the same...
[MK4duo.zip](https://github.com/MKFirmware/MK4duo/files/1399612/MK4duo.zip)

sarpba, 2017-12-10:
same on DUE + RAMPS FD v1 + 4.3.3 stable

I want to use on X and Y axis only, but hang after upload.

[config.zip](https://github.com/MKFirmware/MK4duo/files/1545794/config.zip)

Beherith, 2017-12-21:
I have the same issue, where should I start debugging it? Just a pointer on where to start looking would be helpful. Cheers!

Beherith, 2017-12-27:
I have fixed tmc2130 support for mk4duo and arduino due, by fixing the tmc2130stepper library:

The reason it didnt boot was because it called its own begin() function from its constructor.
The hardware SPI didnt work for me, but the software SPI works, one just has to specify which pins to use. And the ARDUINO_SAM_ARCHIM flag was not passed by the arduino ide, so I changed that to ARDUINO_SAM_DUE

dembach, 2017-12-28:
Hey @Beherith , sounds good... looking forward to test it on my board, because i am stuck at the moment with the same #352 

great work
Markus

MagoKimbra, 2017-12-28:
Good!!

Beherith, 2017-12-28:
Ill ask the tmc2130 library dev if he is willing to backport changes, because calling the begin() function from the constructor is not good practice, and can break other things in subtle ways. But merely commenting that out allows the due to boot.

Beherith, 2017-12-28:
@dembach , please try it with this library, also make sure to set your software SPI pins in the TMC2130stepper.cpp

https://github.com/Beherith/TMC2130Stepper

sarpba, 2017-12-28:
Cool. BIG thankx. It's seems to be working on my hardwer. DUE + RAMPS-FDv1

MagoKimbra, 2017-12-28:
Thank's @Beherith !!!

dembach, 2017-12-28:
@Beherith , i´ll give it a try.....

Beherith, 2017-12-29:
Just for posterity's sake, I am using a Due + Hacked RAMPS 1.4 + 4x TMC2130.
@MagoKimbra , how shall we go about adding this support properly to MK4Duo? The changes I proposed are unlikely to be backported to the TMC2130Stepper library, as they may break existing code (especially the removal of begin() from the constructor where the code depends on it).

dembach, 2017-12-29:
Working !!!!! cheers to @Beherith

sh4jonny, 2018-01-19:
Hi guys how can it work now ? can it work with hardware SPI on DUE boards ? now i try to build it with forked lib on SW_SPI. but have a some errors like

sketch\src\feature\tmc\tmc.cpp: In function 'void _tmc_say_sgt(const char*, uint32_t)':
sketch\src\feature\tmc\../../../src/HAL/HAL_DUE/communication.h:104:76: error: call of overloaded print(const uint32_t&, int)' is ambiguous
#define SERIAL_VAL(val, ...)                (Com::print(val, ## __VA_ARGS__))

thank you!

closed 2018-01-19 by MagoKimbra

jbaiao, 2018-04-26:
@Beherith can you please tell me where i put  SW_SPI() pins in TMC2130stepper.cpp?
Tks!

MagoKimbra, 2018-04-26:
Why do you want to use SPI software? On ultratronics it works great with SPI hardware ...

commit 0373d975 (2018-01-19): Fix #268