MKFirmware/MK4duo #269: HOTEND_AUTO_FAN mixes up the fan index / HOTEND_AUTO_FAN not working

apphead, 2017-10-20:
### Issue description
HOTEND_AUTO_FAN dont start when HOTEND_AUTO_FAN_TEMPERATURE is reached.  

### Firmware version
V4.3.29

### Setup with HOTEND_AUTO_FAN disabled

In Configuration_Pins.h:
#define FAN0_PIN 11 // Heatsink fan
#define FAN1_PIN 9 // Print cooling fan
#define FAN2_PIN 8 // Controller fan
#define FAN3_PIN -1

#define H0_AUTO_FAN_PIN -1
#define H1_AUTO_FAN_PIN -1

*After uploading firmware, I always do a M502, M500, M501*

This is the output of M501 in the console:
Config:  M106 P0 pin:11 min:70 inverted:false
Config:  M106 P1 pin:9 min:70 inverted:false
Config:  M106 P2 pin:8 min:70 inverted:false

That's what I expected, fans respond accordingly

### Enabling HOTEND_AUTO_FAN
#define H0_AUTO_FAN_PIN FAN0_PIN
#define H1_AUTO_FAN_PIN FAN0_PIN

M501 shows:
Config:  M106 P0 pin:11 min:70 inverted:false
Config:  M106 P1 pin:9 min:70 inverted:false
Config:  M106 P2 pin:11 min:70 inverted:false
Config:  M106 P3 pin:11 min:70 inverted:false
Config:  M106 P4 pin:8 min:70 inverted:false

That does not seem right to me.
Issuing M106 P0 does not turn the fan on. Neither does M106 P2. Only the combination of M106 P0 and M106 P3 works. 

### Conclusion:
HOTEND_AUTO_FAN does not work, probably due to the index being mixed up and requiring multiple M106 commands. I guess, the HOTEND_AUTO_FANs should not show up as separate fans P2, P3 etc

MagoKimbra, 2017-10-20:
thank's.. I know this issue... The next step is to be able to choose which fan to associate with the hotend and the activation temperature, but actually the thing is a bit trickier.
For now, set it as a fan1 and fan2 to -1, leaving the pins only for H0_AUTO_FAN_PIN and H1_AUTO_FAN_PIN.

apphead, 2017-10-20:
I'm glad that it's not a bad config - I almost lost my mind and was not sure if it is a bug or me being stupid.
Anyway, I don't understand your workaround...what do you mean by _set it as a fan1 and fan2 to -1_?

And why don't you just remove H[0-3]_AUTO_FAN_PIN from the fan index???

iosonopersia, 2017-10-22:
Hi @apphead . With his latest commits, @MagoKimbra has fixed this. Have a look at the Online Configurator for the 4.3.X dev version. In feature section you'll now have the possibility of defining up to 6 fans (plus an optional controllerfan) and selecting which of them are autofan.

apphead, 2017-10-24:
Configuration_Overall.h

Again, this is tested on RADDS an latest Master

Autofan setup goal: {print cooling, heatsink 0, heatsink 1, controller, unused, unused} 
#define AUTO_FAN {-1,0,1,7,-1,-1} 

#### Test 1
	
#define FAN0_PIN 11
#define FAN1_PIN 9
#define FAN2_PIN 8

#define AUTO_FAN {0,1,-1,-1,-1,-1}
Config:  M106 P0 F250 U11 L70 I0
Config:  M106 P1 F250 U9 L70 I0
Config:  M106 P2 F250 U8 L70 I0

M106 P0 -> mosfet led stays off -> not right
M106 P1 -> on
M106 P2 -> on

#### Test 2

#define FAN0_PIN 8
#define FAN1_PIN 9
#define FAN2_PIN 11

Config:  M106 P0 F250 U8 L70 I0
Config:  M106 P1 F250 U9 L70 I0
Config:  M106 P2 F250 U11 L70 I0

M106 P0 -> on
M106 P1 -> on
M106 P2 -> mosfet led stays off -> not right

At first, I expected my mosfet to be bricked. So, I check Pin 11 with M106 P0 and the led switch on.
Strangely, I have the very same issue with Pin 13 (ORIG_HEATER_0_PIN). No response with MK4duo, but working on Repetier.
No matter if Pin 13 is mapped ORIG_HEATER_0_PIN or ORIG_HEATER_0_PIN or even ORIG_FAN0_PIN, the pin remains dead on MK4duo
.....Took me a whole day to find these issues. I was about to replace my RADDS, but then I found it working with repetier!

Pleeeeeease help. I am always one step away from a working setup when some other problems pops up. (with BLTouch in Repetier being the missing piece of the puzzle)

apphead, 2017-10-25:
These are the results with commit 6a859f5578f074727ae55ed5853a804d7be0a6a0

###Test A / Testing fan pins 8, 9, 11
Config:Fans:
Config:  M106 P0 F250 U9 L70 I0 -> OK
Config:  M106 P1 F250 U11 L70 I0 -> nothing
Config:  M106 P2 F250 U11 L70 I0 -> nothing
Config:  M106 P3 F250 U8 L70 I0 -> OK

Config:Fans:
Config:  M106 P0 F250 U11 L70 I0 -> nothing
Config:  M106 P1 F250 U9 L70 I0 -> OK
Config:  M106 P2 F250 U9 L70 I0 -> OK
Config:  M106 P3 F250 U8 L70 I0 -> OK

**Pin 11 is not responding, the settings do not matter at all. Even when used as heater0, it wont switch on.** Same thing with heaters: No matter if I assign Pin 13 to ORIG_HEATER_0_PIN or ORIG_HEATER_1_PIN (dualX), it just does not fire up. Obviously a show stopper.

###Test B / Starting fans manually
With #define AUTO_FAN {-1,-1,-1,-1,-1,-1} and
#define FAN0_PIN            ORIG_FAN0_PIN -> starts with M106 P0
#define FAN1_PIN            ORIG_FAN1_PIN -> starts with M106 P1
#define FAN2_PIN            ORIG_FAN1_PIN
#define FAN3_PIN            ORIG_FAN2_PIN

, with the exception of pin 11 (FAN3_PIN) which stays off.

###Test C

TC 1.1: M106 P1 / M107 P1 -> U9 Off
TC 1.2: M106 P1 / M107 P2 -> U9 remains On

TC 2.1: M106 P1 / M106 P2 / M107 P1 -> U9 Off
TC 2.2: M106 P1 / M106 P2 / M107 P2 -> U9 Off

In my opinion, TC 2.x don't work as expected. Only M107 P1 and M107 P2 should turn off U9. If I map autofans P1 to heater0 and P2 to heater1, then cooling shuts down on both heater if only one of the temps drop below min_temp.

Unfortunately, I cannot continue my test with only 1 heater working.

MagoKimbra, 2017-10-25:
I did not understand one thing because you have two fans on the same pin?
#define FAN0_PIN ORIG_FAN0_PIN -> starts with M106 P0
#define FAN1_PIN ORIG_FAN1_PIN -> starts with M106 P1
#define FAN2_PIN ORIG_FAN1_PIN
#define FAN3_PIN ORIG_FAN2_PIN

Why?? FAN2_PIN on ORIG_FAN1_PIN and FAN3_PIN on ORIG_FAN2_PIN??

There is a problem if a pin is mapped elsewhere as DIGITAL OUTPUT no longer works as a PWM timer. So if you change pin you have to memorize it in EEPROM and then reset the board.

apphead, 2017-10-25:
> you have two fans on the same pin
Yep. I simply have not enough pins for 2 x heatsink fan, 2 x part cooling and 1 controller fan, so I grouped them.
- Heatsink fans should turn on if one of the temp sensors goes above min_temp. How else would you define that in AUTO_FAN if you don't have two fans on the same pin??
- Part cooling should be default on M106 without P parameter 
- Downside is that fans turn on for both extruders simultaneously, but that is no biggie.

That is what my issue is about and in my understanding that is what you tried to fix with AUTO_FAN

MagoKimbra, 2017-10-25:
I hope I can explain.
You can set the same fan on two hotend. You put the hotend 0 on the fan 1 in configuration.
AUTO_FAN = {-1, 0, -1, -1, -1, -1}
Now with the M106 H command you can add a hotend to control on a fan.
M106 P1 H1 also adds hotend 1 to temperature control.
M106 P1 H-1 clears all hotend
Saving to EEPROM you keep the data.
So the P1 fan controls both Hotend 0 and Hotend 1.

Now the firmware uses hotend and fan hardware timers, only for the pins that are enabled.
For example, pin 11 that you have problems.
In variants.cpp in arduino modules.
...
// 10
   {PIOC, PIO_PC29B_TIOB7, ID_PIOC, PIO_PERIPH_B, PIO_DEFAULT, (PIN_ATTR_DIGITAL | PIN_ATTR_TIMER), NO_ADC, NO_ADC, NOT_ON_PWM, TC2_CHB7}, // TIOB7
// 11
   {PIOD, PIO_PD7B_TIOA8, ID_PIOD, PIO_PERIPH_B, PIO_DEFAULT, (PIN_ATTR_DIGITAL | PIN_ATTR_TIMER), NO_ADC, NO_ADC, NOT_ON_PWM, TC2_CHA8}, // TIOA8
// 12
   {PIOD, PIO_PD8B_TIOB8, ID_PIOD, PIO_PERIPH_B, PIO_DEFAULT, (PIN_ATTR_DIGITAL | PIN_ATTR_TIMER), NO_ADC, NO_ADC, NOT_ON_PWM, TC2_CHB8}, // TIOB8
...
Pin 11 is a hardware pin that uses the timer (PIN_ATTR_TIMER) exactly TC2_CHA8.
I did not try all the pins, but my alligator card works well both the PWM and the timer pins. I can not understand why you have problems with pins 11 and 12.

Repetier firmware not use hardware PWM, but software as MK did before...

apphead, 2017-10-25:
I am not sure that I can follow your explanation. But fact is that 
Config: M106 P0 F250 U11 L70 I0 -> nothing
Config: M106 P1 F250 U9 L70 I0 -> OK
Config: M106 P2 F250 U9 L70 I0 -> OK
Config: M106 P3 F250 U8 L70 I0 -> OK
should actually work for P0 / Pin11 but it does not. That does not make sense in either case, PWM or hardware timer. Right?

Pin11 does actually work in another context. Maybe it is related to https://github.com/MKFirmware/MK4duo/issues/293

iosonopersia, 2018-08-06:
Hi @apphead . As I told you here: https://github.com/MKFirmware/MK4duo/issues/293#issuecomment-410514201 , each timer counter of the SAM3X8E (there are 9 of them) is linked to 2 pins. Since for Due we use only hardware pwm generated by those timers, and because of how we use them, you can't use both the pins connected to the same timer counter at the same time.

So, returning to this issue:
  * Maybe you can't use pin 11 because you already use pin 12 for an heater or something else that needs PWM
  * Maybe you can't use pin 13 for the same reason with the pin 2

> Pin11 does actually work in another context. Maybe it is related to #293

If you use pin 11 for something that doesn't use PWM, than it will be controlled directly from the SAM3X8E and not from the timer counter, so it won't collide with the functioning of pin 12.

closed 2018-08-06 by iosonopersia