MKFirmware/MK4duo #271: Erro compilation in power_supply

Briangsouza, 2017-10-20:
Arduino: 1.8.2 (Windows 10), Placa:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\src/pins.h:33:0,

                 from sketch\base.h:81,

                 from C:\Users\Brian\Dropbox\PROJETOS REPRAP\COREMAKER\MK4duo-4_3_24\MK4duo\MK4duo.ino:257:

sketch\src/boards/33.h:207:0: warning: "ORIG_BEEPER_PIN" redefined

       #define ORIG_BEEPER_PIN   37

 ^

sketch\src/boards/33.h:143:0: note: this is the location of the previous definition

 #define ORIG_BEEPER_PIN -1

 ^

sketch\src/boards/33.h:213:0: warning: "SD_DETECT_PIN" redefined

       #define SD_DETECT_PIN     49

 ^

sketch\src/boards/33.h:146:0: note: this is the location of the previous definition

 #define SD_DETECT_PIN -1

 ^

sketch\src/boards/33.h:214:0: warning: "KILL_PIN" redefined

       #define KILL_PIN          41

 ^

sketch\src/boards/33.h:148:0: note: this is the location of the previous definition

 #define KILL_PIN -1

 ^

In file included from sketch\src/pins.h:299:0,

                 from sketch\base.h:81,

                 from C:\Users\Brian\Dropbox\PROJETOS REPRAP\COREMAKER\MK4duo-4_3_24\MK4duo\MK4duo.ino:257:

sketch\src/../Configuration_Pins.h:154:0: warning: "Z2_MAX_PIN" redefined

 #define Z2_MAX_PIN          ORIG_X_MAX_PIN

 ^

In file included from sketch\base.h:81:0,

                 from C:\Users\Brian\Dropbox\PROJETOS REPRAP\COREMAKER\MK4duo-4_3_24\MK4duo\MK4duo.ino:257:

sketch\src/pins.h:278:0: note: this is the location of the previous definition

   #define Z2_MAX_PIN    -1

 ^

In file included from sketch\src\../src/pins.h:33:0,

                 from sketch\src\../base.h:81,

[... 3051 lines cut ...]

sketch\src\utility\../../src/pins.h:278:0: note: this is the location of the previous definition

   #define Z2_MAX_PIN    -1

 ^

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:24:0,

                 from sketch\src\utility\../../base.h:26,

                 from sketch\src\utility\power_supply.cpp:23:

sketch\src\utility\power_supply.cpp:29:3: error: expected initializer before 'bool'

   bool Power::powersupply_on = 

   ^

exit status 1
Erro compilando para a placa Arduino/Genuino Mega or Mega 2560

Este relatório teria mais informações com
"Mostrar a saida detalhada durante a compilação"
opção pode ser ativada em "Arquivo -> Preferências"

MagoKimbra, 2017-10-20:
Warning warning warning.....
Disabled warning in arduino ide...

Briangsouza, 2017-10-20:
#if HAS_POWER_SWITCH

  Power powerManager *;*<-------------------------------------------

  bool Power::powersupply_on = 
    #if ENABLED(PS_DEFAULT_OFF)
      false
    #else
      true
    #endif
  ;

  void Power::check() {
    static millis_t nextPowerCheck = 0;
    millis_t ms = millis();

iosonopersia, 2017-10-21:
This is the error... Please @Briangsouza edit your issue deleting all those error messages, since 99% of them are completely unnecessary!!! Your issues are polluting this repository...

``` cpp
In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:24:0,

             from sketch\src\utility\../../base.h:26,

             from sketch\src\utility\power_supply.cpp:23:
sketch\src\utility\power_supply.cpp:29:3: error: expected initializer before 'bool'

bool Power::powersupply_on =

^
```

iosonopersia, 2017-10-21:
@MagoKimbra this can be closed, is referred to the 4.3.24 version...

closed 2018-06-02 by Briangsouza