MKFirmware/MK4duo #272: error compilation in ABL

Briangsouza, 2017-10-20:
Arduino: 1.8.2 (Windows 10), Placa:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\src/pins.h:33:0,

                 from sketch\base.h:81,

                 from C:\Users\Brian\Dropbox\PROJETOS REPRAP\COREMAKER\MK4duo-4_3_24\MK4duo\MK4duo.ino:257:

sketch\src/boards/33.h:207:0: warning: "ORIG_BEEPER_PIN" redefined

       #define ORIG_BEEPER_PIN   37

 ^

sketch\src/boards/33.h:143:0: note: this is the location of the previous definition

 #define ORIG_BEEPER_PIN -1

 ^

sketch\src/boards/33.h:213:0: warning: "SD_DETECT_PIN" redefined

       #define SD_DETECT_PIN     49

 ^

sketch\src/boards/33.h:146:0: note: this is the location of the previous definition

 #define SD_DETECT_PIN -1

 ^

sketch\src/boards/33.h:214:0: warning: "KILL_PIN" redefined

       #define KILL_PIN          41

 ^

sketch\src/boards/33.h:148:0: note: this is the location of the previous definition

 #define KILL_PIN -1

 ^

In file included from sketch\src/pins.h:299:0,

                 from sketch\base.h:81,

                 from C:\Users\Brian\Dropbox\PROJETOS REPRAP\COREMAKER\MK4duo-4_3_24\MK4duo\MK4duo.ino:257:

sketch\src/../Configuration_Pins.h:154:0: warning: "Z2_MAX_PIN" redefined

 #define Z2_MAX_PIN          ORIG_X_MAX_PIN

 ^

In file included from sketch\base.h:81:0,

                 from C:\Users\Brian\Dropbox\PROJETOS REPRAP\COREMAKER\MK4duo-4_3_24\MK4duo\MK4duo.ino:257:

sketch\src/pins.h:278:0: note: this is the location of the previous definition

   #define Z2_MAX_PIN    -1

 ^

In file included from sketch\src\../src/pins.h:33:0,

                 from sketch\src\../base.h:81,

[... 1665 lines cut ...]

sketch\src\mechanics\../../src/bedlevel/bedlevel.h:54:42: error: within this context

       #define ABL_BG_SPACING(A) bedlevel.bilinear_grid_spacing_virt[A]

                                          ^

sketch\src\mechanics\../../src/mechanics/mechanics.h:57:38: note: in definition of macro 'LOGICAL_Y_POSITION'

   #define LOGICAL_Y_POSITION(POS)   (POS)

                                      ^

sketch\src\mechanics\core_mechanics.cpp:459:84: note: in expansion of macro 'ABL_BG_SPACING'

         destination[Y_AXIS] = LOGICAL_Y_POSITION(bedlevel.bilinear_start[Y_AXIS] + ABL_BG_SPACING(Y_AXIS) * gcy);

                                                                                    ^

exit status 1
Erro compilando para a placa Arduino/Genuino Mega or Mega 2560

Este relatório teria mais informações com
"Mostrar a saida detalhada durante a compilação"
opção pode ser ativada em "Arquivo -> Preferências"

MagoKimbra, 2017-10-20:
Please disabled warning on your arduino ide...
If just disabled, enabled closed arduino ide, reopen arduino ide disabled and reclosed.
Re open and compile...

Briangsouza, 2017-10-20:
where I deactivated

iosonopersia, 2017-10-20:
File->Preferences...->Compiler warnings select Nothing

Briangsouza, 2017-10-20:
I enabled closed arduino ide, reopen arduino ide disabled and reclosed.
Re open and compile... did not solve

iosonopersia, 2017-10-21:
@MagoKimbra please close this issue. @Briangsouza let's continue this discussion in #276, thanks.

closed 2017-10-21 by MagoKimbra