MKFirmware/MK4duo #273: error compilation LIN ADVANCE

Briangsouza, 2017-10-20:
Arduino: 1.8.2 (Windows 10), Placa:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\src/pins.h:33:0,

                 from sketch\base.h:81,

                 from C:\Users\Brian\Dropbox\PROJETOS REPRAP\COREMAKER\MK4duo-4_3_24\MK4duo\MK4duo.ino:257:

sketch\src/boards/33.h:207:0: warning: "ORIG_BEEPER_PIN" redefined

       #define ORIG_BEEPER_PIN   37

 ^

sketch\src/boards/33.h:143:0: note: this is the location of the previous definition

 #define ORIG_BEEPER_PIN -1

 ^

sketch\src/boards/33.h:213:0: warning: "SD_DETECT_PIN" redefined

       #define SD_DETECT_PIN     49

 ^

sketch\src/boards/33.h:146:0: note: this is the location of the previous definition

 #define SD_DETECT_PIN -1

 ^

sketch\src/boards/33.h:214:0: warning: "KILL_PIN" redefined

       #define KILL_PIN          41

 ^

sketch\src/boards/33.h:148:0: note: this is the location of the previous definition

 #define KILL_PIN -1

 ^

In file included from sketch\src/pins.h:299:0,

                 from sketch\base.h:81,

                 from C:\Users\Brian\Dropbox\PROJETOS REPRAP\COREMAKER\MK4duo-4_3_24\MK4duo\MK4duo.ino:257:

sketch\src/../Configuration_Pins.h:154:0: warning: "Z2_MAX_PIN" redefined

 #define Z2_MAX_PIN          ORIG_X_MAX_PIN

 ^

In file included from sketch\base.h:81:0,

                 from C:\Users\Brian\Dropbox\PROJETOS REPRAP\COREMAKER\MK4duo-4_3_24\MK4duo\MK4duo.ino:257:

sketch\src/pins.h:278:0: note: this is the location of the previous definition

   #define Z2_MAX_PIN    -1

 ^

In file included from sketch\src\../src/pins.h:33:0,

                 from sketch\src\../base.h:81,

                 from sketch\src\MK_Main.cpp:31:

sketch\src\../src/boards/33.h:207:0: warning: "ORIG_BEEPER_PIN" redefined

       #define ORIG_BEEPER_PIN   37

 ^

sketch\src\../src/boards/33.h:143:0: note: this is the location of the previous definition

 #define ORIG_BEEPER_PIN -1

 ^

sketch\src\../src/boards/33.h:213:0: warning: "SD_DETECT_PIN" redefined

       #define SD_DETECT_PIN     49

 ^

sketch\src\../src/boards/33.h:146:0: note: this is the location of the previous definition

 #define SD_DETECT_PIN -1

 ^

sketch\src\../src/boards/33.h:214:0: warning: "KILL_PIN" redefined

       #define KILL_PIN          41

 ^

sketch\src\../src/boards/33.h:148:0: note: this is the location of the previous definition

 #define KILL_PIN -1

 ^

In file included from sketch\src\../src/pins.h:299:0,

                 from sketch\src\../base.h:81,

                 from sketch\src\MK_Main.cpp:31:

sketch\src\../src/../Configuration_Pins.h:154:0: warning: "Z2_MAX_PIN" redefined

 #define Z2_MAX_PIN          ORIG_X_MAX_PIN

 ^

In file included from sketch\src\../base.h:81:0,

                 from sketch\src\MK_Main.cpp:31:

sketch\src\../src/pins.h:278:0: note: this is the location of the previous definition

   #define Z2_MAX_PIN    -1

 ^

In file included from sketch\src\../base.h:81:0,

                 from sketch\src\MK_Main.cpp:31:

sketch\src\MK_Main.cpp: In function 'bool pin_is_protected(uint8_t)':

sketch\src\../src/pins.h:419:24: warning: narrowing conversion of '173' from 'int' to 'const int8_t {aka const signed char}' inside { } [-Wnarrowing]

                        }

                        ^

sketch\src\MK_Main.cpp:423:50: note: in expansion of macro 'SENSITIVE_PINS'

   static const int8_t sensitive_pins[] PROGMEM = SENSITIVE_PINS;

                                                  ^

sketch\src\../src/pins.h:419:24: warning: narrowing conversion of '174' from 'int' to 'const int8_t {aka const signed char}' inside { } [-Wnarrowing]

                        }

                        ^

sketch\src\MK_Main.cpp:423:50: note: in expansion of macro 'SENSITIVE_PINS'

   static const int8_t sensitive_pins[] PROGMEM = SENSITIVE_PINS;

                                                  ^

sketch\src\../src/pins.h:419:24: warning: narrowing conversion of '159' from 'int' to 'const int8_t {aka const signed char}' inside { } [-Wnarrowing]

                        }

                        ^

sketch\src\MK_Main.cpp:423:50: note: in expansion of macro 'SENSITIVE_PINS'

   static const int8_t sensitive_pins[] PROGMEM = SENSITIVE_PINS;

                                                  ^

sketch\src\../src/pins.h:419:24: warning: narrowing conversion of '159' from 'int' to 'const int8_t {aka const signed char}' inside { } [-Wnarrowing]

                        }

                        ^

sketch\src\MK_Main.cpp:423:50: note: in expansion of macro 'SENSITIVE_PINS'

   static const int8_t sensitive_pins[] PROGMEM = SENSITIVE_PINS;

                                                  ^

sketch\src\MK_Main.cpp: In function 'void gcode_M900()':

sketch\src\MK_Main.cpp:8135:28: error: 'class Planner' has no member named 'extruder_advance_k'

     if (newK >= 0) planner.extruder_advance_k = newK;

                            ^

sketch\src\MK_Main.cpp:8145:28: error: 'class Planner' has no member named 'advance_ed_ratio'

     if (newR >= 0) planner.advance_ed_ratio = newR;

                            ^

In file included from sketch\src\../src/HAL/HAL.h:42:0,

                 from sketch\src\../base.h:85,

                 from sketch\src\MK_Main.cpp:31:

sketch\src\MK_Main.cpp:8147:44: error: 'class Planner' has no member named 'extruder_advance_k'

     SERIAL_SMV(ECHO, "Advance K=", planner.extruder_advance_k);

                                            ^

sketch\src\../src/HAL/HAL_AVR/communication.h:68:75: note: in definition of macro 'SERIAL_MV'

 #define SERIAL_MV(msg, val, ...)            (serial_print_pair(PSTR(msg), val, ## __VA_ARGS__))

                                                                           ^

sketch\src\MK_Main.cpp:8147:5: note: in expansion of macro 'SERIAL_SMV'

     SERIAL_SMV(ECHO, "Advance K=", planner.extruder_advance_k);

     ^

sketch\src\MK_Main.cpp:8149:17: error: 'class Planner' has no member named 'advance_ed_ratio'

     if (planner.advance_ed_ratio) SERIAL_VAL(planner.advance_ed_ratio);

                 ^

In file included from sketch\src\../src/HAL/HAL.h:42:0,

                 from sketch\src\../base.h:85,

                 from sketch\src\MK_Main.cpp:31:

sketch\src\MK_Main.cpp:8149:54: error: 'class Planner' has no member named 'advance_ed_ratio'

     if (planner.advance_ed_ratio) SERIAL_VAL(planner.advance_ed_ratio);

                                                      ^

sketch\src\../src/HAL/HAL_AVR/communication.h:61:59: note: in definition of macro 'SERIAL_VAL'

 #define SERIAL_VAL(val, ...)                (serial_print(val, ## __VA_ARGS__))

                                                           ^

exit status 1
Erro compilando para a placa Arduino/Genuino Mega or Mega 2560

Este relatório teria mais informações com
"Mostrar a saida detalhada durante a compilação"
opção pode ser ativada em "Arquivo -> Preferências"

iosonopersia, 2017-10-21:
Hi @Briangsouza , I've summarized your error messages here. Please edit your issue deleting all that text, because it makes difficult to read this issue. Please tell us which version of MK4duo you were trying to compile. I'll make a guess: is it 4.3.2 stable?

``` cpp
sketch\src\MK_Main.cpp:` In function 'void gcode_M900()':

sketch\src\MK_Main.cpp:8135:28: error: 'class Planner' has no member named 'extruder_advance_k'

 if (newK >= 0) planner.extruder_advance_k = newK;

                        ^
sketch\src\MK_Main.cpp:8145:28: error: 'class Planner' has no member named 'advance_ed_ratio'

 if (newR >= 0) planner.advance_ed_ratio = newR;

                        ^
In file included from sketch\src../src/HAL/HAL.h:42:0,

             from sketch\src\../base.h:85,

             from sketch\src\MK_Main.cpp:31:
sketch\src\MK_Main.cpp:8147:44: error: 'class Planner' has no member named 'extruder_advance_k'

 SERIAL_SMV(ECHO, "Advance K=", planner.extruder_advance_k);

                                        ^
sketch\src../src/HAL/HAL_AVR/communication.h:68:75: note: in definition of macro 'SERIAL_MV'

#define SERIAL_MV(msg, val, ...) (serial_print_pair(PSTR(msg), val, ## VA_ARGS))

                                                                       ^
sketch\src\MK_Main.cpp:8147:5: note: in expansion of macro 'SERIAL_SMV'

 SERIAL_SMV(ECHO, "Advance K=", planner.extruder_advance_k);

 ^
sketch\src\MK_Main.cpp:8149:17: error: 'class Planner' has no member named 'advance_ed_ratio'

 if (planner.advance_ed_ratio) SERIAL_VAL(planner.advance_ed_ratio);

             ^
In file included from sketch\src../src/HAL/HAL.h:42:0,

             from sketch\src\../base.h:85,

             from sketch\src\MK_Main.cpp:31:
sketch\src\MK_Main.cpp:8149:54: error: 'class Planner' has no member named 'advance_ed_ratio'

 if (planner.advance_ed_ratio) SERIAL_VAL(planner.advance_ed_ratio);

                                                  ^
sketch\src../src/HAL/HAL_AVR/communication.h:61:59: note: in definition of macro 'SERIAL_VAL'

#define SERIAL_VAL(val, ...) (serial_print(val, ## VA_ARGS))
```

Briangsouza, 2017-10-21:
Arduino: 1.8.5 (Mac OS X), Placa:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch/src/MK_Main.cpp: In function 'void gcode_M900()':
sketch/src/MK_Main.cpp:8135:28: error: 'class Planner' has no member named 'extruder_advance_k'
     if (newK >= 0) planner.extruder_advance_k = newK;

                            ^
sketch/src/MK_Main.cpp:8145:28: error: 'class Planner' has no member named 'advance_ed_ratio'
     if (newR >= 0) planner.advance_ed_ratio = newR;

                            ^
In file included from sketch/src/../src/HAL/HAL.h:42:0,
                 from sketch/src/../base.h:85,
                 from sketch/src/MK_Main.cpp:31:
sketch/src/MK_Main.cpp:8147:44: error: 'class Planner' has no member named 'extruder_advance_k'
     SERIAL_SMV(ECHO, "Advance K=", planner.extruder_advance_k);

                                            ^
sketch/src/../src/HAL/HAL_AVR/communication.h:68:75: note: in definition of macro 'SERIAL_MV'
 #define SERIAL_MV(msg, val, ...)            (serial_print_pair(PSTR(msg), val, ## __VA_ARGS__))
                                                                           ^
sketch/src/MK_Main.cpp:8147:5: note: in expansion of macro 'SERIAL_SMV'
     SERIAL_SMV(ECHO, "Advance K=", planner.extruder_advance_k);

     ^
sketch/src/MK_Main.cpp:8149:17: error: 'class Planner' has no member named 'advance_ed_ratio'
     if (planner.advance_ed_ratio) SERIAL_VAL(planner.advance_ed_ratio);

                 ^
In file included from sketch/src/../src/HAL/HAL.h:42:0,
                 from sketch/src/../base.h:85,
                 from sketch/src/MK_Main.cpp:31:
sketch/src/MK_Main.cpp:8149:54: error: 'class Planner' has no member named 'advance_ed_ratio'
     if (planner.advance_ed_ratio) SERIAL_VAL(planner.advance_ed_ratio);

                                                      ^
sketch/src/../src/HAL/HAL_AVR/communication.h:61:59: note: in definition of macro 'SERIAL_VAL'
 #define SERIAL_VAL(val, ...)                (serial_print(val, ## __VA_ARGS__))
                                                           ^
exit status 1
Erro compilando para a placa Arduino/Genuino Mega or Mega 2560

mk4duo 4.3.24

MagoKimbra, 2017-10-21:
MK4.3.24????? Is very old... please download maste or 4.3.2 stable

Briangsouza, 2017-10-21:
ok I use now master! but error in sdsupport

iosonopersia, 2017-10-21:
@MagoKimbra this can be closed, is referred to the 4.3.24 version...

closed 2017-10-21 by MagoKimbra