MKFirmware/MK4duo #274: Pins for RADDS Extension Board V2 and V3

apphead, 2017-10-20:
A small contribution to your great project:
### Board: RADDS, file \MK4duo\src\boards\1401.h

//Pinout for Extension Boards V2 / V3
//Check for V1
//###EXTRUDER_3
//labelled E4 on V3
#define ORIG_E3_STEP_PIN 35
#define ORIG_E3_DIR_PIN 33
#define ORIG_E3_ENABLE_PIN 37
#define ORIG_E3_CS_PIN -1
#define ORIG_SOL3_PIN -1

//###EXTRUDER_4
//labelled E5 on V3
#define ORIG_E4_STEP_PIN 29
#define ORIG_E4_DIR_PIN 27
#define ORIG_E4_ENABLE_PIN 31
#define ORIG_E4_CS_PIN -1
#define ORIG_SOL4_PIN -1

//###EXTRUDER_5
//labelled E6 on V3, non-existent on V2
#define ORIG_E5_STEP_PIN 67
#define ORIG_E5_DIR_PIN 66
#define ORIG_E5_ENABLE_PIN 68
#define ORIG_E5_CS_PIN -1
#define ORIG_SOL5_PIN -1

iosonopersia, 2017-10-24:
Hi @apphead , thank you so much for your help. Extruder 3 and 4 (aka E4 and E5) have those pins already defined in the board file (src/board/1401.h). Extruder 5 pins have not been defined, maybe @MagoKimbra knows why.

apphead, 2017-10-24:
Strangely, I missed that. About extruder 5 aka E6, it's only available on the latest extension board V3. Older boards have just 2 additional stepper sockets.

closed 2017-10-24 by apphead