MKFirmware/MK4duo #278: open failed, File: init.g

sh4jonny, 2017-10-21:
hi on version 4.3.29 get subj message after connect to printer on Pronterface

here is all out:

Connecting...
start
Printer is now online.
Info:PowerUp
Echo:MK4duo_4.3.29_dev
Echo:Last Updated: Oct 21 2017 17:26:15 | Author: (master, my_web)
Echo:Compiled: Oct 21 2017
Echo:Free Memory: 84072 PlannerBufferBytes: 2560
SD card ok
okHardcoded SD Default Settings Loaded
MKV41 stored settings retrieved (287 bytes; crc 42103)
open failed, File: init.g
SD card ok

### Compile errors - IMPORTANT
no any compile errors

### Firmware version
last git 4.3.29 from 21.10.2017

### My setup
web setup with manual options
nothing special

thank you for support

Prinz12, 2017-10-21:
I have the same bug with RepetierHost. Communication is no longer possible

sh4jonny, 2017-10-21:
what communication you mean ? do you have any other troubles?

Prinz12, 2017-10-21:
It is no longer possible to communicate between host and due with radds. No commands are accepted and executed.

MagoKimbra, 2017-10-21:
init.g is a file for initialize printer to power on. If hou have not this file no problem.

sh4jonny, 2017-10-21:
This is file from “host” side software?

MagoKimbra, 2017-10-21:
no in sd card..

sh4jonny, 2017-10-21:
ok thank you!

closed 2017-10-21 by sh4jonny