MKFirmware/MK4duo #283: Connection with Due not possible.

Prinz12, 2017-10-21:
Hello,
I have a problem connecting to an Arduino Due.
With the current dev it is not possible to establish a fast and reliable connection with the repeater host.
I get the following warning every few lines:
Warning: Seems like we missed a ok, got a wait - continue send.

and

Communication timeout - reset send buffer block

When executing a print job, the commands are sent to the due only occasionally.

I've tried all the baud rates.

With an Arduino Mega this problem does not exist.
[image]

apphead, 2017-10-25:
I can confirm this behaviour with 4.3.29 dev I configured and downloaded from configurator. Pronterface does not accept any commands either.

MagoKimbra, 2017-10-25:
I try this version on Alligator, Ultratronics, and arduino due with Piccolo3D and haven't problem...

apphead, 2017-10-25:
My setup is Due / RADDS 1.5, 250000 baud. Pulled last master commit 6a859f5578f074727ae55ed5853a804d7be0a6a0 and now the connection is back online.

Using Pronterface on Windows 8.1:
Connection is ok with 
//#define EEPROM_SETTINGS
//#define EEPROM_CHITCHAT

Connection is ok with 
#define EEPROM_SETTINGS
#define EEPROM_CHITCHAT 

First test without physical EEPROM, no SD in RADDS. Second run with EEPROM added to I2C, works in Repetier..

Surprisingly, when I try the same with Repetier Host 2.0.5, the board resets every ~10secs, irrespective of EEPROM state. Only when I click disconnect, the boot loop ends. Here is my settings in Repetier Host: (again, no trouble using Repetier firmware)
[image]

Prinz12, 2017-10-25:
Unfortunately, I'm not very good at programming and I don't understand what happened there.
For me commit 3d97cf4d26a0c5fd966d1330c87439524c904ac1 works without restrictions. See pictures
Printer boot
[image]
Printer print
[image]
Transmission is no longer possible with the next commit c1f5fe5f5b03fa650f80c53a01d47eb33ff09d1e10. See pictures 
Printer boot
[image]
Printer print
[image]
Strangely enough, there are also no more "wait" and "ok" messages in the log of RepetierHost.

In MK4duo only the board "Raads" was selected and no other settings were changed.
RepeaterHost 2.0.5 with the same settings as yours except timeout 2 seconds. As test the same gcode as before and absolutely no changes in the settings. Does commit 3d97cf4d26a0c5fd966d1330c87439524c904ac1 still work with you?

apphead, 2017-10-25:
Tried commit https://github.com/MKFirmware/MK4duo/commit/3d97cf4d26a0c5fd966d1330c87439524c904ac1 and just added this to config_overall:
#define BAUDRATE 250000
#define MOTHERBOARD BOARD_RADDS
#define MECHANISM MECH_CARTESIAN

Works without bootloop with Repetier Host. Commuincation Timeout had no impact, tested with 2s and 40s

> Strangely enough, there are also no more "wait" and "ok" messages in the log of RepetierHost.
Maybe you accidentially disabled ACK in Rep.host console??
[image]

apphead, 2017-10-25:
With the same commit I added a 2nd and 3rd extruder (to test my heaterpins) and found a few things that just don't add up.

Nothing changed from commit except my additions to config_overall:
#define BAUDRATE 250000
#define MOTHERBOARD BOARD_RADDS
#define MECHANISM MECH_CARTESIAN
#define EXTRUDERS 2
#define TEMP_SENSOR_0 1 
#define TEMP_SENSOR_1 1 

Heater0 and Heater1 are working (Pin 13 is back!!!). M502 tells me _Hardcoded Default Settings Loaded_

Now the fun part, add another extruder:
#define EXTRUDERS 3
#define TEMP_SENSOR_0 1 
#define TEMP_SENSOR_1 1 
#define TEMP_SENSOR_2 1 

NONE of the heaters respond to my input and M502 tells me.......nothing at all. No warnings in during compilation either and why do 2 Extruder work and the third breaks everything??? 

Just to make sure all heaters have a pin assigned, I checked in 1401.h, and yes they are. So it is supposed to work with the minimal setup above, right!?
#define ORIG_HEATER_0_PIN 13
#define ORIG_HEATER_1_PIN 12
#define ORIG_HEATER_2_PIN 11

The longer I use this firmware, the more I get the feeling that it is seriously bugged and it's going down hill. I would suggest more testing and fixing errors before implementing new features that start acting weird because of the debt already baked in.

MagoKimbra, 2017-10-26:
I'm sorry, but this version is a development, if you have too many problems use the 4.3.2 stable. This version I need to put it on its own to let it try and see where there are problems. I do not have all the boards and I need you to give me feedback to find the possible bugs.

Prinz12, 2017-10-26:
@apphead 
That must have been a misunderstanding. With commit https://github.com/MKFirmware/MK4duo/commit/3d97cf4d26a0c5fd966d1330c87439524c904ac1 the communication runs with me without any problems. From commit c1f5fe5b03fa650f80c53a01d47eb33ff09d1e10 nothing works anymore and the wait and ok comments are missing.

@MagoKimbra  
Of course, the firmware is under development and not everything works at once, so I would like to help you identify this bug. I think the firmware is very good and I would be happy to continue using it.
Let me know if I can help you with anything.

apphead, 2017-10-26:
@MagoKimbra 
I did not mean to offend you, sorry if I did. After spending days and days testing the firmware, I guess I must have lost patience and temper. As you probably know, testing is an arduos and time-consuming process....which made me contemplating the idea of a testing framework that would cut the time required to run tests. Since this is a bit off topic, I openend https://github.com/MKFirmware/MK4duo/issues/294

iosonopersia, 2017-10-26:
c1f5fe5 caused a lot of trouble:

- #289 Full Smart Graphic Controller display doesn't work anymore
- #269 strange auto fan problems
- #293 heater pins not working properly 
- #283 connection with Due (maybe also with Mega??) not reliable

We should check that commit with a lot of attention, unluckily it's a very big commit...

Prinz12, 2017-10-27:
Thank you for the quick troubleshooting. The communication works again without problems.

iosonopersia, 2017-10-28:
Hi, was that fixed also for @apphead ? Just to be 100% sure.

Prinz12, 2017-10-28:
Oh, sorry. I don't know if the bug with @apphead is fixed. I assumed it was because he had the same Setup as me. @apphead you can give feedback if the communication is working without any restrictions?!

apphead, 2017-10-29:
Communication is working for me too. Can't test the display tho as I cannot find it atm.

Prinz12, 2017-10-29:
Okay, thank you. I'm going to close this one.

closed 2017-10-29 by Prinz12

apphead, 2017-10-29:
I still have troubles using Repetier Host 2.0.5. The board keeps rebooting every 10secs....no issues with pronterface. Hmmm

Prinz12, 2017-10-31:
I don't understand why? It works fine for me. If you want to send me your settings, I'll try them.