MKFirmware/MK4duo #289: RepRap full graphics display don't works

Hans007a, 2017-10-24:
After the new changes the RepRap full graphics display don't works...

Any changes on the SPI?

With the version from one week ago was all ok

MagoKimbra, 2017-10-24:
Please write your configuration board.

Hans007a, 2017-10-24:
ups, sorry

RAMPS-FD v1 and Due

sh4jonny, 2017-10-24:
Have this problem too. Work on 4.3.24 and don’t at 4.3.27. Try 4.3.29 with no changes... then rebuild config in web interface and it work! But now #286 ((((

sh4jonny, 2017-10-24:
omg i add missing #define AUTO_FAN {-1,0,-1,-1,-1,-1} to config build and upload. and now my Full Graphics Didplay don't work too (((((

apphead, 2017-10-24:
I don't have a  Full Graphics Didplay on my RADDS (yet), but my gut tells me the new AUTO_FAN implementation messed up a few things. Maybe related to https://github.com/MKFirmware/MK4duo/issues/269#issuecomment-338835047 ??
Are your Fan_Pin1 (heatsink 0) and heater0 working?

iosonopersia, 2017-10-24:
Hi everyone. This is the MK4duo situation just before the autofan implementation was merged: https://github.com/MKFirmware/MK4duo/tree/8c1d1a644f11d131e4691de93120bdb4065bf01e 

Please try downloading the firmware from there and see if the display works, this would tell us if the problem come from there

Hans007a, 2017-10-24:
I was one week in hollidays and the last working version was from the 15.10.2017

sh4jonny, 2017-10-25:
so guys. Alberto and Simone i think if you include new option that critically needed for build in Configuration_Basic.h it will be a good idea. peoples who use old config will have default values and correct build. and will see needed options from commit in Configuration_Basic.h. also will be good if you write about new options in changelog.md

iosonopersia, 2017-10-25:
Meanwhile I've found that 1400.h (Piccolo3D), 1401.h (RADDS) and 1403.h (Ramps FD v1) share a common feature: they're the only Due boards that have I2C_EEPROM enabled by default. In 1400.h and 1403.h that was added on 24th of September. In 1401.h the same happened on 8th of September by me.

Actually there's a little problem with the I2C_EEPROM code: it gets loaded even if EEPROM_SETTINGS is not enabled, which is useless in that case. I don't know if that is the real problem, but maybe we should also fix this

iosonopersia, 2017-10-26:
Ok @Hans007a : we know from other issues that commit c1f5fe5 (8 days ago, which was 18th october) broke a lot of things. #269 , #283 and this issue (#289) all seem to be related to that specific commit. Just to be sure, could you please try with 3d97cf4 (which is the version previous than c1f5fe5)? This will tell us if something bad happened between 15 october and 18 october; otherwise we will be sure that c1f5fe5 broke everything and we will focus on that (unluckily it's a big commit...). Let us know the result of this test.

Hans007a, 2017-10-26:
Ok, i  try to test it today and then post the results :)

Hans007a, 2017-10-27:
Any news about fix this?

sh4jonny, 2017-10-27:
hi! my RRFG display still not work. what i can check for fix it?

Hans007a, 2017-10-27:
use commit from post of @iosonopersia

iosonopersia, 2017-10-27:
Today MagoKimbra got his focus on #293 issue, he is waiting some feedback from @apphead before going on, I suppose

Hans007a, 2017-10-30:
I found another display dug.

I have capacitive z probe, but those is not exact and i switched it off and wanted to use a mechanical switch.
After uploading the firmware the display was broken.
I changed the settings to the z probe and the display works...

TROYANECPRO, 2017-10-30:
[image]
how to fix

closed 2017-10-30 by MagoKimbra

Hans007a, 2017-10-30:
it works now? i wills see..

TROYANECPRO, 2017-10-31:
Yes . Yes thank you

commit 672e1907 (2017-10-30): fix #289