MKFirmware/MK4duo #293: RADDS PIN13 not working - and other curiosities

apphead, 2017-10-25:
Since this issue is troubling my mind for a while now, I decided to simplify things and run my test afterwards. As I mentioned in other issues, running Repetier 0.92 shows that all mosfets / pins actually work. So a HW malfunction can be crossed off the list.
Using latest master 6a859f5578f074727ae55ed5853a804d7be0a6a0

HW-Setup as usual: RADDS / Due / EEPROM on I2C removed / no SD (to exclude another potential source of error). Using Pronterface on Windows 8.1

### Test 1 / Heaters
#define EXTRUDERS 3
#define FAN[0-5]_PIN NoPin
#define TEMP_SENSOR_[0-2] 1 (with thermistors)

TC 1.1: (mosfets / pins known to work as either heater or fan)
#define HEATER_0_PIN        12
#define HEATER_1_PIN        9
#define HEATER_2_PIN        8

Everything is fine here.

TC 1.2: (no luck with pins 13 or 11 so far)
**#define HEATER_0_PIN        13 -> fail**
#define HEATER_1_PIN        12 -> pass
**#define HEATER_2_PIN        11 -> fail**

only HEATER_1_PIN was responsive. 

TC 1.3: (swapping failed and passed pins from TC 1.2)
#define HEATER_0_PIN        12 -> pass
**#define HEATER_1_PIN        13  -> fail**
#define HEATER_2_PIN        8  -> pass

Pin 13 failing

TC 1.4: (swapping heater of failing pin)
**#define HEATER_0_PIN        11 -> pass
#define HEATER_1_PIN        13 -> fail
#define HEATER_2_PIN        12 -> fail**

Now this is strange. All of a sudden, Pin 11 is working, BUT 12 fails????
Checked Pinout with M43 and everything seems in order.
PIN: 11        HEATER_0_PIN                       Input  = 0
PIN: 12        HEATER_2_PIN                       Input  = 0
PIN: 13        HEATER_1_PIN                       Input  = 0
Ran the same test a couple of times with the same result.

TC 1.5: (swapping heater 0 and 1. Expecting to see 13 working as well. Lets see)
#define HEATER_0_PIN        13 -> fail
#define HEATER_1_PIN        11 -> pass
#define HEATER_2_PIN        12 -> fail

Not quite what I was hoping for or expecting. To make a 100% sure that Pin13 is actually working, I flashed Repetier again and voila, both 11 and 13 run heaters 0 and 1 just fine. Definitely a Firmware issue!

Hope, this helps to solve the mistery!

I was gonna do the same tests with all the fans, but I wait for your feedback before I continue my tests and will put my results in this post to keep it tidy.

### Test 2 / Fan

MagoKimbra, 2017-10-26:
Please test with new fix...

MagoKimbra, 2017-10-27:
Do you have a servo?

MagoKimbra, 2017-10-27:
I found the problem. I did the tests with a basic configuration and pins 13 and 11 worked ok, but with the new hardware PWM management system, the pin of servo and heaters on radds go in conflict by blocking the output at 13. Now I moved the timers of the servo on others not used by the pins of the heaters, seems to be all good at least at test. Please confirm it!

apphead, 2017-10-29:
OK, I re-run all the tests:

//TC 1.1: mosfets / pins known to work as either heater or fan
#define HEATER_0_PIN 12 // -> pass
#define HEATER_1_PIN 9 	// -> pass
#define HEATER_2_PIN 8	// -> fail

//TC 1.2: (no luck with pins 13 or 11 so far)
#define HEATER_0_PIN 13 // -> pass
#define HEATER_1_PIN 12 // -> pass
#define HEATER_2_PIN 11 // -> fail

//TC 1.3: (swapping failed and passed pins from TC 1.2)
#define HEATER_0_PIN 12 // -> pass
#define HEATER_1_PIN 13 // -> pass
#define HEATER_2_PIN 8 // -> fail

//TC 1.4: (swapping heater of failing pin)
#define HEATER_0_PIN 11 // -> pass
#define HEATER_1_PIN 13 // -> pass
#define HEATER_2_PIN 12 // -> fail

//TC 1.5: (swapping heater 0 and 1. Expecting to see 13 working as well. Lets see)
#define HEATER_0_PIN 13 // -> pass
#define HEATER_1_PIN 11 // -> pass
#define HEATER_2_PIN 12 // -> fail

### Conclusion: 
All pins tested work, but heater_2 keeps failing - irrespective of assigned pin

###  G-code script
By the way, this is the script I run in pronterface after flashing firmware:

M117 Select Extruder 0 and start heater for 3 seconds
T0;
M104 S100;
G4 S3; 
M104 S0;
M105; // Report current temp

M117 Select Extruder 1 and start heater for 3 seconds
T1; 
M104 S100;
G4 S3; 
M104 S0;
M105; // Report current temp

M117 Select Extruder 2 and start heater for 3 seconds
T2;
M104 S100;
G4 S3; 
M104 S0;
M105; // Report current temp

MagoKimbra, 2017-10-30:
How many extruder have you set??

apphead, 2017-10-30:
3 extruders (even though  I only need two, but RADDS offers three heaters, so that is why I chose three). See my edit in the post before for full setup.

MagoKimbra, 2017-10-30:
Ok, I asked this because in configuration_post HOTENDS = EXTRUDERS, so if the number of extruders is 2 the hotend are 2 and you can only control 2.
Very strange that does not work the third hotend !!!

apphead, 2017-10-30:
Strange indeed. I disabled all fan[0-5]_pins too, just to make sure that there is no interference....

MagoKimbra, 2017-10-30:
Can you see pid valor for all tree hotends? M503 and read it..

apphead, 2017-10-30:
I think I am going crazy! I ran the test again and the results are inconsistent. I noticed that I need to push the reset button on Due AND reconnect pronterface in order to "confirm" the new Pin settings - even though M43 gives the right settings immediately after uploading firmware / without reset. 

So these are the results WITH reset (difference to first run in bold):
//TC 1.1: mosfets / pins known to work as either heater or fan
#define HEATER_0_PIN 12 // -> pass
#define HEATER_1_PIN 9 	// -> pass
**#define HEATER_2_PIN 8	// -> pass**

//TC 1.2: (no luck with pins 13 or 11 so far)
#define HEATER_0_PIN 13 // -> pass
#define HEATER_1_PIN 12 // -> pass
#define HEATER_2_PIN 11 // -> fail

//TC 1.3: (swapping failed and passed pins from TC 1.2)
#define HEATER_0_PIN 12 // -> pass
#define HEATER_1_PIN 13 // -> pass
**#define HEATER_2_PIN 8 // -> pass**
Sometimes there is a delay between pin 12 and pin 8 and pin 8 never turn off after 3 seconds

//TC 1.4: (swapping heater of failing pin)
#define HEATER_0_PIN 11 // -> pass
#define HEATER_1_PIN 13 // -> pass
#define HEATER_2_PIN 12 // -> fail

//TC 1.5: (swapping heater 0 and 1. Expecting to see 13 working as well. Lets see)
#define HEATER_0_PIN 13 // -> pass
#define HEATER_1_PIN 11 // -> pass
#define HEATER_2_PIN 12 // -> fail

// New TC 1.6:
#define HEATER_0_PIN 12 // -> pass
#define HEATER_1_PIN 13 // -> pass
**#define HEATER_2_PIN 9 // -> pass**

// New TC 1.7:
#define HEATER_0_PIN 12 // -> pass
#define HEATER_1_PIN 13 // -> pass
**#define HEATER_2_PIN 11 // -> fail**

M503:
Config:  M301 H0 P40.000000 I7.000000 D60.000000
Config:  M301 H1 P40.000000 I7.000000 D60.000000
Config:  M301 H2 P40.000000 I7.000000 D60.000000

MagoKimbra, 2017-10-30:
Please repeat test with last commit.

apphead, 2017-10-31:
I slightly changed my testing procedure: First, I run the script above, then I turn on heater manually in pronterface. To my surprise, the results are inconsistent again....
Turns out that the first M117 command prevented the rest of the script from running properly!? Consequently, only Heater0 got enabled, the others remained off. After removal of M117, the results from the script were consistent with the manual tests. **Question remains, why that was not the case the last time I ran the tests!**

Using commit 96f2c4d3ac04ff6e943e5f2e25225c37e59cf350
	
[image]

### Conclusion
Heater0 and 1 always work, just heater2 is having hiccups. 
My hypothesis is false too, as it fails with TC 13.
I am running out of ideas...

apphead, 2017-11-04:
Any development I should test?

Prinz12, 2017-11-04:
Hello,
this problem does not affect me because I don't need three heaters, but as I read your table the problem is that pin 11 and pin 12 don't work at the same time.

apphead, 2017-11-04:
@Prinz12 thanks for you input. Did two more tests to prove your point and indeed, you're right:
[image]

apphead, 2017-11-12:
@MagoKimbra Any news on this issue

iosonopersia, 2018-08-05:
@apphead , pins 11 and 12 are linked to the same Timer Counter (TC2 Channel 2) of the SAM3X8E. So you can't use both to drive different hotends... Sorry for the late reply. I believe that this is a problem that we can't overcome via software.

This is the same for pins 2-13, 60-61, 58-92, 4-5 and 3-10...

closed 2018-08-05 by iosonopersia