MKFirmware/MK4duo #294: Feature request: Testing framework

apphead, 2017-10-26:
Outline of the requirements of a testing framework:

### Purpose
- Make it easier to run test configurations (motors, heaters, fans, endstops, leveling etc)
- Add test config to subfolder /testcases and include them either during compilation or ideally at runtime

### Example

iosonopersia, 2017-10-26:
We can make a "Configuration_Test.h" which is loaded in place of "Configuration_Overall.h" when a macro is enabled (let's say something like #define LOAD_TEST_CONFIG). This way you can write in Configuration_Test.h the settings that need to be modified over defaults in order to test a feature. That would simplify things a lot. I'll do this this afternoon...

apphead, 2017-10-26:
Hi iosonopersia , I guess that would make things a lot easier. But my idea is a little more thorough, will keep updating my first post as I make progress. So I suggest you wait a little longer before jumping right into implementing your approach.

iosonopersia, 2017-10-26:
The only configs uploadable at runtime are those which have a dedicated gcode... Some options need recompiling (for example, MECHANICS and MOTHERBOARD). Also, you can set a timeout with G4 S<seconds> or G4 P<milliseconds>. The gcode commands can be stored in a macro as a string, this would allow to parse and execute them automatically after power up. But you will always be forced to recompile and load every time. On the other hand, you will have the possibility to keep your "atomic" tests in a folder and use them occasionally to compile a "testing version" the firmware for a specific test. Perfect solutions hardly ever exist...

apphead, 2017-11-01:
During more testing of https://github.com/MKFirmware/MK4duo/issues/293# , I found that
- pins cannot be configured in config_overall.h but must be in config_pins.h (or boardcode.h)
- having the gcode script to run the test in the same file as the test config would keep everything tidy in one place, but has its downsides (recompile, DUE needs a physical reset for changed pins)

So let's keep it simple as you suggested: 
- #define LOAD_TEST_CONFIG "testcases/heaters.3.h", which loads instead of Configuration_Overall.h
- If LOAD_TEST_CONFIG is defined, then have the pins in the testcase actually override the ones in config_pins.h
- enable different setups within test_config.h (see [heaters.3.h.txt](https://github.com/MKFirmware/MK4duo/files/1435213/heaters.3.h.txt))
- keep accompanying gcode script in a separate file and just keep a reference in test_config.h to 
[heaters.3.gcode.txt](https://github.com/MKFirmware/MK4duo/files/1435219/heaters.3.gcode.txt)

Does that sound like a good plan to you too?

apphead, 2017-11-12:
@iosonopersia what is your take on my idea?

iosonopersia, 2017-11-14:
Hi, sorry. I think that would be a good idea. I'm going to implement it soon and we'll see whether it's ok or not

iosonopersia, 2017-11-15:
Hi @apphead. This morning I've opened a new branch in my repo of MK4duo: https://github.com/iosonopersia/MK4duo/tree/testing?files=1

You can try already using that version. You just have to open MK4duo.h and edit the TESTING_FRAMEWORK section. Since I've encountered some problems while implementing this feature, this is how it works now:

In MK4duo/tests there's a directory called test_config where you can put Overall and Pins configurations that you want to use for testing purposes. You can choose to load them or the ones in the MK4duo folder by enabling/disabling a macro in MK4duo.h

In MK4duo/tests you can find an example of test (warming_heaters.h). You can copy that and edit the gcodes inside (while maintaining the same structure) to create other test files. Then you can choose the one you want to load just by typing its name (without " and .h) in MK4duo.h (just open that file, you should understand without issues how to do it).

Then, compile and load. When the printer will be on, send a M777 command from a host: it will automatically start the test. When it finishes you can restart by sending another M777.

WARNING I'm pretty sure it won't work very well (I "feel" it). I'm building my printer these days so I can't test ir myself. Please give me some feedback

apphead, 2017-11-15:
Hi @iosonopersia  Thanks for your effort and comment. Unfortunately, I am moving my office/shop and have little time to test the next couple of days. Just FYI

iosonopersia, 2018-08-02:
I don't think implementing this directly in the official version of MK4duo would be of any interest for the vast majority of the users. I'm going to close this issue.

closed 2018-08-02 by iosonopersia