MKFirmware/MK4duo #298: Laser mode Raster and M3 laser without function.

Prinz12, 2017-10-27:
After a long time I wanted to use my laser with the current commit https://github.com/MKFirmware/MK4duo/commit/05f88edb480c78a07f13193485e6be7b54085f27
I found out that the laser raster mode G7 doesn't work anymore. 
Furthermore, the laser does not fire permanently with the M3 command, but only flashes briefly. 
Laser fires with G1 without restriction.
I can't tell you which version I had previously loaded.

My setup:
#define BAUDRATE 250000
#define MOTHERBOARD BOARD_RADDS
#define LASER
#define LASER_FIRE_G1
LASER_PWR_PIN 5/6/7 tested

All other settings are unchanged

iosonopersia, 2017-10-28:
Hi, make sure to send a M452 before using laser, that sets the current machine mode from printer to laser. (M451 to return back to printer mode). If that doesn't help, since I strongly believe laser code has not been modified for quite a lot of time, you may need to join this discussion: #118 (which here in Italy is also the ambulance emergency telephone number ... :-| ). In that issue we found that you can use Pronterface with laser raster and some working plugins were listed

Prinz12, 2017-10-28:
Unfortunately, it's not that. The commands M450-M453 are without function and give no feedback in the log of "RepetierHost" I suspect that there is somehow a problem with the multimode. The laser pulsed mode is apparently also without function. Only continuous mode with G1 works.

Prinz12, 2017-10-28:
A problem was the missing entry " #include "multimode/m450_m453. h" in the "gcode. h". Now the commands M450-453 work again but the rastermode is still without function.

iosonopersia, 2017-10-28:
Oops, thanks for spotting that. We'll fix it as soon as possible

Prinz12, 2017-10-29:
I think I identified the problem.

For example, if I execute the command "G1 X10 F60 S100 B1 P1 L1000, the laser should fire once per mm. 
The laser doesn't fire.

When I now execute the command "G1 X10 F60 S100 B1 P1000 L1000, the laser fires exactly once per mm. 
I have now changed line 872 in the Planner. cpp.
From
block->steps_l = labs (block->millimeters * laser. ppm);

towards

block->steps_l = labs (block->millimeters * laser. ppm * 1000);

I don't know if this is the right way to go, but now it seems to work without any problems.

closed 2017-10-30 by Prinz12