MKFirmware/MK4duo #303: RepRap full graphics display still don't works & more issues

Hans007a, 2017-10-31:
RepRap Display still don't works with the new version 4.3.29_dev with Fix #302 from 31.10.2017

Display don't starts and the the heater starts immediatelly to heat bed and hotend...

And in the pronterface I get this error message:
`Stats: Total: 1, Finished: duty <= pPwm->PWM_CH_NUM[ul_channel].PWM_CPRD" failed: file "../source/pwmc.c", line 272, function: PWMC_SetDutyCycle
Exiting with status 1.`

My Board is still Due with
`#define MOTHERBOARD BOARD_RAMPS_FD_V1`

MagoKimbra, 2017-10-31:
Ok please test with last commit.. Now on my board arduino due test have not error.

Hans007a, 2017-10-31:
It looks now good after my changes to the fan (is still inverted, but I think, the fan is physically NOT inverted or double inverted in the firmware).
EEPROM works, but after loading the config the bed and the heater immediately starts to heat and I have to reset the board... I have to switch off the EEPROM in the config because the not controlled heating.

SD Card works now very quick, thank You!

rafaljot, 2017-11-01:
I checked it now EEPROM, RAMPS-FD + Due. 
EEPROM works, 
RepRap full graphics display works

Hans007a, 2017-11-01:
@rafaljot check the loading of the EEPROM settings! the bed and nozzle starts immediately to heat and only reset the board stopt it. it's for me a big security issue!

rafaljot, 2017-11-01:
Yes, right. It happened to me randomly ! I think after switching to EPROM_SD and after hardware reset. It works just after flashing. M502, M500 works. 

I have i2c EPROM too.

Hans007a, 2017-11-01:
then my EEPROM must be defect, but I can't initialise it...

sh4jonny, 2017-11-01:
@Hans007a I had the same problems in the past!!! the bed and nozzle starts immediately to heat! but today i flash last commit from git reset eeprom and save it. everything is ok for first view.

rafaljot, 2017-11-02:
I think pins initiation should be very first task in firmware. especially when heater pins have inverted logic. Now if something goes wrong heater left on.

jmlcatarino, 2017-11-06:
Is this issue fixed by the last commits? Also if someone using ramps4due, the hacked ramps, have you guys managed to get the temps correct? I can only get the extruder heater temp correct but the bed temp is not changing wether I have a sensor or not, it's always on 92ºC. I did managed to get the pins working by turning true all false invert pins on the board .h . I'm using this one because the FD one is not responding to anything.

sh4jonny, 2017-11-06:
@jmlcatarino all work fine at last commit

MagoKimbra, 2017-11-06:
Good!!

jmlcatarino, 2017-11-06:
Ok, I haven't had the time to test the LCD because I've been struggling to put the boards to work. Unfortunately the FD v1 hacked is not working as it should, even after several changes in the code. The ramps4due, the ramps 1.4 hacked is working better but still have one issue to solve, that is the bed temp. For some reason, with or without a temp sensor it shows 92º always. I've been checking everything and comparing the pinout for both Due and Mega, in relation to where they were assigned, and it all seems ok. Only thing to do, tomorrow, is to verify the physical connections on the ramps board. For the FD I still haven't tested the temperature sensors, because it wasn't enabling the motors at all. I would really like to use the FD board but I guess it's going to trash after all the work and learning I had to put it to at least work the MOSFETs. Maybe I'll dedicate some time to it after I get around the temperature issue I'm having now on the other one. I'm hacking a CO2 laser engraver in order to have one of them, the Ramps4due, controlling it, and a new printer that I wanted to have a 32 bit controller with some faster drivers at u32 steps, the Ramps-FD but I guess I'll have to wait for a SmartRamps to arrive to give it a go. Or maybe I'll stick with what I know to work, a Ramps 1.4 and Mega…
At least I'm learning everyday a bit more…bit by bit ;)

jmlcatarino, 2017-11-07:
Update:

uploaded last master branch with latest commits and all except motors work ok on Ramps-FD V1.
Can't seem to find where to turn on the enable stepper pins. Must test ramps4due board. That one was working ok with LCD, steppers but had issues on temperature. Have to check with latest branch.
I @MagoKimbra or any of you guys could help me I'd be thankful.

Hans007a, 2017-11-07:
It look all OK for me (SD card sorts weird, but I think it is so) :)

Now I buy a BLTouch clone to change my capacitive sensor and BLTouch deploy the pin while printing... very weird...

iosonopersia, 2017-11-07:
@Hans007a it would be better to open a new issue for that... (you shouldn't set the BLTouch in the configurator if you use a clone, anyway)

jmlcatarino, 2017-11-07:
To me it's only the steppers not enabled on Ramps-FD and the bed temp acting weird on ramps4due. EEPROM and all other stuff still not tested. BTW found out the IDC10 sockets on the LCD adaptor for the FD board, were in a wrong direction, for the slot guide. Had to remove them and swap them in order to make the LCD work. Weird. If you guys could tell me where could I change anything in order to enable the steppers it would just solve my problems, because I would have the FD working on the printer and would just move the Ramps4due to the laser.

Hans007a, 2017-11-07:
I don't use configurator. I change the values directly in the config files and there is not a setting for clone bltouch...

jmlcatarino, 2017-11-09:
Ok guys, just to update my findings. Had one issue with LCD but I figure it out. The sockets for the IDC 10 were reversed. It works ok now. The issue were the steppers didn't work was solved after lots of thinking and searching but it was my fault, because I had them in the opposite direction. Dumb ass. I thought they would go like on the Ramps1.4 and luckily they didn't fried. Only on the Ramps4Due I have that bed temperature issue where it shows always 92º even with no sensor connected. No problem for now as it's going to the CO2 laser. One side note, even with the modifications, the FD still turns heatbed on power up, but for a short period time. Thank you all for the help and time taken to help.

rafaljot, 2017-11-09:
Do not get me wrong. I'm deadly serious. I'm afraid about you and CO2 laser if you make such mistakes :D.

jmlcatarino, 2017-11-09:
I don't take, you're right to be like that. This is the result of a very poor information on FD and the fact that I'm currently doing some thousand things at the same time. I have to learn on the go, be it electronics, be it Arduino and things related. It has been some extenuating weeks but I got where I wanted because I believed in MK4Duo and my own skills. Thanks to you guys and the open source community. Be assured that when it comes to dealing with the CO2 itself, I'll extra careful. So you can understand why so many projects at the same time, I'm in the process of creating a hack space in my home town and I've built three CNC router machines, three (2.5, one is in building process), 3D printers and next a laser machine. So you can imagine the pressure to get things done, being the only skilled one to do these things, and the fact that sometimes things may be forgot or mistakes made. 
Thank you guys. ;)

MagoKimbra, 2017-11-09:
Thanks to you and to you for believing in MK4duo.
What do you mean by very poor information on FD?

jmlcatarino, 2017-11-09:
I mean, comparing to regular ramps 1.4 there is little information on Ramps-FD except for some youtube videos and the hacking tutorials. For example, I found one tutorial on one of the RepRap forums but the best one to follow was one on an Italian page, with pictures and schematic images. Still, even for something that has been done since 2014 or earlier, there are a few of us doing that mod and even fewer knowing that Mk4Duo exists. Many are still waiting for Marlin to get into the 32 bit world. So, I had to use all of my search, computer and electronics skills, which are basic, and with some trial and error managed to get at this point. Today is the day I'll assemble the board to the printer and do a test run. The other one, ramps4due I won't even bother with that temperature issue.

MagoKimbra, 2017-11-09:
Ok, FD was for rampsFD I did not understand ...
Good .. and thank you!

jmlcatarino, 2017-11-09:
Yes, sorry for using an abbreviation not so clear. ;) 

Grazie tanto Antonio, ;)

Hans007a, 2017-11-11:
I think, the issues are gone and I will close this thread :-)

Thank You!! :-))

closed 2017-11-11 by Hans007a