MKFirmware/MK4duo #305: SerialUSB in DUE

rafaljot, 2017-11-01:
In Arduino DUE Native port is -1 

#define SERIAL_PORT -1
or 
#define SERIAL_PORT SerialUSB
works.
 
There is no way to set it from  Firmware configurator tool

ManBearPigWolf, 2017-11-04:
Forgive me if I'm being ignorant, but why would one use the native port? The programming port is already setup for serial communication. Not disagreeing with you, it should be added to the configurator just so the option is there, but I don't see why you would want to swap to the native port for normal operation.

rafaljot, 2017-11-04:
I use both. Programing port for flashing, native port is connected to OctoPrint.

jf1452, 2018-01-14:
I'm just looking for the same thing. The programming port connector just broke on mine, and I cannot get the native port working.

jf1452, 2018-01-14:
Having manually edited the line in the code; programming by Native takes approx 1/3 of the time compared to the programming port.

MagoKimbra, 2018-01-14:
I had removed the option -1 for the Serial usb, because many who used cards with 8-bit with the mega put this option because they connected with the USB ...

jf1452, 2018-01-15:
@MagoKimbra Thanks for the explanation, I can manually edit Configuration_Overall, thanks to rafaljot for describing the edits required; it would have taken me many hours to work this out on my own.

closed 2018-08-02 by iosonopersia