MKFirmware/MK4duo #307: Ramps-FD pin assignment not responding

jmlcatarino, 2017-11-01:
## Issue template
When I send bed and heater on on Pronterface or serial monitor by means of M code, Due does not respond. 

### Issue description
I've uploaded last master branch and last commit from a few hours ago and I can't get the heaters on. 
I've tested my DUE with basic blink and it works ok on every pin used for heaters and bed. 
If I upload with RAMPS4DUE board it will recognize only heater command. Can't figure out what is this problem because it isn't electronics but fw.

sh4jonny, 2017-11-01:
Ramps-FD this is not RAMPS4DUE i think. check your pin definition

jmlcatarino, 2017-11-01:
@sh4jonny  I know but pin assignment in what heater and bed heater concerns, are the same, with some slight differences. I'm just playing with the Due for now, but I've been all day playing around the two boards, DUE & Ramps-FD in order to figure out why is that I can turn on and off the two parameters on Ramps4due but not on FD. It bugs me off and is messing with my mind. I've hacked two boards in order to have at least one functional but has been a fight.

rafaljot, 2017-11-02:
RAMP-FD is RAMPS_FD_V1 in configuration and it works. I tested it yesterday with lastest dev branche
`#define MOTHERBOARD BOARD_RAMPS_FD_V1`
**be aware, do nor use //#define EEPROM_SD**

jmlcatarino, 2017-11-02:
@rafaljot , hi I know which board it is but every time I issue a heat bed command from Pronterface, for exemp., it just doesn't do anything. I have just the DUE for testing purposes and only when I use V2 or Ramps4due board, the extruder heater and fan work. Another thing that bugs me is that on the board, D8 or pin 8 is the heated bed output and that is correct on pin assignment but for extruder heater 0 it is pin 9 where it should be 10, as per ramps-fd board outputs. Nevertheless, it just doesn't work and I've uploaded last master branch and even corrected with last commit. I know that before, with an older version, which I don't have or know which one, it used to work at least with ramps4due pin assignment, except for the LCD, but that's another story.

rafaljot, 2017-11-02:
> for extruder heater 0 it is pin 9 where it should be 10, as per ramps-fd board outputs.

No, @jmlcatarino,  look at this board.
[http://reprap.org/wiki/File:RAMPS-FD-pinout.jpg](url)
 You have on big blue connector for bed. It is D8, It is correct in src/boards/1403.h
and 3 small green for heaters D9, D10 and D11 (E0,E1,E2), also correct in src/boards/1403.h 
Why are you saying extruder heater 0  should be D10 ?

jmlcatarino, 2017-11-02:
@rafaljot well, I was following the layout form the previous ramps, never mind. The problem is that if I issue the command to heat up the bedd, it simply doesn't work. But I'll do some more tests because I think that just by reloading the sketch and compiling it doesn't change things. I'm using mac so I guess I have to close and open IDE again.

rafaljot, 2017-11-02:
I'm using OSX as well ;) I change printer name every time :D to be sure what I upload.
You don't need restart Arduino IDE.

jmlcatarino, 2017-11-02:
It buggs me off each time I change the original sketch and have to save it with another name, whe I try to upload it again it just doesn't go get the libraries. So I have to manually copy them the that new file.

rafaljot, 2017-11-02:
Hmm, Ty to use Atom + PlatformIO

jmlcatarino, 2017-11-02:
Well, I've uploaded again and still nothing, no heater 0 no heat bed it simply doesn't pass the signal. But if I upload with ramps4due board it's works the D8, D9 and D10 from pronterface. Can't figure out what is causing this issue.

rafaljot, 2017-11-02:
say more. You have RAMPS-FD v1 (this one http://reprap.org/wiki/File:RAMPS-FD-pinout.jpg) with extruder's heater 0 connecteted to D9, green connector next to blue one. if you use 1433  it works when you switch to 1403 doesn't work, no more changes. correct?

jmlcatarino, 2017-11-02:
Yes correct. I use the normal heater button, like PLA 200 and heater bed the same, the ones in the GUI. I even tried using M codes directly but with no avail. Something must change when I choose that board. If I choose V2 it will work on the heater but not on D8. I'm not using the board for the tests, just using a simple LED and resistor directly on DUE. I've hacked both a FD v1 and a regular ramps 1.4 board in order to work with DUE and the only working one is the one with ramps4due. After I have to check the thermistors because for now I'm just using dummies.

rafaljot, 2017-11-02:
Sorry, it is impossible it works with D9 and 1433 board. Check schema and 1433.h file:
#define ORIG_FAN0_PIN 9
You must have mess in configuration, then check terminators are ok, it won't run if don't see terminators.

rafaljot, 2017-11-02:
OK I get it. but I have to imagine it. It happened when somebody won't write all details....
what about power? Do you connected all power lines?  v_motors, v_heatbed as described here
[http://reprap.org/wiki/File:RAMPS-FD-pinout.jpg](http://reprap.org/wiki/File:RAMPS-FD-pinout.jpg)
Probably not. you connected only v_power. Fans are powered by v_power line, in 1433 D9 is fan
#define ORIG_FAN0_PIN 9
fan starts in parallel when you try to start heating, so you think it is heater. 

Call me Holmes :D

jmlcatarino, 2017-11-02:
It works I can assure you. Only difference is that for D9 to work with ramps4due pins, I have to issue a M106 command. Terminators or thermistors? Maybe it's me who is not explaining well but this is the best I can explain. The only thing I changed within the boards were the pin for the given functions and they were set back as soon as I realized it wouldn't work.

rafaljot, 2017-11-02:
OK I get it. but I have to imagine it. It happened when somebody won't write all details....
what about power? Do you connected all power lines? v_motors, v_heatbed as described here
http://reprap.org/wiki/File:RAMPS-FD-pinout.jpg
Probably not. you connected only v_power. Fans are powered by v_power line, in 1433 D9 is fan
#define ORIG_FAN0_PIN 9
fan starts in parallel when you try to start heating, so you think it is heater.

Call me Holmes :D

jmlcatarino, 2017-11-02:
As told before, for tests purposes only use the bare Arduino board and an LED.

rafaljot, 2017-11-02:
yes, just before. and it was a clue.

rafaljot, 2017-11-02:
so. Connect everything right with documentation, select proper board in config and it will works.

_solved_

jmlcatarino, 2017-11-02:
You don't need the ramps board for testing. What it does is just sending a 5 v signal to the pin and that signal activates the MOSFET. That signal is enough to lit an LED and so you can check it's working. I know  the pins on each board have different functions but you can make them work from different ways, I just use this.
And no it won't work. It's not a noob dumb issue.

rafaljot, 2017-11-02:
[https://www.geeetech.com/wiki/images/8/83/RAMPS-FD-Schematic.pdf](https://www.geeetech.com/wiki/images/8/83/RAMPS-FD-Schematic.pdf)
Find D0202  Are you saying about this LED ? So that's why this discussion is so long. You didn't provide all details about your configuration.
So find D0202 and say what do you see, pls.

jmlcatarino, 2017-11-02:
I'm sorry but you are not getting anywhere. Why are you mentioning the schematic and some LED when explicitly wrote " an LED and a resistor"?  And why are insisting on the ramps board when I wrote that I'm using just the Arduino board? Are you understanding what I'm writing ?

jmlcatarino, 2017-11-02:
@iosonopersia can you give some light on this one?

rafaljot, 2017-11-03:
just Due, LED and resistor, whiteout Ramps-FD shield?  What about thermistor? Any heaters won't run whiteout thermistor.

jmlcatarino, 2017-11-03:
If you use the dummy thermistor they will, that's what they're for, for testing purposes. 

From Configuration_Temperature.h
"*       Use these for Testing or Development purposes. NEVER for production machine.                *
 * 998 : Dummy Table that ALWAYS reads 25 degC or the temperature defined below.                     *
 * 999 : Dummy Table that ALWAYS reads 100 degC or the temperature defined below.  "

jmlcatarino, 2017-11-03:
But you might be right in some point about thermistors and that's what I needed to confirm.

rafaljot, 2017-11-03:
and you know, RAMPS-FD configuration has inverted logic for heaters....if you testing with LED on Duo pins.

jmlcatarino, 2017-11-03:
Yes I took that in consideration also, and inverted the input/output of the LED. But with the mod or hacked version, you have to have the 5V on those pins in order to trigger the MOSFET gate. Do you know where can I see that inver

jmlcatarino, 2017-11-03:
Ok, I reviewed the board pins file and set to false the inverted pins because on  hacked FD there's no need for it. it already inverted with the mod. Now to the next issue…
#define INVERTED_HEATER_PINS false
#define INVERTED_BED_PIN false
Thanks @rafaljot for helping me solve this issue and pointing me in the right direction. I had tested this solution but somehow it didn't work at the time.
Will close this issue.

closed 2017-11-03 by jmlcatarino