MKFirmware/MK4duo #308: infinity beep

rafaljot, 2017-11-02:
my configuration is RAMPS-FD v1 + DUE + REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER, Printer HyperCube Evolution with dual Z

When press dialer in menu beeper starts ringing and does it for the next press. 

In buzzer.cpp I added at the end of void Buzzer::buzz(long duration, uint16_t freq)
....
WRITE(BEEPER_PIN, LOW);

and same in ui.cpp at the end of  void beep(uint8_t duration, uint8_t count) {

it works, now I can hear just short beep. I know it isn't right solution.

iosonopersia, 2017-11-05:
Hi, BEEPER_PIN is the value that actually is used inside the firmware, while ORIG_BEEPER_PIN is a value defined in the board files because it's board-specific. That said, if you don't change it with the online configurator, the firmware will automatically get the ORIG_BEEPER_PIN value from the board that you've selected and it will do something like this: `#define BEEPER_PIN ORIG_BEEPER_PIN` inside Configuration_Pins.h

Actually, if in 1400.h there's a BEEPER_PIN, then that's a mistake that should be fixed.

As far as the "infinite beep" issue is concerned, @MagoKimbra has lately released a commit that should bring to a better behaviour of the firmware. Would you mind trying the latest dev version with the same configs as before and checking if the beeper works as expected?

iosonopersia, 2017-11-05:
I've corrected every board file. Now there's only ORIG_BEEPER_PIN.

iosonopersia, 2017-11-07:
@rafaljot, could you give us some feedback on the latest dev version?

rafaljot, 2017-11-07:
OK. It prints!
1. Only very short beep. It could be longer but it is enough 
2. Duble Z with dual Z min endstops works but I don't know how to define it without warnings like this:
`sketch/src/utility/../../src/../Configuration_Pins.h:153:0: warning: "Z2_MIN_PIN" redefined [enabled by default] `
This is my configuration:
- Configuration_overall.h
- Configuration_pins.h
[Configuration HyperCubeEvolution250x300_DualZ_DualZendstops.zip](https://github.com/MKFirmware/MK4duo/files/1451873/Configuration.HyperCubeEvolution250x300_DualZ_DualZendstops.zip)

iosonopersia, 2017-11-07:
Those are harmless warnings, they're not a big issue. It's difficult to track down every redefinition of every macro, since the code is full of macros...

You can add these lines to your confoguration_overall:
``` cpp
#define LCD_FEEDBACK_FREQUENCY_DURATION_MS 100
#define LCD_FEEDBACK_FREQUENCY_HZ 1000
```

Edit those 2 values as you prefer ;-)

rafaljot, 2017-11-07:
Thank you. :) So, time to close.

closed 2017-11-07 by rafaljot