MKFirmware/MK4duo #313: Homing to Z-MIN with fixed z probe and safe z homing disabled does not work

Zwaubel, 2017-11-06:
### Issue description
I'm having trouble to configure MK4duo correctly to use both a fixed z probe and the z min endstop.
When telling the printer to auto home (or giving it a G28) it only moves till it triggers the z probe. But since I'm using a inductive probe which triggers -2.45mm under the nozzle, it never reaches the z min endstop.
I disabled safe homing and made sure that everything is connected correctly (tested with a working Marlin FW setup).

### Firmware version
MK4duo out of the dev branch

### My setup
Custom cartesian printer with RAMPS + Arduino Mega, RRD full graphics display and drv8825 drivers, z probe is connected to z max.
Config files here: 
[Config_dev.zip](https://github.com/MKFirmware/MK4duo/files/1445691/Config_dev.zip)

I'd appreciate any help!

MagoKimbra, 2017-11-06:
The endstop z min serves only for safety if the probe does not work so it is right to stop when triggering the probe.

Zwaubel, 2017-11-06:
okay, thats interesting. I was expecting the behaviour that I know from Marlin FW where a homing command makes the printer move the extruder towards the configured endstops no matter if a probe is configured or not.

sh4jonny, 2017-11-07:
@Zwaubel have you a strange sound and move speed of Z homing and movement when fixed z probe enabled ?

Zwaubel, 2017-11-07:
No, it's all working fine. I was just expecting the printer to home until it hits the min endstop, but MarlinKimbra is handling this differently than other firmwares as @MagoKimbra explained.

> strange sound and move speed of Z homing and movement < to me sounds like there might be something wrong with your motor current settings or steps per unit regarding the z stepper driver, although it wouldn't make sense that this occurs only with enabled z probe.

closed 2018-03-06 by Zwaubel