MKFirmware/MK4duo #315: Compiling Error Z2_Enable

wext1, 2017-11-07:
### Issue description

i cant compile my Sketch, 
when i compile with "#define Z_TWO_STEPPER" no problem but
when i try to compile with "#define Z_THREE_STEPPER" 
i get this Compile ERROR

### Compile errors - IMPORTANT

In file included from sketch\src\../src/motion/stepper.h:50:0,

                 from sketch\src\../base.h:122,

                 from sketch\src\MK_Main.cpp:31:

sketch\src\MK_Main.cpp: In function 'void gcode_M18_M84()':

sketch\src\../src/motion/stepper_indirection.h:595:85: error: 'Z2_ENABLE_WRITE' was not declared in this scope

   #define disable_Z() do{ Z_ENABLE_WRITE(!Z_ENABLE_ON); Z2_ENABLE_WRITE(!Z_ENABLE_ON);  Z3_ENABLE_WRITE(!Z_ENABLE_ON); axis_known_position[Z_AXIS] = false; }while(0)

                                                                                     ^

sketch\src\MK_Main.cpp:7630:27: note: in expansion of macro 'disable_Z'

       if (code_seen('Z')) disable_Z();

                           ^
### Firmware version
MK4DUO 4.3.2

iosonopersia, 2017-11-07:
@wext1, please, download again the firmware and reopen MK4duo.ino. It seems that old code has been mixed up with the new code... Are you sure having downloaded the 4.3.29 dev version?

iosonopersia, 2017-11-07:
@wext1, you can share your configuration with us by saving it in a .txt file and then Drag-and-dropping it inside the textarea of your post.

wext1, 2017-11-08:
Hi,

okay now i downloaded the Firmware again, and now i get an other Error.

ERROR:
In file included from sketch\MK4duo.h:138:0,

                 from C:\Users\eowegs\Downloads\MK4duo_08-11-2017_121547\MK4duo\MK4duo.ino:288:

sketch\src/feature/bedlevel/bedlevel.h:68:18: error: 'matrix_3x3' does not name a type

           static matrix_3x3 matrix; // Transform to compensate for bed Level

i use an Arduino DUE with RADDS and 3 Z-Stepper.

wext1, 2017-11-08:
[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/1453663/Configuration_Overall.txt)

iosonopersia, 2017-11-09:
@wext1, please, still a little thing to do. Can you delete the configuration from your initial  post? This issue needs to be cleaned up before starting to dig into it...

iosonopersia, 2017-11-10:
@MagoKimbra , in src/feature/bedlevel/bedlevel.h , at line 42, in order to make it work,
``` cpp
#if ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_UBL)
```
should be replaced by this:
``` cpp
#if ABL_PLANAR || ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_UBL)
```

But then, the other errors related to Z_TWO_STEPPERS occur again...

MagoKimbra, 2017-11-10:
Ok  i fond the error..

In configuration_pins:

```
// Z2 axis pins
#if ENABLED(Z_TWO_STEPPER)
  #define Z2_STEP_PIN       ORIG_E1_STEP_PIN
  #define Z2_DIR_PIN        ORIG_E1_DIR_PIN
  #define Z2_ENABLE_PIN     ORIG_E1_ENABLE_PIN
  #define Z2_CS_PIN         ORIG_E1_CS_PIN
#endif

// Z3 axis pins
#if ENABLED(Z_THREE_STEPPER)
  #define Z3_STEP_PIN       ORIG_E2_STEP_PIN
  #define Z3_DIR_PIN        ORIG_E2_DIR_PIN
  #define Z3_ENABLE_PIN     ORIG_E2_ENABLE_PIN
#endif

// Z4 axis pins
#if ENABLED(Z_FOUR_STEPPER)
  #define Z4_STEP_PIN       ORIG_E3_STEP_PIN
  #define Z4_DIR_PIN        ORIG_E3_DIR_PIN
  #define Z4_ENABLE_PIN     ORIG_E3_ENABLE_PIN
#endif
```

iosonopersia, 2017-11-10:
@wext1 now it should compile, try it out please!

closed 2018-08-04 by iosonopersia