MKFirmware/MK4duo #318: Compiling error - version from 08.11.2017

Hans007a, 2017-11-08:
sketch\src\feature\bedlevel\abl\abl.cpp: In member function 'void AutoBedLevel::bilinear_line_to_destination(float, uint16_t, uint16_t)':

sketch\src\feature\bedlevel\abl\abl.cpp:353:81: error: 'ABL_BG_FACTOR' was not declared in this scope

       int cx1 = (mechanics.current_position[X_AXIS] - bilinear_start[X_AXIS]) * ABL_BG_FACTOR[X_AXIS],

                                                                                 ^

sketch\src\feature\bedlevel\abl\abl.cpp:359:7: error: 'cy1' was not declared in this scope

       cy1 = constrain(cy1, 0, ABL_BG_POINTS_Y - 2);

       ^

sketch\src\feature\bedlevel\abl\abl.cpp:360:7: error: 'cx2' was not declared in this scope

       cx2 = constrain(cx2, 0, ABL_BG_POINTS_X - 2);

       ^

sketch\src\feature\bedlevel\abl\abl.cpp:361:7: error: 'cy2' was not declared in this scope

       cy2 = constrain(cy2, 0, ABL_BG_POINTS_Y - 2);

       ^

In file included from sketch\src\feature\bedlevel\abl\../../../../MK4duo.h:52:0,

                 from sketch\src\feature\bedlevel\abl\abl.cpp:23:

sketch\src\feature\bedlevel\abl\abl.cpp:383:45: error: 'gcy' was not declared in this scope

       else if (cy2 != cy1 && TEST(y_splits, gcy)) {

                                             ^

sketch\src\feature\bedlevel\abl\../../../../src/macros.h:102:36: note: in definition of macro '_BV'

   #define _BV(b)            (1 << (b))

                                    ^

sketch\src\feature\bedlevel\abl\abl.cpp:383:30: note: in expansion of macro 'TEST'

       else if (cy2 != cy1 && TEST(y_splits, gcy)) {

                              ^

Multiple libraries were found for "Wire.h"
 Used: D:\Arduino\arduino-1.8.3\portable\packages\arduino\hardware\sam\1.6.11\libraries\Wire
 Not used: D:\Arduino\arduino-1.8.3\libraries\Wire
exit status 1
Error compiling for board Arduino Due (Programming Port).

iosonopersia, 2017-11-08:
I've just proposed the fix, someone else told us about this problem some minutes ago... Thanks

Hans007a, 2017-11-08:
I found the bug by myself in abl.cpp at line 351:

```
    void AutoBedLevel::bilinear_line_to_destination(float fr_mm_s, uint16_t x_splits/*= 0xFFFF*/, uint16_t y_splits/*= 0xFFFF*/) {

      int cx1 = (mechanics.current_position[X_AXIS] - bilinear_start[X_AXIS]) * ABL_BG_FACTOR(X_AXIS),
          cy1 = (mechanics.current_position[Y_AXIS] - bilinear_start[Y_AXIS]) * ABL_BG_FACTOR(Y_AXIS),
          cx2 = (mechanics.destination[X_AXIS]      - bilinear_start[X_AXIS]) * ABL_BG_FACTOR(X_AXIS),
          cy2 = (mechanics.destination[Y_AXIS]      - bilinear_start[Y_AXIS]) * ABL_BG_FACTOR(Y_AXIS);
```

closed 2017-11-08 by Hans007a