MKFirmware/MK4duo #320: BLTouch clone deploy after start of print

Hans007a, 2017-11-08:
Hello,

After my tests with my new BLTouch clone I have allways an issue.
Directly after start of print (a few seconds), after bed leveling (without problems), the probe deploy the pin and is going into error state after few seconds. The pin drags on the bed...

I checked all settings, with and without #define BLTOUCH with different delays for BLTOUCH and disable servo (too tiny delays got the nozzle to go up)

My printer is a prusa i3 clone with RAMPS-FD v1 and Due with #define ENDSTOP_INTERRUPTS_FEATURE enabled.

Hans007a, 2017-11-09:
I tryed with #define BLTOUCH and the deploy and stow works better, but the issue with the deploy after start print (extrude filament) after 10-20sec is still existent...

Hans007a, 2017-11-10:
I found the issue, very dumb by my own... bad servo connection...

closed 2017-11-10 by Hans007a

jbaiao, 2017-12-12:
Can you share Confuguration file for bltouch clone?