MKFirmware/MK4duo #321: cannot compile

dembach, 2017-11-09:
In file included from sketch/src/HAL/HAL_AVR/../../../MK4duo.h:96:0,
                 from sketch/src/HAL/HAL_AVR/HAL_AVR.cpp:57:
sketch/src/HAL/HAL_AVR/../../../src/pins.h:33:36: fatal error: boards/BOARD_RAMPS_13_HFB.h: No such file or directory
 #include INCLUDE_BY_MB(MOTHERBOARD)

                                    ^
compilation terminated.
Mehrere Bibliotheken wurden für "Boards.h" gefunden
 Benutzt: /Users/dembach/Documents/Arduino/libraries/Firmata
 Nicht benutzt: /Users/dembach/Desktop/Arduino.app/Contents/Java/libraries/Firmata
 Nicht benutzt: /Users/dembach/Desktop/Arduino.app/Contents/Java/libraries/Firmata
 Nicht benutzt: /Users/dembach/Desktop/Arduino.app/Contents/Java/libraries/Firmata
 Nicht benutzt: /Users/dembach/Desktop/Arduino.app/Contents/Java/libraries/Firmata
exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino Mega or Mega 2560.

dembach, 2017-11-09:
Update: 
cloned complete Repository and now i am able to compile with errors:
As readable in config, TOWER Angles are set, also Z-MIN-PIN is set for ZPROBE:

In file included from sketch/src/sanitycheck.h:70:0,
                 from sketch/MK4duo.h:98,
                 from /Users/dembach/Desktop/BiQu Kossel Plus/MK4duo-master-09-11-2017/MK4duo/MK4duo.ino:288:
sketch/src/temperature/sanitycheck.h:145:13: error: missing expression between '(' and ')'
 #if (PIDTEMP)
             ^
sketch/src/temperature/sanitycheck.h:159:16: error: missing expression between '(' and ')'
 #if (PIDTEMPBED)
                ^
In file included from sketch/src/sanitycheck.h:82:0,
                 from sketch/MK4duo.h:98,
                 from /Users/dembach/Desktop/BiQu Kossel Plus/MK4duo-master-09-11-2017/MK4duo/MK4duo.ino:288:
sketch/src/mechanics/delta/sanitycheck.h:84:6: error: #error DEPENDENCY ERROR: Missing setting TOWER_A_ANGLE_ADJ
     #error DEPENDENCY ERROR: Missing setting TOWER_A_ANGLE_ADJ
      ^
sketch/src/mechanics/delta/sanitycheck.h:87:6: error: #error DEPENDENCY ERROR: Missing setting TOWER_B_ANGLE_ADJ
     #error DEPENDENCY ERROR: Missing setting TOWER_B_ANGLE_ADJ
      ^
sketch/src/mechanics/delta/sanitycheck.h:90:6: error: #error DEPENDENCY ERROR: Missing setting TOWER_C_ANGLE_ADJ
     #error DEPENDENCY ERROR: Missing setting TOWER_C_ANGLE_ADJ
      ^
In file included from sketch/src/sanitycheck.h:93:0,
                 from sketch/MK4duo.h:98,
                 from /Users/dembach/Desktop/BiQu Kossel Plus/MK4duo-master-09-11-2017/MK4duo/MK4duo.ino:288:
sketch/src/feature/probe/sanitycheck.h:74:6: error: #error "A probe needs a pin! Use Z_MIN_PIN or Z_PROBE_PIN."
     #error "A probe needs a pin! Use Z_MIN_PIN or Z_PROBE_PIN."
      ^
In file included from sketch/MK4duo.h:119:0,
                 from /Users/dembach/Desktop/BiQu Kossel Plus/MK4duo-master-09-11-2017/MK4duo/MK4duo.ino:288:
sketch/src/temperature/temperature.h:56:17: error: missing expression between '(' and ')'
     #if (PIDTEMP) && ENABLED(PID_ADD_EXTRUSION_RATE)
                 ^
Mehrere Bibliotheken wurden für "LiquidCrystal.h" gefunden
 Benutzt: /Users/dembach/Documents/Arduino/libraries/LiquidCrystal
 Nicht benutzt: /Applications/Arduino.app/Contents/Java/libraries/LiquidCrystal
exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino Mega or Mega 2560.

iosonopersia, 2017-11-09:
@dembach, hey, come on... Didn't you read the message while typing this issue? Please, delete the configuration from your post!!! It's unreadable like this, we cannot help you. Please delete it, leave only the issue description and then copy Configuration_Overall.h content into a thenamethatyouwant.TXT file. Now, you can Drag&Drop that file in your issue post. You can modify it, by clicking a button. Please, please, please...

iosonopersia, 2017-11-09:
Also it would be helpful if you put the erros in a txt file and upload it here...

dembach, 2017-11-09:
[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/1458931/Configuration_Overall.txt)

dembach, 2017-11-09:
better now ?

iosonopersia, 2017-11-09:
Waaaaay better :-) Thanks!

iosonopersia, 2017-11-10:
Hi @dembach . Actually the problem is that you used the configurator version related to the `4.3.2 stable` branch, while using the latest `4.3.29 dev` source code.

Since some settings changed their names in the meanwhile, you need to use the proper version of the configurator to configure the latest code.

Here's the correct link: http://marlinkimbra.it/configurator/v4_3_x/firmware_configurator.php

dembach, 2017-11-10:
i reused old configuration...... 
now i did a fresh one... attached here:
[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/1462716/Configuration_Overall.txt)
but now it is getting even more worse.....
[error.txt](https://github.com/MKFirmware/MK4duo/files/1462715/error.txt)

iosonopersia, 2017-11-10:
@dembach, it's not worse. You have to disable compiler warnings from the preferences tab... Of that endless file you posted, only this is an actual little error...

sketch/src/lcd/ultralcd.cpp: In function 'void _lcd_move_xyz(const char*, AxisEnum)':
sketch/src/lcd/ultralcd.cpp:2830:28: error: 'class Endstops' has no member named 'soft_endstop_min'
             min = endstops.soft_endstop_min[axis];

                            ^
sketch/src/lcd/ultralcd.cpp:2833:28: error: 'class Endstops' has no member named 'soft_endstop_max'
             max = endstops.soft_endstop_max[axis];

                            ^

dembach, 2017-11-10:
it´s a mac.... "compiler warnings" = none

iosonopersia, 2017-11-10:
No problem, but 99% of that output is from harmless and useless warnings. We'll try to fix that error soon

dembach, 2017-11-10:
ahh... switched to compiler warnings = enhanced and then back to none... now its fine....

dembach, 2017-11-10:
saw your changes and downloaded the complete repo...
the issue with the lcd is gone. perfect.... next one appeared:

`sketch/src/mechanics/delta/delta_mechanics.cpp: In member function 'virtual void Delta_Mechanics::report_current_position_detail()':
sketch/src/mechanics/delta/delta_mechanics.cpp:738:15: error: 'leveled' was not declared in this scope
     Transform(leveled);  // writes delta[]
               ^`

iosonopersia, 2017-11-11:
Try now

dembach, 2017-11-11:
Hey Simone,
working ! 
thanks

iosonopersia, 2017-11-11:
Great! You can close this issue

closed 2017-11-11 by dembach