MKFirmware/MK4duo #322: Rostock 301

gitamani, 2017-11-11:
Amo, creo e promuovo open hardware e software. Ho acquistato una stampante 3D Geeetech Rostock 301 senza riuscire a farla funzionare (ma non mi interessava farlo cos' com'era). Ho appena sostituito l'elettronica ma, con la procedura online, non riesco a produrre un firmware che non produca errori. La guida è troppo vecchia per riuscire a fare le modifiche sul firmware completo. Qualcuno mi può aiutare? Non mi interessa farla funzionare in color mixing quello è il problema minore: Basta che si muovano i motori. La mia Prusa Mendel è stata la prima in Trentino. Grazie in anticipo. Giuseppe Tamanini Trovate i miei lavori su filoviva.it, e su thingiverse e youtube.

iosonopersia, 2017-11-11:
Ciao, sono Simone. Forse la cosa migliore sarebbe se tu provassi a fare una configurazione e a mandarci sia il tuo Configuration_Overall.h (lo trovi dentro alla cartella zippata che scarichi dal configuratore online) sia il messaggio di errore di Arduino IDE. Per farlo, salva entrambi i file con estensione txt e fai drag&drop qui per condividerceli.

gitamani, 2017-11-11:
Grazie infinite per la celere risposta. Senza selezionare l'opzione Color Mixing Extruder la compilazione è stata completata senza errori.
Con Pronterface la stampante si connette e da le temperature. Ora devo alimentare la Ramps con i 12V e provare a  vedere se funzionano i motori, in che verso e se ho inserito i parametri giusti della Delta.
Appena ho testato tutto vi chiederò informazioni. Non so come si impostano i valori dei finecorsa che sono dalla parte opposta delle Cartesian.
Poi, per ultimo, come impostare il Color Mixing.
Grazie ancora per la disponibilità:

Giuseppe

iosonopersia, 2017-11-11:
Ciao, hai un sensore di prossimità (o comunque una probe per l'asse Z)? Nel caso questa guida potrebbe tornarti utile per la calibrazione della tua stampante: https://mk4duowiki.altervista.org/wiki/doku.php/guide:lucvandaele_autocalibrazione_delta

closed 2018-08-04 by iosonopersia