MKFirmware/MK4duo #324: Case_Light not working properly M355 S1 does not turn on lights

lifelight3d, 2017-11-16:
**Please, follow this template when opening a new issue: this will help us a lot in the process of helping you.**

### Issue description
Tell us here what happened: don't miss any valuable detail, be clear and concise, don't use UPPERCASE or abbreviations which can make understanding what you wrote quite difficult.

### Compile errors
If you have problems compiling MK4duo, report here the error messages Arduino IDE gives you.

Before posting this issue, however, take a long breath and check this two things:
1. If you're not using the latest version of Arduino IDE, please try again with the latest one.
2. Always check if an error (or more than one) starts with `DEPENDENCY ERROR: _something wrong!_`: **DO NOT POST DEPENDENCY ERRORS**, please! They are messages which tell you what config option was set badly. Fix your configuration and try again. If you also get other types of error, please be sure to have fixed dependency errors **before** posting this issue!

### Firmware version
MK4duo's versions that have this issue.

### My setup
If you used the online configurator, copy the content of `Configuration_Overall.h` and `Configuration_Pins.h` in two text files. **Then Drag&Drop them here. DO NOT COPY THE CONTENT DIRECTLY HERE!!!**

Have you configured MK4duo manually? Then write here a list of the settings that may be related to the issue:
- SETTING_CHANGED_1    true
- SETTING_CHANGED_2    25
- SETTING_CHANGED_3    "blablabla"
- ...

_If needed, you'll be asked for other settings you modified._

**Please be patient, we'll be trying to help you as soon as possible.**

lifelight3d, 2017-11-16:
M355 S0 works turning off case lights but M355 S1 also turns off case lights

lifelight3d, 2017-11-16:
also if its possible to implement a bit of code so a Nextion can switch case lights on or off without a relay or Enhanced Nextion.. the same way the fan can be toggled on and off (i have my lights in pin 11) as i use the 3.5 inch standard Nextion  just had to resize all the pictures and mod the nextion_cpp to connect to it.

maukcc, 2017-12-06:
Is it PWM based? you probably need more then 1 :) 255 is full on
if it is on a digital pin you probably need a value over 127
So try M355 S255

iosonopersia, 2018-01-09:
@lifelight3d please tell us if the @maukcc suggestion fixed your problem with caselight!

MagoKimbra, 2018-01-09:
No, M355 have S and P subcode, S is boolean 0-1 off-on P is brightness P0-255

iosonopersia, 2018-01-09:
@lifelight3d Have a look here:
```cpp
/**
   * M355: Turn case light on/off and set brightness
   *
   *   P<byte>  Set case light brightness (PWM pin required - ignored otherwise)
   *
   *   S<bool>  Set case light on/off
   *
   *   When S turns on the light on a PWM pin then the current brightness level is used/restored
   *
   *   M355 P200 S0 turns off the light & sets the brightness level
   *   M355 S1 turns on the light with a brightness of 200 (assuming a PWM pin)
   */
```

maukcc, 2018-01-09:
You can also try M42 (change pin state) to see if the pin works(unless it is a restricted pin)
So:
M42 P11 S0 is off
M42 P11 S255 is on

closed 2018-08-04 by iosonopersia