MKFirmware/MK4duo #325: Compile error with CNCROUTER option

darfij, 2017-11-19:
### Issue description
switching #define CNCROUTER in Configuration_Feature.h causes compile error 

#define LASER works as expected!

### Compile errors
In file included from C:\Users\tata\AppData\Local\Temp\arduino_build_622837\sketch\src\feature\cncrouter\../../../src/HAL/HAL_AVR/HAL_AVR.h:83:0,

                 from C:\Users\tata\AppData\Local\Temp\arduino_build_622837\sketch\src\feature\cncrouter\../../../src/HAL/HAL.h:43,

                 from C:\Users\tata\AppData\Local\Temp\arduino_build_622837\sketch\src\feature\cncrouter\../../../base.h:84,

                 from C:\Users\tata\AppData\Local\Temp\arduino_build_622837\sketch\src\feature\cncrouter\cncrouter.cpp:42:

C:\Users\tata\AppData\Local\Temp\arduino_build_622837\sketch\src\feature\cncrouter\../../../src/HAL/HAL_AVR/fastio.h:72:29: error: pasting "DIO" and "-" does not give a valid preprocessing token

 #define _SET_OUTPUT(IO) do {DIO ## IO ## _DDR |= _BV(DIO ## IO ## _PIN); } while (0)
                             ^

C:\Users\tata\AppData\Local\Temp\arduino_build_622837\sketch\src\feature\cncrouter\../../../src/HAL/HAL_AVR/fastio.h:84:25: note: in expansion of macro '_SET_OUTPUT'

 #define SET_OUTPUT(IO)  _SET_OUTPUT(IO)

,,,,,,,,,,,,,,,,,,,,,,
,,,,,,,,,,,,,,,,,,,,,
,,,,,,,,,,,,,,,,,,,,,
more simillar s _SET_OUTPUT'errors
,,,,,,,,,,,,,,,,,,,,
,,,,,,,,,,,,,,,,,,,,
,,,,,,,,,,,,,,,,,,,,
C:\Users\tata\AppData\Local\Temp\arduino_build_622837\sketch\src\feature\cncrouter\cncrouter.cpp:96:25: note: in expansion of macro 'get_Speed'

           saved_speed = get_Speed();

                         ^

C:\Users\tata\AppData\Local\Temp\arduino_build_622837\sketch\src\feature\cncrouter\cncrouter.cpp: In static member function 'static void Cncrouter::disable_router()':

C:\Users\tata\AppData\Local\Temp\arduino_build_622837\sketch\src\feature\cncrouter\cncrouter.cpp:165:26: error: 'WRITE_CNCROUTER' was not declared in this scope

       WRITE_CNCROUTER(LOW);

                          ^

C:\Users\tata\AppData\Local\Temp\arduino_build_622837\sketch\src\feature\cncrouter\cncrouter.cpp: In static member function 'static void Cncrouter::setRouterSpeed(uint32_t, bool)':

C:\Users\tata\AppData\Local\Temp\arduino_build_622837\sketch\src\feature\cncrouter\cncrouter.cpp:206:28: error: 'WRITE_CNCROUTER' was not declared in this scope

       WRITE_CNCROUTER(!!rpm);  

                            ^

### Firmware version
Version 4.3.2

### My setup
using
MOTHERBOARD BOARD_RAMPS_13_HFF
REPRAP_DISCOUNT_SMART_CONTROLLER 
(also tried with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER)
only board and lcd have been changed in setup

MagoKimbra, 2017-11-19:
Have you set CNCROUTER_PIN in configuration_pin.h?

darfij, 2017-11-19:
nope, I havent 
now that I have 

it is compiling fine !!!

thanks!!

closed 2017-11-19 by darfij