MKFirmware/MK4duo #334: Thermocouples & Ultratronics

pbrgle1, 2017-12-02:
**Please, follow this template when opening a new issue: this will help us a lot in the process of helping you.**

### Issue description
Tell us here what happened: don't miss any valuable detail, be clear and concise, don't use UPPERCASE or abbreviations which can make understanding what you wrote quite difficult.

### Compile errors
If you have problems compiling MK4duo, report here the error messages Arduino IDE gives you.

Before posting this issue, however, take a long breath and check this two things:
1. If you're not using the latest version of Arduino IDE, please try again with the latest one.
2. Always check if an error (or more than one) starts with `DEPENDENCY ERROR: _something wrong!_`: **DO NOT POST DEPENDENCY ERRORS**, please! They are messages which tell you what config option was set badly. Fix your configuration and try again. If you also get other types of error, please be sure to have fixed dependency errors **before** posting this issue!

### Firmware version
MK4duo's versions that have this issue.

### My setup
If you used the online configurator, copy the content of `Configuration_Overall.h` and `Configuration_Pins.h` in two text files. **Then Drag&Drop them here. DO NOT COPY THE CONTENT DIRECTLY HERE!!!**

Have you configured MK4duo manually? Then write here a list of the settings that may be related to the issue:
- SETTING_CHANGED_1    true
- SETTING_CHANGED_2    25
- SETTING_CHANGED_3    "blablabla"
- ...

_If needed, you'll be asked for other settings you modified._

**Please be patient, we'll be trying to help you as soon as possible.**

iosonopersia, 2018-08-25:
@pbrgle1 please test the latest version of MK4duo, we're trying to fix this...

MagoKimbra, 2018-09-01:
Remember the pins sensor is saved into eeprom...